WillCodeForCats/solaredge-modbus-multi #305: Storage default mode works for only 2 seconds

FranciscoBigWings, 2023-05-07:
The first thing that I have to say is that I coulnt find information in the Wiki related to:
- Storage commande mode
- Storage default mode
- Storage command timeout

While I select storage control mode to "Remote control mode" and all storage controls appear in the StorEdge inverter:
- I cant change "Storage commande mode", it is fix to "Charge from Clipped Solar Power", which was the state when I disabled the Time of use mode I was using through the Solaredge monitoring profile.
- I can change Sotrage discharge/charge limit
- I can change "Storage Default mode" but I cant get the battery to do what I want, if for example I choose the command "Maximize self consumption" while I am importing energy from the grid, the battery doesnt start to discharge to supply that energy instead of keep importing it. I can see that the battery receives the commands because it changes its lights for 2-5 seconds but after it goes back to the previous state of "doing nothing".

My configuration:
- SE leader is a EV of 5Kw
- SE follower is the Storedge of 5Kw with a LG RESU 10H
- Before I tried using the storage remote I disabled Storage control in leader and follower inverters and I disabled the Time of use profile in the SolarEdge monitoring portal linked to this plant.

Besides this issue I have some questions that I couldnt find in the Wiki either:
- What is the difference between "Storage commande mode" and "Storage default mode"?
- What is the use of Negative Site Limit?

Thanks, this is all a great work

WillCodeForCats, 2023-05-07:
I don't have batteries, I can't help with or answer any of these questions.

WillCodeForCats, 2023-05-07:
Moving to discussions since this isn't an integration issue.