WillCodeForCats/solaredge-modbus-multi #309: Incorrect calculation of values with Template "Sensors for Power and Energy"

bastelbert, 2023-05-08:
Hi, I've just installed this awesome plugin, thanks for your great work!
I get all the data from my SE17k, but the template for calculating the values does not seem to work correctly, at least for me. 
In fact, 6.4 kW were produced earlier, approx. 300 W were consumed in the house and the rest was fed into the grid. 
However, a consumption of just under 6.7 kW and a mains supply of 300W was calculated.
It seems as if the values in the template are reversed.

[image]
[image]
[image]
[image]
[image]

Used template:
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Sensors-for-Power-and-Energy

`template:                                                                       
  - sensor:                                                                     
      - name: "Power - Solar Generation"                                        
        unit_of_measurement: "W"                                                
        device_class: "power"                                                   
        state_class: "measurement"                                              
        # DC power from inverter
        state: "{{ float(states('sensor.solaredge_i1_dc_power'), 0) }}"
                                                                                                
      - name: "Power - Consumption"                                    
        unit_of_measurement: "W"                                                
        device_class: "power"                                                   
        state_class: "measurement"        
        # The net of inverter and grid power
        state: "{{ float(states('sensor.solaredge_i1_ac_power'), 0) - float(states('sensor.solaredge_m1_ac_power'), 0) }}"

      - name: "Power - Grid - Load"                                             
        unit_of_measurement: "W"                                                
        device_class: "power"                                                   
        state_class: "measurement"                                              
        state: "{{ min([float(states('sensor.solaredge_m1_ac_power'), 0), 0]) | abs() }}"
      - name: "Power - PV - Grid"                                             
        unit_of_measurement: "W"                                                
        device_class: "power"                                                   
        state_class: "measurement"                                              
        state: "{{ max([float(states('sensor.solaredge_m1_ac_power'), 0), 0]) | abs() }}"    
      - name: "Power - PV - Load"                                               
        unit_of_measurement: "W"                                                
        device_class: "power"
        state_class: "measurement"                                              
        # What is being generated less what is being exported, always positive.
        state: "{{ max([float(states('sensor.power_solar_generation'), 0) - float(states('sensor.power_pv_grid'), 0), 0]) }}"`

WillCodeForCats, 2023-05-08:
The templates were contributed, I've never used them personally.

bastelbert, 2023-05-08:
That explains a lot ;) How do you calculate the actual values?

WillCodeForCats, 2023-05-08:
I actually have two meters: an E+I meter and a consumption meter, so I have sensor values for import/export (+ or - depending which direction), and consumption values without templates. But if I didn't have the consumption meter I believe I'd only need a template sensor for the difference between import/export and solar production to give me a consumption value. So I haven't really looked at or checked any of the templates, I just trust that anyone who is actually using them are updating them if there are problems.

I only use Energy dashboard and it uses sensors from the E+I meter and solar production which don't need any templates to work.