WillCodeForCats/solaredge-modbus-multi #313: Try removing the inverter-battery check for storage commands

WillCodeForCats, 2023-05-10:
Hi @WillCodeForCats, I found the issue.

When sending a command to the inverter, this HA integration sends only the order to the StorEdge inverter which is the one that controls the battery, this is ok if your leader inverter is the StorEdge.

In my case the leader inverter is the SE EV charger (SolarEdge said that is the only way that both inverters can work and in fact the system doesn't work if my leader inverter were the StorEdge SE). Then if the follower is the StorEdge, the command doesn't work, it has to be the leader inverter that has to control the follower "Energy Control".

At least I tested my theory while using the Time of Use through SetApp:
- If I access through SetApp Power Control/Energy Manager/Energy Control in the leader inverter and disable it, it doesn't matter that the SE StorEdge follower inverter has Time of Use in Energy Control, it won't work
- If I select in both inverters, Power Control/Energy Manager/Energy Control --> Time of Use; then everything works

Summarising, I think that it should be as easy as giving the possibility in the HA integration to manage the StorEdge Control Mode in the leader inverter even if it is not in fact a StorEdge inverter. Then if both inverters have the same Energy Control mode, should work.

I guess that there are not many installations like mine where there are two inverters and the follower inverter is the StorEdge, so I can test it if you can give the possibility through configuration of the inverters the access to StorEdge Control Mode even if the inverter is not in fact a StorEdge; you could leave the feature as disabled by default.

Thanks

_Originally posted by @FranciscoBigWings in https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/307#discussioncomment-5854352_

WillCodeForCats, 2023-05-10:
The purpose of this change would be to always have storage controls available on any inverter that reports those functions are available over modbus, regardless if there is a battery associated with the same inverter modbus address or not. The possibility of such a condition was reported in discussion #307.

If there isn't a battery associated with the inverter's modbus ID but the inverter reports the storage control functions are available, then the associated entities will be disabled by default. This should sort of match existing behavior where the entities aren't created without an associated battery.

This is an architectural change that could have unintended results, so testers will be needed since I don't have any real batteries to check with.

FranciscoBigWings, 2023-05-10:
Hi. Once you start releasing the changes, I will be able to test it.

If energy controls appears in my non-StorEdge inverter also should appear in yours as well as on others, at least in the HD-WAVE. There are other options in SetApp that you can enable in inverter and don’t have any results.

WillCodeForCats, 2023-05-10:
My inverters do not. SolarEdge doesn't enable modbus control registers universally.

WillCodeForCats, 2023-05-10:
Yes I know how setapp works, but none of my inverters support controls over modbus. I've done a lot of work with them to make this integration. It's a common issue, but I've never cared to contact solaredge because there's nothing useful about controls for me with how my utility net metering works.

You can find other discussions about it here: https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues

WillCodeForCats, 2023-05-17:
The next pre-release will have this change if you are able to test it.

WillCodeForCats, 2023-05-17:
[Release v2.3.2-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.2-pre.1) is now available.

FranciscoBigWings, 2023-05-18:
> The next pre-release will have this change if you are able to test it.

How can I try it? How can I install the pre-release?

WillCodeForCats, 2023-05-18:
If you're using HACS you need to turn on beta versions.

FranciscoBigWings, 2023-05-19:
> If you're using HACS you need to turn on beta versions.

I installed it well, then I first tried that everything was stable without changing anything related to Storedge commands.

I enabled manually the entities in the leader inverter (not Storedge inverter) of:
- Storage Control Mode
- Storage Default Mode
- Storage Charge Limit
- Storage Discharge Limit

After enable one by one, all them are read and it is possible to write them

As I suspected, if you change any command in the follower inverter related to Storage control nothing happens but when you change it in the leader everything works well

Important notes:
- If you set a command in the leader inverter you must set the same command in the follower inverter related to "Storage Control Mode". For example if you set "Remote Control" in the leader inverter, you must set the same command in the follower inverter otherwise it doesnt work.
- If you set the command "Storage Default Mode" in the leader inverter, you dont need to change it in the follower inverter. The leader inverter changes it automatically.
- Storage Charge Limit and Storage Discharge Limit entities in leader inverter doesnt change anything. You must change those values in the Storedge follower inverter. Those entities could be disabled in the leader

Summarize:
- Entities Storage Control Mode and Storage Default Mode are needed in the leader
- Storage Control Mode entity value should be changed at the same time in both leader and follower. I will create an HA automation to change both at the same time if I change one
- Storage Default Mode is not needed in the follower, the leader shares it value to the follower. This should only be changed in the leader.
- Charge and Discharge Limits are not needed in the leader, those values should be changed in the follower

I am open to do any further tests. Thank you for your work @WillCodeForCats 

For the moment I will keep in v2.3.2-pre.1 and report any issues

WillCodeForCats, 2023-05-22:
PR #321 removes charge and discharge limit entities if the inverter doesn't have a battery attached.

FranciscoBigWings, 2023-05-23:
Hi.

v2.3.2-pre.1 had been working already for 4 days without any issues 

Today I installed v2.3.2-pre.2 and everything seems stable too

WillCodeForCats, 2023-06-05:
[Relase v2.3.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.2) is now available.

closed 2023-06-05 by WillCodeForCats