WillCodeForCats/solaredge-modbus-multi #317: integration not ready yet; Retrying in background

stepir, 2023-05-12:
**Describe the bug**
When I setup integration, even after adding batteries discovery - integration load fine and I have all my entities. Upon restart the integration remain in the following state. If I remove and readd integration it work until service restart.

[image]

**Expected behavior**
Note that I have 4 inverters and 3 batteries. Maybe 30 seconds are not enough for fetching all the data? is there a (user) configurable way to increase it?

**Logs**
**Debug Logs**

```
2023-05-13 00:42:15.145 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 30.000 seconds (success: False)
2023-05-13 00:42:15.145 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet; Retrying in background
```

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: **2023.4.5**
 - solaredge-modbus-multi Version: **v2.3.0**

**Additional context**
Add any other context about the problem here.

WillCodeForCats, 2023-05-12:
turn on debug

WillCodeForCats, 2023-05-12:
Two log lines doesn't provide any useful information, the full cycle of debug logging needs to be uploaded (do not edit any logs).

stepir, 2023-05-13:
I did not saw anything else relevant but here it is:

[home-assistant_2023-05-13T00-03-49.851Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/11468388/home-assistant_2023-05-13T00-03-49.851Z.log)

Thanks

WillCodeForCats, 2023-05-13:
Well, it's relevant to me figuring out what calls are happening when and what the timing is for those calls.

I'm already actually working on a dynamic timing property in PR #316 to eliminate the fixed 30 second timeout value.

stepir, 2023-05-13:
sure thing - 5s per inverter would be 20s in my case, which might be not enough. I noticed that after xx minutes the integration actually loads, lead me thinking that - at least in my config - 30 seconds is not far away from what I would need in the worst case scenario

WillCodeForCats, 2023-05-13:
I see some issues where there are two modbus reads in a row to the same inverter but there is a 3 second delay for the response to the second call. It's not a failure but the extra seconds adds up when you have many inverters and batteries.

I'm going to try and clean up some of these modbus reads.

stepir, 2023-05-13:
Sound good - let me know when you want me to test it

WillCodeForCats, 2023-05-13:
I changed updates for inverters and meters where a delay between two reads was seen that could be combined into one read, so that should take care of a few delays I saw in your logs (about 10 seconds worth).

I'm testing it on my dev system now and will do a pre-release after that.

WillCodeForCats, 2023-05-13:
Please try [Release v2.3.1-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.1-pre.1)

stepir, 2023-05-13:
upgraded and restarted HA twice - both times the integration started right away 👍

closed 2023-05-13 by WillCodeForCats