WillCodeForCats/solaredge-modbus-multi #322: Solaredge battery status

GregEdge, 2023-05-24:
I've noticed a battery status not listed in the integration.
Status 10. Power Saving mode

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.5.4
 - solaredge-modbus-multi Version: v2.3.1
[image]

WillCodeForCats, 2023-05-24:
It doesn't looks like you're actually running release v2.3.1.

In 2.3.0 the status entities were changed to display a value from an ENUM list of values, not a numeric value. Your sensor still has a numeric value for status, which is an indication of an older version.

The constant for power saving mode was added in PR #178 which was in [Release v2.2.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7)

GregEdge, 2023-05-24:
Understand what you are saying.  The status 10 was from a previous release
are there 10 in the current release?
I did an enumerate and there was a count of 9.

WillCodeForCats, 2023-05-24:
Numeric value of 10 was added but there are not 10 known states.

GregEdge, 2023-05-24:
Ah, ok.
Thanks for the clarification

WillCodeForCats, 2023-05-24:
The specs only define 7 total battery states. The extra ones were added from observation comparing modbus values to app or web displays.

In release 2.3.0 and higher it will display as `B_STATUS_POWER_SAVING`.

closed 2023-05-25 by WillCodeForCats