WillCodeForCats/solaredge-modbus-multi #329: Move the "storage Controle Mode" configuration item below the "Limit Control Mode"

jovana, 2023-06-03:
**Describe the feature**
In the configuration for the Battery setup, the Storage Control Mode selector is in the wrong place. Because by default it is disabled. Once enabled the items above are enabled also. So for the user experience it's a bit confusing.

**Screenshots**
In the below screenshot:
- Place the selector in the red box more above. 
- The items in the green boxs are by default disabled. After change the setting (red box) the items in green boxs are also enabled.
[image]

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.4.4 - 20230411.1
 - solaredge-modbus-multi Version: 2.3.1

**Additional context**
This integration is the best for my SolarEdge equipment setup :-)

jovana, 2023-06-03:
> **Describe the feature** In the configuration for the Battery setup, the Storage Control Mode selector is in the wrong place. Because by default it is disabled. Once enabled the items above are [...]

One more thing, it looks like the "storage command mode" and the "storage default mode" are doing the same thing here. Check the blow gif, when changing the "storage command mode" it also automatically changes the "storage command mode".
[image]

WillCodeForCats, 2023-06-03:
Home Assistant frontend sets the order, not the integration. Make your own dashboard with the layout you prefer.

WillCodeForCats, 2023-06-03:
We send commands to the inverter and read the results back. If the inverter is changing other settings you'll need to contact solaredge support or your installer. Or maybe someone who has experience with batteries can help in discussions, I don't have batteries so I can't really help with battery questions.

closed 2023-06-03 by WillCodeForCats

jovana, 2023-06-03:
> Home Assistant frontend sets the order, not the integration. Make your own dashboard with the layout you prefer.

Aha okay understood! 
I will close this then.

jovana, 2023-06-03:
> We send commands to the inverter and read the results back. If the inverter is changing other settings you'll need to contact solaredge support or your installer. Or maybe someone who has experience [...]

Okay, I will keep an eye on this. It seems this settings are the same under the hood. So let me monitor for a while to see what it does. 

Thx for the reply!