WillCodeForCats/solaredge-modbus-multi #333: Add port to unique_id of the device

robbinjanssen, 2023-06-12:
**Describe the current behavior**
At this moment in time, the `unique_id` used by the configured device is the `host`. For example `1.2.3.4`.

**What should be updated or changed?**
 In our case we have a modbus proxy which we use to expose multiple inverters on the same ip with different ports, for example: `1.2.3.4:1502` and `1.2.3.4:1503`. The current implementation doesn't allow us to add the host with a different port because it is "already configured"
 
 Adding the port to the unique ID would resolve this problem.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.6.0
 - solaredge-modbus-multi Version: v2.3.2

WillCodeForCats, 2023-06-12:
This integration doesn't need a proxy to help split multiple inverters, it can address them all directly behind a single IP.

robbinjanssen, 2023-06-13:
They are on the same IP, but not on the same port. The config flow doesn't allow me to add the same IP with a different port. Or am I doing something wrong? 🤔

WillCodeForCats, 2023-06-13:
My advice would be don't use a proxy. Why do you think you need one?

WillCodeForCats, 2023-06-13:
Do you have a link for this proxy I can look at documentation or something? The entire point of a modbus bus is to address multiple devices over a single connection, so either you can change the config do that or it doesn't really support modbus properly.

WillCodeForCats, 2023-06-15:
If I can learn more about the proxy I'll take a look at that aspect to see if there's a solution that involves the proxy addressing modbus devices correctly instead of splitting them by port, otherwise I don't think it's wise to allow duplicate IP addresses to be configured since that can lead to someone pointing config at the same physical inverter multiple times with different port numbers.

You could work around that by assigning multiple IP addresses to the proxy instead of splitting by port.

robbinjanssen, 2023-06-16:
Let me clarify our situation a bit more. Currently we have 3 inverters.
- The first two are connected with each other through RS485, they are accessible on `1.2.3.4:1502`
- The last inverter is accessible on `1.2.3.4:1503`

We use a proxy because we have systems, other than home assistant, that need the data (primarily for "smart charging" our carpark).

We use this proxy: https://github.com/tiagocoutinho/modbus-proxy

However we've decided to use another IP for the "last inverter" so I think we can close this issue :)

jovana, 2023-06-16:
@robbinjanssen probably I have a workaround for you. I don't have tested and not sure if it could work. But it something you can try safely.

**But here it comes:**
1. Add the inverter by entering a wrong ip, like 192.168.100.2.
2. You now see it will try to set up (red bar because it can not connect). 
3. Shutdown HA. 
4. Go to your config folder and open /.storage/core.config_entries (First make a backup from this file before you make any changes)
5. Find your device (it should be on the bottom of the file) and change the wrong hostname to the correct hostname:1234. (Again make sure you have a copy of this file before making any changes). 
6. Start HA and see what happens. Worst case it keeps the red bar and does work, so you can delete the device the normal way using HA gui. 

Not sure, but this is how I want to give it a try.

Jo

WillCodeForCats, 2023-06-16:
Yes, you can manually change the config entries file in storage. The check is only done once when you add the integration in HA.

Knowing more about the situation allows me to come up with ideas. So while I don't think it's a good idea to just change it to allow duplicate hosts, it would be possible to add a step to confirm the duplicate during setup.

By design this integration can coexist with other instances of itself on different systems, or other things that close the connection immediately after reading, since it doesn't hold the connection open by default (unless the keep open option is enabled or the poll time is less than 10 seconds).

closed 2023-07-08 by WillCodeForCats