WillCodeForCats/solaredge-modbus-multi #337: Cannot import name 'host_valid' from 'custom_components.solaredge_modbus_multi.helpers'

syphernl, 2023-06-15:
**Describe the bug**
After updating to v2.3.3 this shows up in the log.

**Expected behavior**
No error

**Screenshots**
N/A

**Logs**
```
023-06-15 07:23:24.782 ERROR (MainThread) [homeassistant.helpers.integration_platform] Unexpected error importing solaredge_modbus_multi/repairs.py
Traceback (most recent call last):
  File "/srv/hass/lib/python3.11/site-packages/homeassistant/helpers/integration_platform.py", line 50, in _async_process_single_integration_platform_component
    platform = integration.get_platform(platform_name)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/hass/lib/python3.11/site-packages/homeassistant/loader.py", line 833, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/hass/lib/python3.11/site-packages/homeassistant/loader.py", line 850, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/hass/.homeassistant/custom_components/solaredge_modbus_multi/repairs.py", line 15, in <module>
    from .helpers import host_valid
ImportError: cannot import name 'host_valid' from 'custom_components.solaredge_modbus_multi.helpers' (/home/hass/.homeassistant/custom_components/solaredge_modbus_multi/helpers.py)
```

**Debug Logs**
N/A

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.6.2
 - solaredge-modbus-multi Version: 2.3.3

**Additional context**

WillCodeForCats, 2023-06-15:
That should be easy to duplicate, but I actually can't find the problem. It loads fine on my live and test systems so I'm not sure how that's happening.

jovana, 2023-06-15:
> is anyone else having this problem?

Still running 2.3.1. I will update to the latest version and will let you know.

Largelos, 2023-06-18:
> is anyone else having this problem?
No, unable to reproduce also

WillCodeForCats, 2023-06-20:
Unless there are other reports of this I will have to close this issue as unable to reproduce.

Beyond that I would suggest reinstalling or using HACS to manage installs.

syphernl, 2023-06-20:
> Beyond that I would suggest reinstalling or using HACS to manage installs.

This integration has been installed using HACS.

WillCodeForCats, 2023-06-20:
I only develop (and run my person system) with Home Assistant Operating System, so if it's being run in a different environment I probably won't be able to reproduce the issue.

WillCodeForCats, 2023-07-17:
I've not seen this on any of my systems in the past month of work and testing I've been doing, so I am going to have to close this as unable to reproduce.

closed 2023-07-17 by WillCodeForCats