WillCodeForCats/solaredge-modbus-multi #342: Power consumption negative

cburghardt, 2023-06-20:
**Describe the bug**
Especially in the morning, when the second inverter is not active yet, I get negative power consumption.

**Expected behavior**
Power consumption should always be positive

**Screenshots**
Grafana dashboard
[image]

**Logs**
Not related but anyway:
```
Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:229
Integration: SolarEdge Modbus Multi Device (documentation, issues)
First occurred: 19. Juni 2023 um 23:11:56 (1 occurrences)
Last logged: 19. Juni 2023 um 23:11:56

Error fetching SolarEdge Coordinator data: Could not open Modbus/TCP connection to 192.168.178.64
```

**Debug Logs**

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.6.2
 - solaredge-modbus-multi Version: 2.3.3

**Additional context**
I have 2 inverters so I combine the AC value like that:
`      - name: "Power - AC Total" 
        unique_id: "power_ac_total"
        unit_of_measurement: "W"                                                
        device_class: "power"                                                   
        state_class: "measurement" 
        icon: "mdi:solar-power"
        state: "{{ float(states('sensor.solaredge_modbus_i1_ac_power'), 0) + float(states('sensor.solaredge_modbus_i2_ac_power'), 0) }}"          `

Power consumption:
`      - name: "Power - Consumption"                                             
        unit_of_measurement: "W"                                                
        device_class: "power"                                                   
        state_class: "measurement"                                              
        # The net of inverter and grid power                                    
        state: "{{ float(states('sensor.power_ac_total'), 0) - float(states('sensor.solaredge_modbus_m1_ac_power'), 0) }}"`

WillCodeForCats, 2023-06-20:
Template results are not useful for an issue report. I need to see the actual sensor values, not the calculated results.

Upload a diagnostic file when the problem is occurring. Not enough info about your system.

cburghardt, 2023-06-20:
[image]
Readings of both inverters. Looks like no. 1 is returning negative values during that time.

cburghardt, 2023-06-20:
[config_entry-solaredge_modbus_multi-b86bc24f1ca70258f7ffd41889955498.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/11796777/config_entry-solaredge_modbus_multi-b86bc24f1ca70258f7ffd41889955498.json.txt)

Diagnostics

WillCodeForCats, 2023-06-20:
It appears you have a battery: negative inverter power AC/DC values are valid in certain battery charging conditions.

Export+Import meters will return positive power values during export and negative values during import.

cburghardt, 2023-06-20:
Thanks, so how do I get the correct AC value (PV generation) and house consumption (power consumption) then?

WillCodeForCats, 2023-06-20:
I can't help with that: I don't have a system with batteries personally.

WillCodeForCats, 2023-06-20:
You might try searching for or asking in discussions, or look at the templates that have been contributed in the wiki to see if they apply to your situation.

WillCodeForCats, 2023-06-20:
This issue was opened as power should always be positive, but that's not the case: power can be positive or negative depending on what it's indicating. So as far as an issue report this is resolved.

Template stuff I would defer to discussions where someone who has experience with calculations related to batteries might be able to help. Again, I don't have batteries, so I can't really help with anything battery related outside the integration.

This discussion has a lot of battery related template stuff: https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084

closed 2023-06-20 by WillCodeForCats

mpredfearn, 2023-06-20:
When `sensor.solaredge_i1_ac_power` is negative - i.e. power is flowing into the inverter (to the battery) in a single inverter system that power *must* be coming from the grid - so `sensor.solaredge_m1_ac_power` will be positive. And the net of these 2 cancel each other out and you are left with the load consumption.

In your case, with 2 inverters, the power that is flowing into inverter 1 battery could be (a looks like it is in your graph) coming from inverter 2. The net of inverter 1 AC power, inverter 2 AC power and the grid meter should all still work out to be the load consumption though. What you're likely seeing though is spikes due to different sampling frequency / times to the 2 inverters. This happens, to a lesser extent, with single inverter / single meter systems where the meter and inverter registers are sampled milliseconds apart - so you do not get an atomic snapshot of the system state. So, you're best bet is probably to filter / smooth the raw data from each inverter before using it in the calculations.

WillCodeForCats, 2023-06-20:
I'm not entirely sure what to make of the first part of the graph where the two inverters are showing opposite DC power: maybe the power from I1 is being used to charge the battery on I2? According to the diagnostic file there's only one battery and it's attached to I2.