WillCodeForCats/solaredge-modbus-multi #353: No "Configure" button for the integration

bjruberg, 2023-07-01:
**Describe the bug**
I would like to configure the integration for advanced power options - but the "configure" button for the integration is not showing up for me. 

**Expected behavior**
I would expect a "configure" button on the screenshot as this integration obviously supports this. 

**Screenshots**

[image]

SolarEdge Modbus Multi: v2.3.3
Home Assistant 2023.6.3
(have never seen that button in previous versions)

WillCodeForCats, 2023-07-01:
I have no idea why the HA frontend wouldn't show the button. Configuration options have been supported since the very first v2.0.0 release over a year ago.

It sounds like you might have an HA frontend problem but I wouldn't know how to debug that since I don't know anything about frontend development.

bjruberg, 2023-07-01:
I do not think it is a frontend problem - I am a frontend developer and from what I can see, the DOM elements are not added to the page.
Something on the server-side decides to not show this button.

WillCodeForCats, 2023-07-01:
It's always worked for me and has been a basic function of every release, so I'm not sure why it wouldn't show up.

[image]

WillCodeForCats, 2023-07-01:
I've checked several examples for options flow in HA core just to make sure, and about the only improvement I can see would be to add a type hit for the callback. I don't think it changes anything but I can push these into a pre-release:

[image]

bjruberg, 2023-07-01:
Well, I would test it when I can fetch it :)

bjruberg, 2023-07-01:
I am sorry - it seems that I was actually using the "[other solaredge modbus](https://github.com/binsentsu/home-assistant-solaredge-modbus)"  implementation. Although I could swear I was directed to the issue page here from it - whatever.
Works now. And finally I have site power controls. Well done!

closed 2023-07-01 by bjruberg

Habbie, 2023-08-01:
Just putting notes here for anybody else who runs into this.

I found this on my system:

```
homeassistant:/config/custom_components# head -50 *modbus*/manifest.json
==> solaredge_modbus/manifest.json <==
{
  "domain": "solaredge_modbus",
  "name": "SolarEdge Modbus Multi-Inverter",
  "documentation": "https://github.com/WillCodeForCats/solaredge-modbus-multi",
  "issue_tracker": "https://github.com/WillCodeForCats/solaredge-modbus-multi/issues",
  "requirements": ["pymodbus==2.5.3"],
  "dependencies": [],
  "codeowners": ["@WillCodeForCats"],
  "config_flow": true,
  "iot_class": "local_polling",
  "version": "1.1.3"
}

==> solaredge_modbus_multi/manifest.json <==
{
  "domain": "solaredge_modbus_multi",
  "name": "SolarEdge Modbus Multi",
  "codeowners": ["@WillCodeForCats"],
  "config_flow": true,
  "dependencies": [],
  "documentation": "https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki",
  "integration_type": "hub",
  "iot_class": "local_polling",
  "issue_tracker": "https://github.com/WillCodeForCats/solaredge-modbus-multi/issues",
  "loggers": ["custom_components.solaredge_modbus_multi"],
  "requirements": ["pymodbus>=3.1.1"],
  "version": "2.3.4"
}
```

Looking in git history, indeed version 1.1.3 called itself `solaredge_modbus`, so this looks like an upgrade across the name change is perhaps keeping both around?

WillCodeForCats, 2023-08-01:
All 1.x releases were removed a long time ago, but we changed the domain since 2.0.0 which required manually removing the old directory in custom_components if the upgrade steps weren't followed (remove first then upgrade). The steps were documented in the first v2.0.0 release notes.

Habbie, 2023-08-01:
Thank you for the quick response! Indeed, clearly I (and the user above) skipped those steps.

WillCodeForCats, 2023-08-01:
There was unfortunately no way to automate that step.

Habbie, 2023-08-01:
I have now followed the v2.0.0 release notes and everything works. Thanks!