WillCodeForCats/solaredge-modbus-multi #356: Connection refused

s50u, 2023-07-05:
**Describe the bug**
After a power outage modbus connection is refused. Does anyone have any tips on how to fix the problem? MySolarEdge app is running fine. 3-phase SE9K-RW0TEBEN4 , DSP2 Version | 2.20.1103, Communication Board (CPU) Version | 4.17.221, DSP1 Version | 1.20.1517. 

**Logs**
2023-07-04 20:51:47.264 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solaredge_modbus_multi which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-04 20:51:47.265 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-04 20:51:58.476 ERROR (SyncWorker_6) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:51:58.488 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: Could not open Modbus/TCP connection to 192.168.2.222; Retrying in background
2023-07-04 20:52:07.087 ERROR (SyncWorker_6) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:52:17.612 ERROR (SyncWorker_4) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:52:38.047 ERROR (SyncWorker_1) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:53:18.290 ERROR (SyncWorker_0) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:54:38.569 ERROR (SyncWorker_4) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:55:58.994 ERROR (SyncWorker_2) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:57:19.202 ERROR (SyncWorker_4) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:58:39.495 ERROR (SyncWorker_2) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 20:59:59.960 ERROR (SyncWorker_4) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:01:20.064 ERROR (SyncWorker_1) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:02:40.281 ERROR (SyncWorker_4) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:04:00.415 ERROR (SyncWorker_2) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:05:20.778 ERROR (SyncWorker_2) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:06:40.953 ERROR (SyncWorker_1) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:08:01.084 ERROR (SyncWorker_2) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused
2023-07-04 21:09:21.142 ERROR (SyncWorker_0) [pymodbus.logging] Connection to (192.168.2.222, 1502) failed: [Errno 111] Connection refused

**Home Assistant
 - Home Assistant 2023.6.3, Supervisor 2023.06.4, Operating System 10.3
 - solaredge-modbus-multi Version: v2.3.4
 - Installation Type: NUC Generic x86-64

jovana, 2023-07-05:
You may check the IP address of the inverter. Probably it has a new/other IP address? Also, check if the Modbus is still enabled and check the port in the inverter.

WillCodeForCats, 2023-07-05:
This is absolutely not an issue with the integration.

s50u, 2023-07-05:
Thanks i checked this earlier. My fixed local ip is still correctly assigned and the default port is turned on on the inverter. If I find a problem I report it here.

closed 2023-07-06 by WillCodeForCats