WillCodeForCats/solaredge-modbus-multi #361: Update cycles to reset battery doesn't work as expected

Largelos, 2023-07-14:
**Describe the bug**
Sometimes battery energy counters reset before update count reaches the "Update cycle to resets" value. 
In debug log, only 5 pool cycle with "Battery Export Energy went backwards: 0 < 242478" exist, while in integration settings threshold set to "40".
In past, i can see up to 11 cycles without reset in logs, so it looks like this floating bug and possible somehow related to ModBus connection error just before that.

**Expected behavior**
Energy counters should be reset only after reach "update cycle to resets" threshold.

**Screenshots**
[image]

**Diagnostic File**
[config_entry-solaredge_modbus_multi-dd1018dc8fc1c298fc03bc4e7311a826.json(2).txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12053945/config_entry-solaredge_modbus_multi-dd1018dc8fc1c298fc03bc4e7311a826.json.2.txt)

**Logs**
```
2023-07-14 09:44:08.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 16.244 seconds (success: True)
2023-07-14 09:44:08.720 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Export Energy went backwards: 0 < 242478
2023-07-14 09:44:08.720 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 0 < 246965
2023-07-14 09:44:08.722 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Utility B4 Energy Export received an invalid new state from sensor.solaredge_b1_energy_export_3 : unknown
2023-07-14 09:44:08.722 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Utility B4 Energy Import received an invalid new state from sensor.solaredge_b1_energy_import_3 : unknown
2023-07-14 09:44:23.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
...
2023-07-14 09:44:27.662 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 4.192 seconds (success: True)
2023-07-14 09:44:27.666 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Export Energy went backwards: 0 < 242478
2023-07-14 09:44:27.666 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 0 < 246965
2023-07-14 09:44:42.471 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
...
2023-07-14 09:44:47.321 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 4.850 seconds (success: True)
2023-07-14 09:44:47.325 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Export Energy went backwards: 0 < 242478
2023-07-14 09:44:47.325 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 0 < 246965
2023-07-14 09:45:02.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
...
2023-07-14 09:45:05.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.503 seconds (success: True)
2023-07-14 09:45:05.976 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Export Energy went backwards: 0 < 242478
2023-07-14 09:45:05.976 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 0 < 246965
2023-07-14 09:45:20.470 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
...
2023-07-14 09:45:24.170 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.700 seconds (success: True)
2023-07-14 09:45:24.174 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Export Energy went backwards: 0 < 242478
2023-07-14 09:45:24.174 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 0 < 246965
2023-07-14 09:45:39.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
...
2023-07-14 09:45:42.946 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.476 seconds (success: True)
2023-07-14 09:45:57.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
```

**Debug Logs**
[home-assistant_solaredge_modbus_multi_2023-07-14T19-17-56.636Z_trimmed.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12053889/home-assistant_solaredge_modbus_multi_2023-07-14T19-17-56.636Z_trimmed.log)

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.7.1
 - solaredge-modbus-multi Version: v2.3.5-pre.1
 - Installation Type: HACS

WillCodeForCats, 2023-07-14:
v2.3.5-pre.1 will not be developed; change to v2.4.0-pre.1

Largelos, 2023-07-14:
Do you think problem would not reproduced with 2.4.0-pre.1 version? I don't see any changes for this feature in release notes and commits.

WillCodeForCats, 2023-07-14:
I can add cycle count to the debug log, otherwise I'm not sure why it wouldn't go the full count unless the integration was reloaded or HA restarted. Cycle count can't persist across an integration reload or restart though. Also remember that changing options for an integration causes HA to reload that integration, which will clear the internal count as well and start it from values read at integration startup.

Any work I do now will be based on the 2.4.x releases, there will not be a v2.3.5 release and no more fixes or updates to any 2.3.x releases.

Largelos, 2023-07-14:
> otherwise I'm not sure why it wouldn't go the full count unless the integration was reloaded or HA restarted. Cycle count can't persist across an integration reload or restart though.
HA or integration was not restarted, and settings wasn't changed also, but there is errors with connection refused probably caused by internal solaredge inverter bugs and multiple tracebacks exists in debug recently before that.

WillCodeForCats, 2023-07-14:
Connection error should not be able to reset the cycle counter, which is internal to the entity.

At this point the only thing I can do is add more debug logging so I'll do that under a pre-release.

Largelos, 2023-07-14:
Thx, i'll keep an eye on next updates and left debug always on in case this problem repeat again.

WillCodeForCats, 2023-07-14:
I'm trying it in my modbus simulator first, if there's something obvious to fix.

WillCodeForCats, 2023-07-14:
[Release v2.4.0-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.0-pre.2) contains the additional debug logging for battery energy reset cycles.

WillCodeForCats, 2023-07-14:
I was not able to duplicate any issues with my simulator and the cycle counter was maintained across inverter communication failures.

Largelos, 2023-07-15:
Hello, I checked code and did not found any reset for variable `self._count` when modbus counter report normal value and doesn't went backwards. Look like variable resets to zero only at integration init and always increasing until reach reset cycle limit? I think this is the reason, the counter has a cumulative effect from the previous wrong values reported by the inverter since integration/ha restart.

WillCodeForCats, 2023-07-15:
Sure, I can make it set the count to 0 on normal value.

WillCodeForCats, 2023-07-15:
Fixed in [Release v2.4.0-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.0-pre.3)

closed 2023-07-15 by WillCodeForCats