WillCodeForCats/solaredge-modbus-multi #367: Please try the v2.4.0 pre-releases

WillCodeForCats, 2023-07-17:
I ended up doing an unplanned major internal architectural change while trying to figure out some other behavior I wasn't completely satisfied with in testing, so the v2.4.0 release was created.

Since it's a big internal change - but nothing external except adding a new grid control switch - please test it to make sure it works the same as it did before under 2.3.x, but hopefully better in some cases with command writes.

I've been using it on my system without any issues, but I don't have batteries and I my inverters don't accept commands over modbus, so outside of my simulator those aspects are still untested against a real inverter.

It depends on using a current version of pymodbus from HA 2023.7.

thargy, 2023-07-19:
I am using it, but it is a new install, so probably not as informative.  However, it is currently working fine.

WillCodeForCats, 2023-07-19:
Reports that it works without errors are good too.

mpredfearn, 2023-07-21:
I've been using on my system for the last few days and everything seems to be working fine.

WillCodeForCats, 2023-07-21:
I will probably promote the latest 2.4.0 pre-release to full release after a week or two of bake in testing (assuming no major issues are reported).

purcell-lab, 2023-07-26:
Working fine here.  2x three phase inverters, leader/ follower, consumption meter, no DC coupled battery.

My zero export scripts worked without changes.

[image]

Unsure about what the new 'Grid Control' and 'External Production' switches activate.

[image]

```
MMPPT Events entity is no longer being provided by the solaredge_modbus_multi integration.
```

I wasn't using this entity so its disappearance doesn't worry me.

WillCodeForCats, 2023-07-31:
External production is probably not going to be used often, maybe never in a home setting, but it's a setting that is available so the setting exists. It's for indicating you have production coming from a source outside of the system. They are disabled by default for that reason, but they aren't new to this version.

Same for grid control: it's mentioned so it's being exposed. Although the docs say using that switch is required for some settings, those settings still seem to work without it so how accurate the documentation really is when referring to it is questionable. Grid control switch is new to this version.

WillCodeForCats, 2023-07-31:
Add some changes to timeout handling based on discussion #373 and a new feature (button entity) to request immediate polling to update the data.

WillCodeForCats, 2023-07-31:
Oops my mistake, grid control switch is enabled by default.

WillCodeForCats, 2023-08-02:
Time for a release, I think it's ready.

closed 2023-08-02 by WillCodeForCats