WillCodeForCats/solaredge-modbus-multi #368: Support for inverters with no consecutive IDs

thargy, 2023-07-19:
**Describe the feature**
Support for non-consecutive IDs for inverters

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.7.2
 - solaredge-modbus-multi Version: 2.4.0-pre.4

**Additional context**
This feature was original requested in #47, which is now closed, as the poster moved their second inverter from 3->2.

_However_, I installed the new (for UK) HomeHub last week.  The installer had problems getting it working, until he called SolarEdge and was reminded that ID's 1 & 2 are reserved for masters and slaves have to start from ID 3.  Changed the slave inverter to 3 resolved the issue.  My installer recalled he'd had the same issue previously.  As such, I'm reluctant to attempt changing the ID back to 2 (even if that was possible and I knew how), and I know my installer won't.  It seems easier to provide support for custom IDs.

thargy, 2023-07-19:
I decided to try and change the modbus ID of the slave inverter from 3 back to 2 and everything seems to be working fine.  So no idea why the SolarEdge support and installer both insisted it should be 3.

closed 2023-07-19 by thargy

WillCodeForCats, 2023-07-19:
Modbus ID range is 1 to 247, there are no "reserved" ones.

thargy, 2023-07-19:
> Modbus ID range is 1 to 247, there are no "reserved" ones.

I was with the installer whilst he spoke to support, and he definitely said about the 'reservation' which is bizarre because it's clearly not true!  The weird thing is, the lead inverter couldn't detect the follower until it's ID was changed to 3.  I wonder if there's an issue with the SetApp's follower inverter detection routine, which you run as part of the initial commissioning?  Regardless, when I changed it back from 3 to 2, it carried on working.

WillCodeForCats, 2023-07-19:
Usually the reason people say you can't use ID 2 is because they have a meter that's on modbus ID 2 (they ship factory set for 2). Or in the case of installers they know meters come with ID 2 so they skip to 3 even if a meter isn't installed since they might come back later to add a meter when the customer wonders why they don't get consumption or E+I data. But actually the inverter leader/follower chain and the bus for meters are separate, where duplicate IDs only apply to devices on the same bus. But they're not really on the same bus so having a meter with ID 2 and a follower with ID 2 is not actually a conflict. The inverter interprets data it has from meters and batteries to modbus/tcp like a proxy.

I've never heard or read anyone saying IDs 1 and 2 are reserved for leaders, so I'm not sure where that came from. There are no reserved use purposes for any specific ID: you can have a leader on ID 45 and followers on 46, 47, etc. if you really liked those numbers instead of 1,2,3.