WillCodeForCats/solaredge-modbus-multi #369: Wiki Documentation detailing Energy Dashboard setup

thargy, 2023-07-19:
**Describe the feature**
Add documentation to the wiki on setting up the Energy Dashboard in HA.

**Screenshots**
[image]

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.07.1
 - solaredge-modbus-multi Version: 2.3.4
 
**Additional context**
The readme advertises support for the energy dashboard, but it's not immediately obvious which sensors to add where.  For example, I'm at a loss as to which to use for the solar panels?

WillCodeForCats, 2023-07-19:
There is only one solar production sensor on the inverters: `sensor.solaredge_i*_ac_energy_kwh`

If you have a production meter that would provide another option.

WillCodeForCats, 2023-07-19:
These are on my system:

[image]

[image]

WillCodeForCats, 2023-07-19:
Yeah I don't have any experience with how that entity interacts with battery.

thargy, 2023-07-19:
> No worries at all. This is my configuration for a single inverter, single battery configuration if it helps anyone. [gist.github.com/Ashpork/f80fb0d3cb22356a12ed24734065061c](https://gist.github.com/Ashpork/f80fb0d3cb22356a12ed24734065061c)

Perfect! I have one battery per inverter, so I should be able to adapt.

WillCodeForCats, 2023-07-19:
> No worries at all. This is my configuration for a single inverter, single battery configuration if it helps anyone. https://gist.github.com/Ashpork/f80fb0d3cb22356a12ed24734065061c

Added to [wiki/YAML-Sample-Configurations](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/YAML-Sample-Configurations)

thargy, 2023-07-19:
> No worries at all. This is my configuration for a single inverter, single battery configuration if it helps anyone. [gist.github.com/Ashpork/f80fb0d3cb22356a12ed24734065061c](https://gist.github.com/Ashpork/f80fb0d3cb22356a12ed24734065061c)

As this is in the Wiki, might want to re-indent [Line 50](https://gist.github.com/Ashpork/f80fb0d3cb22356a12ed24734065061c#file-ha-power-sensors-yaml-L50).

requiemmg, 2023-07-20:
Thanks, I'll give it a shot. I just got my batteries a week ago, everything was easy before. I now switched from Solaredge Modbus to Modbus multi and redid my dashboards, but energy dashboard is giving me a headache.

I have this in my energy export/import from/to grid:
[image]

result is an empty energy usage graph, but the entities themselves show their expected values from the modbus and also the history of the entities is there.

Even more strange:
[image]

I have never excluded anything from the recorder and wonder what happened here...

thargy, 2023-07-20:
> I have never excluded anything from the recorder and wonder what happened here...

It needs at least 2 data points to generate statistics and can take up to 2 hours.

I have made an updated version of the above gist for 2 inverters/2 batteries, also fixed several issues and made it easier to deploy as a package:

https://gist.github.com/thargy/f9713cd6c06bc77a22a4b323ed0f83af

In current testing, it is working very well.  Once I have a few days of comparables, I may consider making the package configurable so that you can specify combinations of inverters/panels/batteries and have it create the summary statistics for you.  That is doable in YAML, though you wouldn't have the intermediate sensors.  In the meantime, if you are pretty familiar with YAML, it provides enough insight to be converted for any installation.

thargy, 2023-07-20:
I have made an updated version of the above gist for 2 inverters/2 batteries, also fixed several issues and made it easier to deploy as a package:

https://gist.github.com/thargy/f9713cd6c06bc77a22a4b323ed0f83af

In current testing, it is working very well.  Once I have a few days of comparables, I may consider making the package configurable so that you can specify combinations of inverters/panels/batteries and have it create the summary statistics for you.  That is doable in YAML, though you wouldn't have the intermediate sensors.  In the meantime, if you are pretty familiar with YAML, it provides enough insight to be converted for any installation.

requiemmg, 2023-07-20:
> > I have never excluded anything from the recorder and wonder what happened here...
> 
> It needs at least 2 data points to generate statistics and can take up to 2 hours.
> 
> Also, if you see the [gists](https://gist.github.com/thargy/f9713cd6c06bc77a22a4b323ed0f83af), you'll note that you can't just add the Inverters AC Energy to get the Solar Panel production. That is because it also includes any energy coming from the battery.
> 
> Instead, we create a new sensor which uses the inverter's DC power and removes the battery's contribution to calculate the remainder, which, presumably, is what is being generated by solar.

Thanks, this is what I was looking for.
I had it runing with the new config for >24 hours and the parameters are still not shown.
Nevermind, I think my database has some nasty leftovers and duplicate entries which might cause issues. I'll use this as an opportunity to start with a fresh HA installation and restore only my needed addons which are a bit more complex to configure from a partial backup. And I'll use an SSD instead of SD card....

thargy, 2023-07-20:
> Thanks, this is what I was looking for. I had it runing with the new config for >24 hours and the parameters are still not shown. Nevermind, I think my database has some nasty leftovers and [...]

Yeah, I dumped my Pi and changed is to be a second Zigbee2MQTT coordinator for my sensor network (which is so spammy it made the primary network unstable).  I replaced it with a relatively cheap [Imported N305 NUC](https://www.aliexpress.com/item/1005004009585758.html?spm=a2g0o.order_list.order_list_main.59.21ef1802qEUNpX) which has been rock solid ever since (and seriously overpowered).

However, you can do a lot of clean-up via judicious use of the "System-> Devices & Services -> Entities" and "Developer Tools-> Statistics" pages without throwing away all the juicy data, if you're motivated sufficiently!

requiemmg, 2023-07-20:
Thanks, I know, but somehow I have a situation where some entities are duplicate, some are not working properly, I think something got messed up in my database and I like to start over with a fresh installation. Just hope that all my backups are working properly.

requiemmg, 2023-07-24:
Hi again...I just did a fresh reinstall of Home Assistant and set up Solaredge Modbus Multi. As I have a single inverter/single battery system, I tried @Ashpork 's config file. I tried putting it in my packages dir as well as in the config.yaml, both give no errors and the sensors will show up but not the utility meters. Any idea why?
Second idea would be editing @thargy 's config to fit my needs, but I'm not sure if i am capable to do so...