WillCodeForCats/solaredge-modbus-multi #370: Values from Solaredge B1 missing

requiemmg, 2023-07-25:
Hi,
I have a single inverter single battery (SolarEdge HomeHub SE8KRWB48 with SolarEdge StorEdge 9,2kWh Battery) setup. Your addon was always getting correct data from modbus. Modbus data comes thru the Modbus-Proxy Addon; Clients are your addon and my OpenWB Wallbox.
I did a fresh install of Home Assistant this week and installed everything from scratch. Worked flawlessly. Today, I noticed that some values were suddenly missing from the B1 device: B1 DC Voltage, B1 Energy Export and B1 Energy import.
Everything else is there. First, I restarted Home assistant - Exergy export and Import reappeared, but not B1 DC voltage. I have no idea why and I was not even at home at the time it stopped reporting, so I sure didnt change anything about the config.

**Expected behavior**
Continuous reporting of the values to the entity.

**Screenshots**
[image]
[image]
[image]

**Logs**

There are no logfiles in /config/custom_components/solaredge_modbus_multi
Debug logs attached below, started debug logs and restarted HA.

This is a log from the Modbus-Proxy, 2 clients polling, no errors at all.
```
2023-07-25 15:01:38,424     INFO modbus-proxy.Client(192.168.0.170:34044): connection closed
2023-07-25 15:01:43,392     INFO modbus-proxy.Client(192.168.0.181:54529): new client connection
2023-07-25 15:01:43,634     INFO modbus-proxy.Client(192.168.0.181:54529): client closed connection
2023-07-25 15:01:43,634     INFO modbus-proxy.Client(192.168.0.181:54529): closing connection...
2023-07-25 15:01:43,635     INFO modbus-proxy.Client(192.168.0.181:54529): connection closed
2023-07-25 15:01:44,695     INFO modbus-proxy.Client(192.168.0.181:52551): new client connection
2023-07-25 15:01:44,926     INFO modbus-proxy.Client(192.168.0.181:52551): client closed connection
2023-07-25 15:01:44,926     INFO modbus-proxy.Client(192.168.0.181:52551): closing connection...
2023-07-25 15:01:44,927     INFO modbus-proxy.Client(192.168.0.181:52551): connection closed
2023-07-25 15:01:53,389     INFO modbus-proxy.Client(192.168.0.181:58663): new client connection
2023-07-25 15:01:53,693     INFO modbus-proxy.Client(192.168.0.181:58663): client closed connection
2023-07-25 15:01:53,694     INFO modbus-proxy.Client(192.168.0.181:58663): closing connection...
2023-07-25 15:01:53,696     INFO modbus-proxy.Client(192.168.0.181:58663): connection closed
2023-07-25 15:01:54,690     INFO modbus-proxy.Client(192.168.0.181:55255): new client connection
2023-07-25 15:01:54,991     INFO modbus-proxy.Client(192.168.0.181:55255): client closed connection
2023-07-25 15:01:54,992     INFO modbus-proxy.Client(192.168.0.181:55255): closing connection...
2023-07-25 15:01:54,993     INFO modbus-proxy.Client(192.168.0.181:55255): connection closed
2023-07-25 15:02:02,699     INFO modbus-proxy.Client(192.168.0.181:38789): new client connection
2023-07-25 15:02:02,973     INFO modbus-proxy.Client(192.168.0.181:38789): client closed connection
2023-07-25 15:02:02,973     INFO modbus-proxy.Client(192.168.0.181:38789): closing connection...
2023-07-25 15:02:02,975     INFO modbus-proxy.Client(192.168.0.181:38789): connection closed
2023-07-25 15:02:04,043     INFO modbus-proxy.Client(192.168.0.181:44031): new client connection
2023-07-25 15:02:04,290     INFO modbus-proxy.Client(192.168.0.181:44031): client closed connection
2023-07-25 15:02:04,291     INFO modbus-proxy.Client(192.168.0.181:44031): closing connection...
2023-07-25 15:02:04,294     INFO modbus-proxy.Client(192.168.0.181:44031): connection closed
2023-07-25 15:02:08,145     INFO modbus-proxy.Client(192.168.0.170:39042): new client connection
2023-07-25 15:02:08,339     INFO modbus-proxy.Client(192.168.0.170:39042): client closed connection
2023-07-25 15:02:08,339     INFO modbus-proxy.Client(192.168.0.170:39042): closing connection...
2023-07-25 15:02:08,340     INFO modbus-proxy.Client(192.168.0.170:39042): connection closed
```
**Diagnostic File**
[config_entry-solaredge_modbus_multi-ac9198dc514e43f1dc4670578282ecf2.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12161437/config_entry-solaredge_modbus_multi-ac9198dc514e43f1dc4670578282ecf2.json.txt)

**Debug Logs**
[home-assistant_solaredge_modbus_multi_2023-07-25T13-12-44.814Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12161542/home-assistant_solaredge_modbus_multi_2023-07-25T13-12-44.814Z.log)

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version:  2023.7.3
 - solaredge-modbus-multi Version: 2.3.4
 - Installation Type: HAOS

requiemmg, 2023-07-25:
Addendum: Tried deleting the config, uninstall from HACS, reboot, Install 2.4pre4, reboot, reconfigured: still the same.
And as I noticed, the value seems to appear and disappear periodically when I look at longer statistics (not too long as I said, fresh install of HA. Didnt see this before, though).
[image]

requiemmg, 2023-07-25:
Known issue of SolarEdges modbus integration that is? If so, there'd be nothing you could do about it,right?

requiemmg, 2023-07-25:
Right,this is a low voltage DC system.
To make sure it is not the proxy, i disabled it and connected directoy with your addon, but the result is the same, so it seems to be incorrect value from modbus (or incorrect interpretation of the data ar least with this model?)

WillCodeForCats, 2023-07-25:
The value coming from the inverter is `0x444d5296` which is a floating point number, converted to decimal is `821.290405273`

The only thing I can do is the upper limit to 1000 since that's what the modbus data coming from the inverter is.

requiemmg, 2023-07-25:
I will ask SE support about this. Strange.

WillCodeForCats, 2023-07-25:
PR  #371 will change the display max to 1000 volts and I will make a new pre-release with it.

requiemmg, 2023-07-25:
Thanks for the support. I just have to keep in mind that the values are BS.
Any idea why in first instance values for charged and duscharged kWh were probably missing? Never seen before. Sorry, no logs from that.

WillCodeForCats, 2023-07-25:
The kWh counters resetting or going backwards is a known issue with batteries: [wiki/Known-Issues#battery-energy](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues#battery-energy)

There is a configuration option to allow battery energy to reset and an option to set how many polling cycles for it.

WillCodeForCats, 2023-07-25:
Release v2.4.0-pre.5 contains this change.

closed 2023-07-25 by WillCodeForCats