WillCodeForCats/solaredge-modbus-multi #385: After Update 06.08.2023: Detects 2nd battery where there ist none

Ojoachim71, 2023-08-06:
**Describe the bug
Hello! Sine today, a second battery device shows up (B2). All values of this are unknown or empty. B1 works AS expected.

**Expected behavior**
It should only detect one battery, like in the release before, where there is only one.

Nothing else was changed, plugin accesses data from evcc modbus proxy.

Not possible top deleted 2nd device.

Regards,
Oliver

masterx1981, 2023-08-06:
I have freshly installed the integration, and the first time has detected 1 inverter and 2 meters and no battery, after a reinstall got 1 inverter, 1 meter but still no battery... maybe something wrong with detection

WillCodeForCats, 2023-08-06:
Please resubmit with all of the requested information in the template: there is no information here for me to work with.

closed 2023-08-06 by WillCodeForCats

deqw, 2023-08-07:
> Please resubmit with all of the requested information in the template: there is no information here for me to work with.

Hi Seth @WillCodeForCats. I think there isn't much more to add. After the v2.4.0 update a 2nd battery device is added to the configuration while that battery does not exist. This B2 device cannot be deleted from the integration and all information/sensors with this device is unknown.

Ojoachim71, 2023-08-07:
👍

WillCodeForCats, 2023-08-07:
There is an entire bug report template that was ignored with a big warning at the top not to delete it. I ask for all that information for a reason. I cannot help without it. I hate wasting my time asking for basic information the template asks for before starting. Opening an issue and deleting the template because why? Why would anyone open a ticket and think I don't think I need all the information I ask for?

If someone wants to actually help me go open a new bug report and see what the template asks for:

https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/new/choose

deqw, 2023-08-07:
Yep saw the template, I had already started there this morning and halfway the template I'd quit (and found later this already closed topic) because all is asked isn't necessary to point you to a minor problem that was introduced with v2.4.0. But we'll now patiently have to wait for someone to do the entire template.

WillCodeForCats, 2023-08-07:
I don't have access to batteries, I have no way to see what could even possibly be the problem.

deqw, 2023-08-07:
I understand Seth, that is not very convenient for development. I have no development environment setup myself so I can't help you with that, but if you can think of any other information I can provide you with, beside the template of course ;o) I'll be glad to help. The B2 isn't in the way otherwise so it isn't a big problem.

WillCodeForCats, 2023-08-12:
[Release v2.4.2-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.2-pre.2)