WillCodeForCats/solaredge-modbus-multi #388: Communication

CSH12, 2023-08-08:
**Describe the bug**
After a restart of HA the communication to modbus fails. 

**Expected behavior**
Communication established also after restart

**Screenshots**
[image]

**Debug Logs**
2023-08-08 10:02:05.166 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=True, keep_modbus_open=True, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, battery_rating_adjust=0, 
2023-08-08 10:02:05.166 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-08 10:02:05.174 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.2.113', 1502)
2023-08-08 10:02:05.176 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.010 seconds (success: False)
2023-08-08 10:02:05.419 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.2.113', 1502)

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: Hassos 10.4, HA 2023.8.1 
 - solaredge-modbus-multi Version: v2.4.1
 - Installation Type: Supervised

Note: Your issue must be duplicated in an HAOS VM. If your issue is related to running Core you will need to do your own debugging.

**Additional context**
After deleting the inegration, restarting HA and reintegrating of solaredge modbus multi, the integration is working properly

WillCodeForCats, 2023-08-08:
The "Connect call failed" error is from pymodbus, not this integration.

You can try rebooting your inverter or the server Home Assistant is running on.

froggi1930, 2023-08-09:
same issue here. reboot of Inverter and HA did not change anything. No other instance is connecting to the Converter Modbus

WillCodeForCats, 2023-08-09:
Also everyone needs to upload their diagnostic files and debug logs.

froggi1930, 2023-08-09:
> Also everyone needs to upload their diagnostic files and debug logs.

I am new to Home Assistant. What files do you need and where do I find them?

WillCodeForCats, 2023-08-09:
Go to any inverter or meter or battery device info screen and there is a button that says "download diagnostics", it will download a file that can be uploaded here.

froggi1930, 2023-08-09:
> Go to any inverter or meter or battery device info screen and there is a button that says "download diagnostics", it will download a file that can be uploaded here.

Thanks... I will set up a fresh installation and set up the SolardEdge again. In case I run into the same issues I will post the log files.

froggi1930, 2023-08-10:
I have made a fresh installation and get the same error. If I read the modbus with the same settings in iobroker I do not fail.
I can not upload the debug file here... :-(

froggi1930, 2023-08-10:
2023-08-10 15:04:04.303 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solaredge_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-08-10 15:04:04.305 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-08-10 15:04:27.487 DEBUG (MainThread) [custom_components.solaredge_modbus] Setup solaredge_modbus.solaredge
2023-08-10 15:04:28.652 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.176.143, 1502) failed: [Errno 111] Connection refused
2023-08-10 15:04:33.683 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.176.143, 1502) failed: [Errno 111] Connection refused
2023-08-10 15:04:33.692 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 102, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 175, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.176.143:1502)]
2023-08-10 15:04:38.683 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.176.143, 1502) failed: [Errno 111] Connection refused
2023-08-10 15:04:38.687 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 102, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 175, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.176.143:1502)]
2023-08-10 15:04:39.042 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 591, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'unknown'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template_entity.py", line 397, in _async_template_startup
    result_info.async_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1022, in async_refresh
    self._refresh(None)
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1202, in _refresh
    self.hass.async_run_hass_job(self._job, event, updates)
  File "/usr/src/homeassistant/homeassistant/core.py", line 625, in async_run_hass_job
    hassjob.target(*args)
  File "/usr/src/homeassistant/homeassistant/helpers/template_entity.py", line 366, in _handle_results
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 746, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 808, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 752, in _stringify_state
[... 43 lines cut ...]
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.176.143:1502)]
[... cut ...]
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 175, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.176.143:1502)]
2023-08-10 15:04:58.712 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.176.143, 1502) failed: [Errno 111] Connection refused
2023-08-10 15:04:58.714 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):

WillCodeForCats, 2023-08-10:
Those logs are from a different integration, not this one.

WillCodeForCats, 2023-08-10:
I don't know what iobroker is, show debug logs from that and your inverter modbus config.

WillCodeForCats, 2023-08-10:
Also you do know the inverter can only take one connection at a time, if whatever iobroker is connected nothing else can connect.