WillCodeForCats/solaredge-modbus-multi #389: Additional (non-existent) battery detected since 2.4 (pre?)

requiemmg, 2023-08-08:
**Describe the bug**
An additional battery is detected since one of the last upgrades (2.4pre Number ?) where there is none.
Deleting the hub or uninstalling the addon and reinstalling does not fix this.

**Expected behavior**
Only present batteries are being detected.

**Screenshots**
[image]
[image]

**Logs**
Copy any logs in Home Assistant from "custom_components.solaredge_modbus".
--> No logs in that dir. Debug logs below.

**Debug Logs**
(Explanation abou what I did: Removed the hub, started debug logging (which throws some errors due to the missing entities there), then re-added and reconfigured the hub. 2 Batteries were detected, in fact there is only one. This worked perfectly in 2.34 (and I THINK in some 2.4pre or 2.41pre-Versions also, but I am not sure about the latter)

[home-assistant_solaredge_modbus_multi_2023-08-08T13-48-09.321Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12291980/home-assistant_solaredge_modbus_multi_2023-08-08T13-48-09.321Z.log)

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: Home Assistant 2023.8.1
 - solaredge-modbus-multi Version: 2.4.1.-pre.1
 - Installation Type: HAOS

WillCodeForCats, 2023-08-08:
The change that probably caused this was in v2.4.0-pre.8 and I reversed it for 2.4.1

I see what's happening here is that even though you don't have B2 the inverter is still giving a manufacture and model for B2. In the other issue someone had a battery that stopped showing up because the inverter stopped sending "model".

The difficulty is coming up with a detection method that works for all since the current one made it possible for a battery to disappear if the inverter stopped reporting the model name reported in issue #379.

requiemmg, 2023-08-08:
Addition: 2.4.1 fixed the issue. THe additional battery device was deletable now, everything back to normal (for me). Still, #379 might need a workaround.

WillCodeForCats, 2023-08-12:
PR #397 will change to use the rated energy value from the battery identification group.

WillCodeForCats, 2023-08-12:
[Release v2.4.2-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.2-pre.2)

closed 2023-08-12 by WillCodeForCats