WillCodeForCats/solaredge-modbus-multi #394: Integration fails with "Failed setup, will retry"

ExcelsiorEyer, 2023-08-11:
### Describe the bug

The integration fails to setup when trying with just one inverter and one revenue grade meter

### Expected behavior

The required hub and entities are expected to be setup

### Screenshots

[image]

### Diagnostic File

Option unavailable

### Debug logs

```shell
2023-08-11 11:32:14.099 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solaredge_modbus_multi which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-08-11 11:32:22.201 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=2, detect_meters=True, detect_batteries=False, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, battery_rating_adjust=0, 
2023-08-11 11:32:22.201 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-11 11:32:22.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2023-08-11 11:32:22.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2023-08-11 11:32:22.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2023-08-11 11:32:22.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
2023-08-11 11:32:22.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Model SE10000H-US000BNI4
2023-08-11 11:32:22.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Option 
2023-08-11 11:32:22.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Version 0004.0017.0221
2023-08-11 11:32:22.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SerialNumber 730ABDC1
2023-08-11 11:32:22.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Device_address 0x2
2023-08-11 11:32:22.844 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_DID 0x1
2023-08-11 11:32:22.844 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Length 0x41
2023-08-11 11:32:22.844 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Units 0x0
2023-08-11 11:32:22.844 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 is NOT Multiple MPPT
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SunSpec_DID 0x1
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SunSpec_Length 0x41
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Manufacturer SolarEdge
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Model SE-RGMTR-1D-240C-A
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Option Production
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Version 0
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SerialNumber 0
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Device_address 0x1
2023-08-11 11:32:23.073 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Found meter 1 on inverter ID 2
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SunSpec_DID 0x2bd
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SunSpec_Length 0x99
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Manufacturer c	
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Model B
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Option 
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Version 
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SerialNumber 
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Device_address 0x8000
2023-08-11 11:32:23.473 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M2: Meter 2 ident incorrect or not installed.
2023-08-11 11:32:23.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SunSpec_DID 0xffff
2023-08-11 11:32:23.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SunSpec_Length 0x0
2023-08-11 11:32:23.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Manufacturer 
2023-08-11 11:32:23.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Model 
2023-08-11 11:32:23.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Option 
2023-08-11 11:32:23.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Version 
2023-08-11 11:32:23.745 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SerialNumber A
2023-08-11 11:32:23.745 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Device_address 0xffff
2023-08-11 11:32:23.745 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M3: Meter 3 ident incorrect or not installed.
2023-08-11 11:32:26.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 4.642 seconds (success: False)
2023-08-11 11:32:26.843 WARNING (MainThread) [homeassistant.config_entries] Config entry 'waveshare' for solaredge_modbus_multi integration not ready yet: Modbus timeout: ; Retrying in background
2023-08-11 11:33:50.811 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=2, detect_meters=True, detect_batteries=False, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, battery_rating_adjust=0, 
2023-08-11 11:33:50.812 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-11 11:33:50.857 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2023-08-11 11:33:50.857 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2023-08-11 11:33:50.857 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2023-08-11 11:33:50.857 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
[... 117 lines cut ...]
2023-08-11 11:34:32.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Model 
2023-08-11 11:34:32.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Option 
2023-08-11 11:34:32.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Version 
2023-08-11 11:34:32.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SerialNumber A
2023-08-11 11:34:32.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Device_address 0xffff
2023-08-11 11:34:32.469 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M3: Meter 3 ident incorrect or not installed.
2023-08-11 11:34:38.920 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 6.743 seconds (success: False)
```

### Home Assistant Version

2023.8.1

### solaredge-modbus-multi Version

2.4.1

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

WillCodeForCats, 2023-08-11:
The inverter is timing out when requesting data from it, so setup never completes.

ExcelsiorEyer, 2023-08-11:
Not sure I quite understand since I see "some" data about the inverter and the meter in the logs. What sort of data is awaited from the inverter? I tried manually polling the modbus holding registers and they seem to return the data just fine with <500ms response time. Any thoughts on how best to diagnose this. Is this an inverter issue?

WillCodeForCats, 2023-08-11:
The debug logs only show the initial device discovery steps, but when it moves past that into the step where it tries to request the data points to create entities it fails with a timeout:

`2023-08-11 11:33:58.904 WARNING (MainThread) [homeassistant.config_entries] Config entry 'waveshare' for solaredge_modbus_multi integration not ready yet: Modbus timeout: ; Retrying in background`

I can make a guess that it could be one of the several not always supported modbus requests that normally return an invalid address, but it's possible the inverter is timing out instead on those.

WillCodeForCats, 2023-08-11:
These are polled as well as part of the setup/discovery process:

Start: 61440
Size: 4

Start: 61762
Size: 2

Start: 57344
Size: 4

Start: 57362
Size: 2

Start: 61762
Size: 2

These should return invalid address on an inverter that doesn't support them, but if it's timing out instead that will halt the process.

ExcelsiorEyer, 2023-08-11:
Yes. All the four addresses return Modbus error code 2 (Which I believe is for Illegal Data Address??). I did observe that the Response time is 2-3 seconds for every read before the error is returned.

[image]

[image]

[image]

[image]

WillCodeForCats, 2023-08-11:
That's the problem then, the read timeout is 3 seconds so it's taking longer than timeout to give that response.

WillCodeForCats, 2023-08-11:
I have the same series of inverter that you do, except that I'm on firmware 0004.0014.0229 where as you are on 0004.0017.0221.

SolarEdge says the response time is always under 1000ms for Modbus/TCP, but if you're seeing a 2-3 second delay then that's clearly not correct anymore.

ExcelsiorEyer, 2023-08-11:
> That's the problem then, the read timeout is 3 seconds so it's taking longer than timeout to give that response.

Ah interesting! Is that timeout configurable to a slightly higher value like 5 seconds or is that controlled by Home Assistant? Just for giggles I tried the other Solaredge Modbus binding and that seems to set it up correctly but I'm not seeing any Import/Export data in the entities.

WillCodeForCats, 2023-08-11:
That other one doesn't support everything that my integration does, so it's probably not even polling those registers.

WillCodeForCats, 2023-08-11:
It can be configurable I think, but also I don't want to just blanket increase a timeout. What I will need to do is see if I can make it only increase timeout if 61440 or 61762 time out.

ExcelsiorEyer, 2023-08-11:
> I have the same series of inverter that you do, except that I'm on firmware 0004.0014.0229 where as you are on 0004.0017.0221.
> 
> SolarEdge says the response time is always under 1000ms for Modbus/TCP, but if you're seeing a 2-3 second delay then that's clearly not correct anymore.

It's just the Illegal address where I seem to have this lag, but even for that, for the most part it does seem to respond sub 3 seconds. I just tried a stopwatch and I was getting it in 2-2.5 seconds on 10-15 consecutive tries. The response on 40XXX registers is in milliseconds. Not sure if the firmware version upgrade is causing this. Unfortunately, the firmware upgrade happened when I used the SetApp mobile app.

WillCodeForCats, 2023-08-15:
I think I have a solution to increase timeouts in this case, I will make a pre-release soon that you can try.

WillCodeForCats, 2023-08-21:
Please try [Release v2.4.3-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.3-pre.1)

WillCodeForCats, 2023-08-21:
Sorry, there's an error in that release. Quickly making another one.

WillCodeForCats, 2023-08-21:
Please try [Release v2.4.3-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.3-pre.2)

ExcelsiorEyer, 2023-08-21:
```
2023-08-21 18:20:51.805 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=2, detect_meters=True, detect_batteries=False, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
2023-08-21 18:20:51.805 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Model SE10000H-US000BNI4
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Option 
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Version 0004.0017.0221
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SerialNumber 730ABDC1
2023-08-21 18:20:51.843 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Device_address 0x2
2023-08-21 18:20:51.901 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_DID 0x1
2023-08-21 18:20:51.901 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Length 0x41
2023-08-21 18:20:51.901 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Units 0x0
2023-08-21 18:20:51.901 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 is NOT Multiple MPPT
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SunSpec_DID 0x1
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SunSpec_Length 0x41
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Manufacturer SolarEdge
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Model SE-RGMTR-1D-240C-B
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Option Production
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Version 78
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SerialNumber 613503288
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Device_address 0x1
2023-08-21 18:20:51.972 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Found meter 1 on inverter ID 2
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SunSpec_DID 0x2bd
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SunSpec_Length 0x99
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Manufacturer {|c	
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Model Q
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Option 
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Version 
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SerialNumber 
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Device_address 0x8000
2023-08-21 18:20:52.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M2: Meter 2 ident incorrect or not installed.
2023-08-21 18:20:52.123 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SunSpec_DID 0xffff
2023-08-21 18:20:52.123 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SunSpec_Length 0x0
2023-08-21 18:20:52.123 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Manufacturer 
2023-08-21 18:20:52.123 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Model 
2023-08-21 18:20:52.124 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Option 
2023-08-21 18:20:52.124 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Version 
2023-08-21 18:20:52.124 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SerialNumber A
2023-08-21 18:20:52.124 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Device_address 0xffff
2023-08-21 18:20:52.124 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M3: Meter 3 ident incorrect or not installed.
2023-08-21 18:20:55.199 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.394 seconds (success: False)
```

WillCodeForCats, 2023-08-21:
Hmm, it's not increasing the timeout. I'll do some more testing. It should go from `modbus client timeout is 3` to `modbus client timeout is 6`.

ExcelsiorEyer, 2023-08-21:
Thanks for looking into this. I removed the device completely and created a fresh one. However, it made no difference in modbus client timeout.

WillCodeForCats, 2023-08-22:
Sorry, my mistake, the way HA handles setup time errors is it reloads the integration every time it retries so the increase is not saved.

I want to get this working for you so what I'm going to do is just make this a configuration option to set the modbus response timeout (default 3 seconds) for cases like yours where the inverter is slow to respond for reasons only SolarEdge will ever know.

WillCodeForCats, 2023-08-22:
Use [Release v2.4.3-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.3-pre.3)

This will add a new config option:
[image]

Try setting the client timeout to 4-6 seconds.

ExcelsiorEyer, 2023-08-23:
Unfortunately, not much luck. Tried gradually bumping it up all the way to 10 secs. I tested the modbus response multiple times to check if there is any degraded response times. All are at around 3 seconds.

I'm curious as to why there is no error message in logs that there is No response or modbus timeout. It just ends abruptly. 

```
2023-08-22 22:22:16.104 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=2, detect_meters=True, detect_batteries=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, mb_client_timeout=10, 
2023-08-22 22:22:16.104 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-22 22:22:16.104 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] modbus client timeout is 10
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Model SE10000H-US000BNI4
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Option 
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Version 0004.0017.0221
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SerialNumber 730ABDC1
2023-08-22 22:22:16.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Device_address 0x2
2023-08-22 22:22:16.186 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_DID 0x1
2023-08-22 22:22:16.186 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Length 0x41
2023-08-22 22:22:16.186 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Units 0x0
2023-08-22 22:22:16.186 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 is NOT Multiple MPPT
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SunSpec_DID 0x1
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SunSpec_Length 0x41
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Manufacturer SolarEdge
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Model SE-RGMTR-1D-240C-B
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Option Production
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Version 78
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_SerialNumber 613503288
2023-08-22 22:22:16.250 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 1: C_Device_address 0x1
2023-08-22 22:22:16.251 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Found meter 1 on inverter ID 2
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SunSpec_DID 0x2bd
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SunSpec_Length 0x99
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Manufacturer 	
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Model q
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Option 
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Version 
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_SerialNumber 
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 2: C_Device_address 0x8000
2023-08-22 22:22:16.319 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M2: Meter 2 ident incorrect or not installed.
2023-08-22 22:22:16.391 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SunSpec_DID 0xffff
2023-08-22 22:22:16.391 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SunSpec_Length 0x0
2023-08-22 22:22:16.391 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Manufacturer 
2023-08-22 22:22:16.391 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Model 
[... cut ...]
2023-08-22 22:22:56.670 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_SerialNumber A
2023-08-22 22:22:56.670 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2 meter 3: C_Device_address 0xffff
2023-08-22 22:22:56.670 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2M3: Meter 3 ident incorrect or not installed.
2023-08-22 22:23:06.672 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 10.326 seconds (success: False)
```

WillCodeForCats, 2023-08-23:
There would be the first time it happens. the first one if it's repeating the same error, which is this one:

`2023-08-11 11:33:58.904 WARNING (MainThread) [homeassistant.config_entries] Config entry 'waveshare' for solaredge_modbus_multi integration not ready yet: Modbus timeout: ; Retrying in background

It's HA design spec: "Log a warning once when unavailable, log once when reconnected."

WillCodeForCats, 2023-08-23:
Well, we're at the point where obviously I can't duplicate the issue firsthand and it's unique to your inverter for whatever reason (maybe it's specific to the firmware version), so I can keep guessing and trying different things or adding step debug logging.

I'm relatively sure that it's stopping at the point where it asks your inverter if 61762 or 57344 are supported because after it does the stuff it does log, after the last meter check it moves on to querying the inverter for data points.

Are you installing through HACS or manually?

WillCodeForCats, 2023-08-23:
I'll just add step debug logging for every read in the inverter section and see what comes out of that.

WillCodeForCats, 2023-08-23:
Try [Release v2.4.3-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.3-pre.4)

ExcelsiorEyer, 2023-08-23:
No difference in logs observed. No extra lines, nothing. Then I had this nagging feeling about errors at pymodbus level. So turned on debug for pymodbus as well. I see timeouts happening after at 3 seconds eventhough the modbus timeout is set for 10 seconds. Logs below for reference.

```
2023-08-23 13:00:47.649 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=2, detect_meters=True, detect_batteries=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, mb_client_timeout=10, 
2023-08-23 13:00:47.649 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-23 13:00:47.649 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] modbus client timeout is 10
2023-08-23 13:00:47.649 DEBUG (MainThread) [pymodbus.logging] Connecting to 10.20.13.22:502.
2023-08-23 13:00:47.649 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-08-23 13:00:47.651 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2023-08-23 13:00:47.652 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x2 0x3 0x9c 0x40 0x0 0x45
2023-08-23 13:00:47.652 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x8d 0x2 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x31 0x30 0x30 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x49 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x37 0x2e 0x30 0x32 0x32 0x31 0x0 0x0 0x37 0x33 0x30 0x41 0x42 0x44 0x43 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x8d 0x2 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x31 0x30 0x30 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x49 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x37 0x2e 0x30 0x32 0x32 0x31 0x0 0x0 0x37 0x33 0x30 0x41 0x42 0x44 0x43 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x8d 0x2 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x31 0x30 0x30 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x49 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x37 0x2e 0x30 0x32 0x32 0x31 0x0 0x0 0x37 0x33 0x30 0x41 0x42 0x44 0x43 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] [21365, 28243, 1, 65, 21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 12592, 12336, 12360, 11605, 21296, 12336, 16974, 18740, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12340, 11824, 12337, 14126, 12338, 12849, 0, 14131, 12353, 16964, 17201, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2]
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] handle: [b'Su', b'nS']
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Model SE10000H-US000BNI4
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Option 
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Version 0004.0017.0221
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SerialNumber 730ABDC1
2023-08-23 13:00:47.721 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Device_address 0x2
2023-08-23 13:00:47.721 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x2 0x3 0x9c 0xb9 0x0 0x9
2023-08-23 13:00:47.722 DEBUG (MainThread) [pymodbus.logging] Adding transaction 2
[... cut ...]
2023-08-23 13:00:57.964 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 10.315 seconds (success: False)
2023-08-23 13:00:57.964 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2023-08-23 13:00:57.964 DEBUG (MainThread) [pymodbus.logging] Getting transaction 6
2023-08-23 13:00:58.052 DEBUG (MainThread) [pymodbus.logging] Connection to Modbus server established. Socket ('10.20.11.14', 43538)
2023-08-23 13:00:58.053 DEBUG (MainThread) [pymodbus.logging] Current transaction state - TRANSACTION_COMPLETE

```

WillCodeForCats, 2023-08-23:
Do you have other modbus stuff running? It kind of looks like the logging is mixed in with stuff I don't recognize.

ExcelsiorEyer, 2023-08-23:
Oh yes... The other Solaredge modbus integration is running. Didn't realize that it could be the one using the pymodbus. I don't have batteries.

WillCodeForCats, 2023-08-23:
Also remember that solaredge inverters are only able to accept one modbus connection at a time. Have you been trying to run them together from the beginning?

Disable anything else that tries to access the inverter and only run one thing at a time when testing.

ExcelsiorEyer, 2023-08-23:
No. I had just enabled the other one after repeated failed attempts. Its just that I grabbed the logs from the last section without disabling this integration. My bad. Here is the updated log with no other modbus integration active.

```
2023-08-23 14:33:18.172 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=2, detect_meters=True, detect_batteries=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, mb_client_timeout=10, 
2023-08-23 14:33:18.172 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-23 14:33:18.172 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] modbus client timeout is 10
2023-08-23 14:33:18.172 DEBUG (MainThread) [pymodbus.logging] Connecting to 10.20.13.22:502.
2023-08-23 14:33:18.172 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-08-23 14:33:18.194 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2023-08-23 14:33:18.202 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x2 0x3 0x9c 0x40 0x0 0x45
2023-08-23 14:33:18.202 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-08-23 14:33:19.629 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x8d 0x2 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x31 0x30 0x30 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x49 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x37 0x2e 0x30 0x32 0x32 0x31 0x0 0x0 0x37 0x33 0x30 0x41 0x42 0x44 0x43 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2
2023-08-23 14:33:19.629 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x8d 0x2 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x31 0x30 0x30 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x49 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x37 0x2e 0x30 0x32 0x32 0x31 0x0 0x0 0x37 0x33 0x30 0x41 0x42 0x44 0x43 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2
2023-08-23 14:33:19.629 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x8d 0x2 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x31 0x30 0x30 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x49 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x37 0x2e 0x30 0x32 0x32 0x31 0x0 0x0 0x37 0x33 0x30 0x41 0x42 0x44 0x43 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2
2023-08-23 14:33:19.630 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-08-23 14:33:19.630 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-08-23 14:33:19.634 DEBUG (MainThread) [pymodbus.logging] [21365, 28243, 1, 65, 21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 12592, 12336, 12360, 11605, 21296, 12336, 16974, 18740, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12340, 11824, 12337, 14126, 12338, 12849, 0, 14131, 12353, 16964, 17201, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2]
2023-08-23 14:33:19.634 DEBUG (MainThread) [pymodbus.logging] handle: [b'Su', b'nS']
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Model SE10000H-US000BNI4
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Option 
2023-08-23 14:33:19.634 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Version 0004.0017.0221
2023-08-23 14:33:19.635 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SerialNumber 730ABDC1
2023-08-23 14:33:19.635 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Device_address 0x2
2023-08-23 14:33:19.635 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x2 0x3 0x9c 0xb9 0x0 0x9
2023-08-23 14:33:19.635 DEBUG (MainThread) [pymodbus.logging] Adding transaction 2
[... cut ...] _energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, mb_client_timeout=10, 
2023-08-23 14:33:41.274 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-08-23 14:33:41.274 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] modbus client timeout is 10
2023-08-23 14:33:41.274 DEBUG (MainThread) [pymodbus.logging] Connecting to 10.20.13.22:502.
2023-08-23 14:33:41.274 DEBUG (MainThread) [pymodbus.logging] Connecting comm
```

WillCodeForCats, 2023-08-23:
Thanks, that's much clearer.

WillCodeForCats, 2023-08-23:
It's following the 10 second timeout, but clearly increasing the timeout isn't the issue.

```
2023-08-23 14:33:20.840 DEBUG (MainThread) [pymodbus.logging] Adding transaction 6
2023-08-23 14:33:30.842 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2023-08-23 14:33:30.842 DEBUG (MainThread) [pymodbus.logging] Getting transaction 6
```

I will need to look at the pymodbus code to find out what `lost comm due to None` could mean.

This line means it's asking the inverter if there is a Battery 1 attached:

```
2023-08-23 14:33:20.840 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x6 0x0 0x0 0x0 0x6 0x2 0x3 0xe1 0x0 0x0 0x4c
```

And at that point it fails with no response or lost communications.

ExcelsiorEyer, 2023-08-23:
I just downloaded the pymodbusTCP on python. It looks like it is not returning an error and instead is simply timing out (may be because of the firmware??). Even with the default 30 seconds of timeout, there was still no response.

```
Python 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyModbusTCP.client import ModbusClient
>>> c = ModbusClient(host="inverter.home.arpa", port=502, unit_id=2, auto_open=True)
>>> print(c.debug)
False
>>> print(c.timeout)
30.0
>>> regs = c.read_holding_registers(40000, 2)
>>> print(regs)
[21365, 28243]
>>> regs = c.read_holding_registers(57600, 76)
>>> c.timeout = 10
>>> c.debug=True
>>> regs = c.read_holding_registers(57600, 76)
Tx
[84 73 00 00 00 06 02] 03 E1 00 00 4C
timeout error
>>> print(regs)
None
>>>
```

The register value is returned as None by pymodbus. May be it is that??

WillCodeForCats, 2023-08-24:
Since there appears to be something going on here that is out of my control at the moment (pymodbus) I'm going to wrap those discovery calls into the "Power Control Options" option so they won't even be called unless that option is enabled.

WillCodeForCats, 2023-08-24:
Try [Release v2.4.3-pre.5](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.3-pre.5)

Make sure detect batteries and power control options are turned off, and it should no longer try discovery on the registers your inverter doesn't like.

ExcelsiorEyer, 2023-08-24:
Thanks @WillCodeForCats. I have a working integration now!

WillCodeForCats, 2023-08-25:
[Release v2.4.3-pre.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.3-pre.7) should be the final iteration for this issue.

You will want to turn off the option "Auto-Detect Additional Entities" which will be enabled by default. This will prevent it from querying the additional feature locations that are causing the problem you're seeing.

closed 2023-08-25 by WillCodeForCats