WillCodeForCats/solaredge-modbus-multi #395: Battery marked as not usable

DougMunford, 2023-08-12:
### Describe the bug

Prior to 2.4.1 batteries were set up as devices. Afterwards they are being detected and marked as not usable.

Setup:
Inverter 1 + Battery
Inverter 2 + Battery
Inverter 3.

### Expected behavior

Create battery devices on configuration start.

### Diagnostic File

[config_entry-solaredge_modbus_multi-f2f5b2f77efaaacd5aaeec643ebabc99.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12327064/config_entry-solaredge_modbus_multi-f2f5b2f77efaaacd5aaeec643ebabc99.json.txt)

### Debug logs

```shell
2023-08-12 10:26:49.761 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=3, start_device_id=1, detect_meters=True, detect_batteries=True, keep_modbus_open=False, adv_storage_control=True, adv_site_limit_control=False, allow_battery_energy_reset=True, battery_rating_adjust=0,
2023-08-12 10:26:49.761 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 40.8
2023-08-12 10:26:49.776 WARNING (MainThread) [custom_components.solaredge_modbus_multi.hub] Power Control Options: Storage Control is enabled. Use at your own risk!
2023-08-12 10:26:49.870 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2023-08-12 10:26:49.871 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2023-08-12 10:26:49.871 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2023-08-12 10:26:49.871 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Manufacturer SolarEdge
2023-08-12 10:26:49.871 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Model SE4000H-RW000BNN4
2023-08-12 10:26:49.873 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Option
2023-08-12 10:26:49.873 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Version 0004.0018.0032
2023-08-12 10:26:49.874 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SerialNumber 740EDF41
2023-08-12 10:26:49.874 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Device_address 0x1
2023-08-12 10:26:49.919 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_DID 0x1
2023-08-12 10:26:49.919 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_Length 0x41
2023-08-12 10:26:49.919 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_Units 0x0
2023-08-12 10:26:49.919 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 is NOT Multiple MPPT
2023-08-12 10:26:49.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_DID 0x1
2023-08-12 10:26:49.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_Length 0x41
2023-08-12 10:26:49.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Manufacturer WattNode
2023-08-12 10:26:49.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Model WND-3Y-400-MB
2023-08-12 10:26:49.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Option
2023-08-12 10:26:49.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Version 25
2023-08-12 10:26:49.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SerialNumber 4130879
2023-08-12 10:26:49.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Device_address 0x2
2023-08-12 10:26:49.961 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Found meter 1 on inverter ID 1
2023-08-12 10:26:50.010 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_SunSpec_DID 0x2bd
2023-08-12 10:26:50.010 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_SunSpec_Length 0x99
2023-08-12 10:26:50.010 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_Manufacturer \c H
2023-08-12 10:26:50.010 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_Model 
2023-08-12 10:26:50.010 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_Option
2023-08-12 10:26:50.011 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_Version H
2023-08-12 10:26:50.011 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_SerialNumber
2023-08-12 10:26:50.011 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_Device_address 0x8000
2023-08-12 10:26:50.011 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: Meter 2 ident incorrect or not installed.
2023-08-12 10:26:50.049 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_SunSpec_DID 0xffff
2023-08-12 10:26:50.049 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_SunSpec_Length 0x0
2023-08-12 10:26:50.049 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_Manufacturer 
2023-08-12 10:26:50.050 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_Model 
2023-08-12 10:26:50.050 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_Option 
2023-08-12 10:26:50.050 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_Version
2023-08-12 10:26:50.050 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_SerialNumber A
2023-08-12 10:26:50.050 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_Device_address 0xffff
2023-08-12 10:26:50.051 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: Meter 3 ident incorrect or not installed.
2023-08-12 10:26:50.110 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Manufacturer SolarEdge
2023-08-12 10:26:50.110 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Model
2023-08-12 10:26:50.110 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Version SMCU 1.1.9 DCDC 0.0.81 BMS 0.6.9
2023-08-12 10:26:50.110 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_SerialNumber 6307661F
2023-08-12 10:26:50.111 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Device_Address 0x661f
2023-08-12 10:26:50.111 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: Reserved 0x0
2023-08-12 10:26:50.111 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_RatedEnergy 0x46179000
[... 117 lines cut ...]
2023-08-12 10:26:53.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 3 batt 2: Reserved 0x0
2023-08-12 10:26:53.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 3 batt 2: B_RatedEnergy 0xff7fffff
2023-08-12 10:26:53.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 3 batt 2: B_MaxChargePower 0xff7fffff
2023-08-12 10:26:53.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 3 batt 2: B_MaxDischargePower 0xff7fffff
2023-08-12 10:26:53.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 3 batt 2: B_MaxChargePeakPower 0xff7fffff
2023-08-12 10:26:53.713 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 3 batt 2: B_MaxDischargePeakPower 0xff7fffff
2023-08-12 10:26:53.714 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I3B2: Battery 2 not usable.
```

### Home Assistant Version

2023.8.2

### solaredge-modbus-multi Version

2.4.1

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

WillCodeForCats, 2023-08-12:
This is because the inverter is returning the battery "model" as blank, see issue #379 for more information.

WillCodeForCats, 2023-08-12:
[Release v2.4.2-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.2-pre.2)

closed 2023-08-12 by WillCodeForCats