WillCodeForCats/solaredge-modbus-multi #396: Entities becoming unavailable after Inverter not responding.

requiemmg, 2023-08-12:
### Describe the bug

Solaredge Inverters are obviously becoming unresponsive for a minute or so every day. See https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084/392?u=requiemmg

If the modbus is polled in this time, it will throw an error, thus making some entities unavailable.
All other entities derived from the unavailable entities will become unavailable, too as the timout generally seem to happen during the night hours when the inverter is not producing power.

The entites will become available again either after a reboot or the next day when the inverter starts producing again.

### Expected behavior

As for a short timeout error, the addon should be fault tolerant and keep the previous values so that the entities will not become unavailable at once.

### Screenshots

[image]

### Diagnostic File

This log file is from Modbus Proxy, showing the error. Happened exactly 24 hours before in the same manner, duration less than a minute, then startet working again.

Below is the snippet from the debuglog of that specific occurrence.

```
2023-08-11 22:57:03,319    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:03,325    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:07,338    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:07,342    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:09,069    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:09,074    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:09,885    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:09,891    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:22,305    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: TimeoutError()
2023-08-11 22:57:32,308    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: TimeoutError()
2023-08-11 22:57:39,509    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:39,515    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:39,520    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:39,525    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:39,531    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:39,536    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:43,377    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [1/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
2023-08-11 22:57:43,382    ERROR modbus-proxy.ModBus(192.168.0.175:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.0.175', 502)")
```

### Debug logs

```shell
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_DC_Voltage 0x444d4e0f
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_DC_Current 0x3f7d038a
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_DC_Power 0xc44b0000
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Export_Energy_WH 0x2b8f
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Import_Energy_WH 0x29ba
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Energy_Max 0x460fc000
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Energy_Available 0x460014cd
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_SOH 0x42c60000
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_SOE 0x4262aaab
2023-08-11 22:56:49.264 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Status 0x4
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Status_Vendor 0x3
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log1 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log2 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log3 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log4 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log5 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log6 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log7 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log8 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor1 0x0
2023-08-11 22:56:49.265 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor2 0x0
2023-08-11 22:56:49.266 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor3 0x0
2023-08-11 22:56:49.266 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor4 0x0
[... 183 lines cut ...]
2023-08-11 22:57:48.933 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] B_Import_Energy went backwards: 0 < 10682 cycle 1 of 0
2023-08-11 22:57:48.933 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] B_Import_Energy reset at cycle 1
2023-08-11 22:57:48.951 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Solar Imported Power Daily Solar Edge received an invalid new state from sensor.solaredge_m1_imported_kwh : unknown
2023-08-11 22:57:48.965 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Solar Imported Power Weekly Solar Edge received an invalid new state from sensor.solaredge_m1_imported_kwh : unknown
2023-08-11 22:57:48.973 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Solar Imported Power Monthly Solar Edge peak received an invalid new state from sensor.solaredge_m1_imported_kwh : unknown
2023-08-11 22:57:48.978 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Solar Imported Power Yearly Solar Edge received an invalid new state from sensor.solaredge_m1_imported_kwh : unknown
2023-08-11 22:58:08.060 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 10.8
```

### Home Assistant Version

2023.8.1

### solaredge-modbus-multi Version

2.4.1

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

WillCodeForCats, 2023-08-12:
When an inverter goes offline (returns an error, incomplete read, or other errors) entities are supposed to become unavailable so stale or bad data is not reported: this a by design, not a bug. Entities should be able to handle this condition and recover from it.

This is a template issue that needs to be modified to check the entity is numeric, add a default value, or possibly add ` availability_template` to it. I don't really use templates much myself so I don't know the correct answer to give here.

`2023-08-11 22:57:25.547 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: round got invalid input 'unavailable' when rendering template '{{ states('sensor.solaredge_b1_state_of_energy') | round(0) }}' but no default was specified') while processing template 'Template<template=({{ states('sensor.solaredge_b1_state_of_energy') | round(0) }}) renders=786>' for attribute '_attr_native_value' in entity 'sensor.battery_in_percent'`

Instead of using round like that, this was my recommendation:

https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084/261

Use `states('sensor.X', rounded=True)` which avoids the use of `round(0)` but also applies the entity apply the default precision setting.

More information on `availability_template`:

https://www.home-assistant.io/integrations/template/

https://www.home-assistant.io/docs/configuration/templating/#states

https://community.home-assistant.io/t/template-sensor-with-unavailable-sensor/416676

https://community.home-assistant.io/t/how-to-prevent-unavailable-in-template-sensors/398409/3

WillCodeForCats, 2023-08-12:
I'm not using `utility_meter` helpers, but I added some to my installation to observe the result the next time my inverter #2 cycles through being unavailable for a minute.

WillCodeForCats, 2023-08-22:
I've been running utility meter as solar daily production since August 12th and I have not been able to duplicate this issue.

requiemmg, 2023-08-23:
I simply use the sensors from the packages dir here: https://github.com/Remko76/hasolarcfg
Not really utility meters,are they?

closed 2023-08-24 by WillCodeForCats