WillCodeForCats/solaredge-modbus-multi #399: Entities not available anymore after automatic firmware update

john8329, 2023-08-12:
### Describe the bug

It looks like that after today's update of the inverter to v4.18.32 everything related to the solaredge integration broke. Entities are not available anymore except some basic ones. The battery isn't available too. Everything stopped working since exactly the time of the release, and I suspect it's 90% some integration change. The module is able to connect via Modbus, and isn't reporting errors.

I tried upgrading to 2.4.1 but nothing changed. There's no mention about protocol changes in the [release notes](https://www.solaredge.com/en/support/setapp-inverters-firmware#_Known_Issues_and32) it seems.

### Expected behavior

I expect to see all entities. Right now solar production and battery levels are completely missing.

### Diagnostic File

[config_entry-solaredge_modbus_multi-566b2f71e309bda3d0d7947a9df6bb8a.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12328364/config_entry-solaredge_modbus_multi-566b2f71e309bda3d0d7947a9df6bb8a.json.txt)

### Home Assistant Version

2023.7

### solaredge-modbus-multi Version

2.4.1

### Installation Type

Container

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

### Additional Context

I should really disconnect the damn inverter from the internet next time.

WillCodeForCats, 2023-08-12:
Data looks fine to me.

Post a screenshot of your device page.

john8329, 2023-08-12:
[image]

There you go, thanks for the support.

WillCodeForCats, 2023-08-12:
And of the actual device page once you click on it to show all the sensors and stuff.

john8329, 2023-08-12:
[image]

As you can see some entities are available, but not all of them. I remember being way more.
I believe the official app glitched also today, but it worked after a while.

WillCodeForCats, 2023-08-12:
I mean that looks normal to me. There aren't more, unless you turn on the ones classified as advanced.

WillCodeForCats, 2023-08-12:
As far as the battery you will need to run at least [Release v2.4.2-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.2-pre.2) and newer because SE started omitting an identifier used for battery detection.

john8329, 2023-08-12:
I'm missing the entity `solar_panel_production_w` which is crucial to know the current solar production. Not sure 100% if it's provided by this module or another calculation, I'll have to check. All I know is that it correlates precisely with the firmware.

I upgraded to the prerelease (was afraid to do it) and now the battery seems to work! Also the production value is now zero and not NaN, I'll have to check tomorrow when photons start raining again.

Guess this is solved with the beta. Thanks again, this integration is amazing and well maintained.

closed 2023-08-12 by john8329

WillCodeForCats, 2023-08-12:
This is my inverter still running firmware 0004.0014.0229:
[image]

So you aren't missing anything in that regard from the update.

john8329, 2023-08-12:
Good to know (there's one more entity though). Again, not sure where the production value is taken from, but the main problem was the battery. I can dig deeper myself into that in case I encounter more issues.

WillCodeForCats, 2023-08-12:
There is no entity in this integration called `solar_panel_production_w`.

The default name for inverter power is `sensor.solaredge_iN_ac_power` where `N` is the inverter ID number.

john8329, 2023-08-12:
You're right, it's calculated as a difference, which is why it glitches sometimes.

WillCodeForCats, 2023-08-12:
If it's a template sensor there's a couple things to be aware of to make template entities more robust.

First is don't use `round()` in templates. It will error if the entity goes into any non-numeric state (like offline). Instead try using the `rounded=True` property with `states()` like this: `states('sensor.X', rounded=True)` which avoids the use of `round()`, does the same thing, but also applies the entity the default precision setting from the UI.

Second, with templates you can include `availability_template` which defines a template to get the available state of the entity. More information on `availability_template`:

https://www.home-assistant.io/integrations/template/

https://www.home-assistant.io/docs/configuration/templating/#states

https://community.home-assistant.io/t/template-sensor-with-unavailable-sensor/416676

https://community.home-assistant.io/t/how-to-prevent-unavailable-in-template-sensors/398409/3

john8329, 2023-08-12:
Thanks for the info. I believe a possible cause for the glitch could be a timing difference between sampling from the inverter and the meter. Even half a second can produce a few watts of difference, which then make the panels look like they're producing. 

Not a real concern, and I'm going off topic :)

WillCodeForCats, 2023-08-12:
Yes that's possible, although they are read as rapidly as possible the inverter data block and the meter data block are indeed separate reads. They can't be read together because it would exceed modbus maximum response size to read at the same time.