WillCodeForCats/solaredge-modbus-multi #400: Integration stoped working after update to HAAS2023.8.2

LBHatHA, 2023-08-14:
### Describe the bug

Integration not able to connect via ModbusTCP to Inverter after updating HAAS to 2023.8.2 version. At the same time as well Home Assistant operating system was upgraded to 10.4

### Expected behavior

It was working fine before HAAS upgrade, Inverter data and smart meter data was available.

### Screenshots

[image]

[image]

### Diagnostic File

[home-assistant_solaredge_modbus_multi_2023-08-14T15-49-14.697Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12337020/home-assistant_solaredge_modbus_multi_2023-08-14T15-49-14.697Z.log)

### Debug logs

```shell
2023-08-14 16:49:30.288 WARNING (MainThread) [custom_components.solaredge_modbus_multi] Polling frequency < 10, requiring keep modbus open.
2023-08-14 16:49:33.316 ERROR (SyncWorker_5) [pymodbus.logging] Connection to (192.168.1.234, 1502) failed: timed out
2023-08-14 16:49:33.347 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: Could not open Modbus/TCP connection to 192.168.1.234; Retrying in background
```

### Home Assistant Version

2023.8.2

### solaredge-modbus-multi Version

2.4.1

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

### Additional Context

No other configuration changes were done recently, invertor is visible on the LAN and communicates with solaredge ecosystem.

WillCodeForCats, 2023-08-14:
You attached a log file, not the diagnostic file. But with these type of connection errors I don't really know what I'm supposed to do: if pymodbus says it can't connect, it can't connect and that's all there is about it.

I run 2023.8.2 as HAOS and the latest pre-relse of my own integration, so this falls under "unable to duplicate".

LBHatHA, 2023-08-14:
ok, thank you for explanation.

Jan

WillCodeForCats, 2023-08-14:
The thing is I would never make a release that didn't work to my own satisfaction on my Home Assistant installation (other than a mistake then quickly pushing a new release), so if there was actually an integration-related problem causing connection failures I would probably see them first and fix it without waiting for anyone else to report it.

I'm running HA 2023.8.2, HAOS, OS 10.4, and release [v2.4.2-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.2-pre.2) installing through HACS. I do not use wireless connections anywhere.

If we're running the same versions same HAOS environment, where mine connects but yours doesn't connect, then the problem is likely somewhere else. The only advice I can really give as far as an issue report with the information I see is to try and power cycle your hardware (network and inverters).

LBHatHA, 2023-08-14:
Hello,

I appreciate all your work on this plugin and your help! Looks like it
start working again. I have physically shut down my Haas device. Solaredge
ModBus plugin started working like 25 minutes after powering it up again.
Absolutely no idea what caused the outage, looks like it was just
coincidence it lost connection to inverter right after Hass update. I
didn't notice any issue with network nor inverter during the outage,
hopefully it will remain stable now.

Thank you again, please feel free to close the issue report I logged .

Many thanks for your support
Jan

closed 2023-08-14 by WillCodeForCats