WillCodeForCats/solaredge-modbus-multi #401: still errore on multiple commands

Condorello, 2023-08-17:
### Describe the bug

I have an automation that First change the storage default mode to "solar only" (or Change only etc) and After set the site limit to 1, or 5000 deoending on the automation. The script Works fine, but often Just One setting Is changed,  because the script terminate with this error:

Choose: Option 1 executed
Stopped because an error was encountered at 17 agosto 2023 alle ore 11:03:39 (runtime: 0.01 seconds)
Connection to inverter ID 1 failed.

I already tried to inserti a delay of 1 minute between the command, but nothinf changes.

If i run again the automation by hand also the second setting Is applied.

### Expected behavior

All commands must be processed

### Diagnostic File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "6.1.39",
    "supervisor": "2023.08.1",
    "host_os": "Home Assistant OS 10.4",
    "docker_version": "23.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "frigate": {
      "version": "4.0.0",
      "requirements": [
        "pytz==2022.7"
      ]
    },
    "solaredge_modbus_multi": {
      "version": "2.4.2-pre.2",
      "requirements": [
        "pymodbus>=3.3.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "solaredge_modbus_multi",
    "name": "SolarEdge Modbus Multi",
    "codeowners": [
      "@WillCodeForCats"
    ],
    "config_flow": true,
    "dependencies": [],
    "documentation": "https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki",
    "integration_type": "hub",
    "iot_class": "local_polling",
    "issue_tracker": "https://github.com/WillCodeForCats/solaredge-modbus-multi/issues",
    "loggers": [
[... 335 lines cut ...]
        "B_Event_Log_Vendor5": "0x0",
        "B_Event_Log_Vendor6": "0x0",
        "B_Event_Log_Vendor7": "0x0",
        "B_Event_Log_Vendor8": "0x0"
      }
    }
  }
}

### Home Assistant Version

2023.8.2

### solaredge-modbus-multi Version

2.4.2 pre 2

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

closed 2023-08-17 by WillCodeForCats

WillCodeForCats, 2023-08-17:
Try [Release v2.4.2-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.2-pre.3) and setting Inverter Command Delay between 3-10 seconds.

Condorello, 2023-08-22:
Seems to be solved with pre.3 and 20ms delay + 30sec pause during the automation... Up and working since 3 days, thanks!