WillCodeForCats/solaredge-modbus-multi #419: Recorded energy values are wrong

FranciscoBigWings, 2023-08-27:
### Describe the bug

Hi,

I have an issue with energy configuration and recorded energy values since 2.4.X

My inverters and battery configuration:
- 1 EV inverter (leader) of 5kW
- 1 StorEdge inverter (follower) 5kW
- 1 LG RESU battery 10kW

Everything worked after updating to 2.4. and then, (I think that it happened when a command of Storage Default Mode is changed) the recorded value of solar production spikes (even if at the time there is no solar production). After the spike in solar production, the sensor for AC energy kwh in energy configuration changes to unavailable and no more values are recorded in energy tab; but then (day after) in my energy tab appears another spike even if the sensor is unavailable and no data is recorded.

Notes: 
- To fix the mentioned issue, the only solution I found was to configure again the integration but I found out that it happened again.
- The energy value that spikes is only for EV inverter (leader)
- The energy values of Storedge inverter (follower) are always fine, no spikes and always available
- In 2.3.X this didnt happen
- I noted that since 2.4 when changing value of Storage Default Mode all sensors of the integration change to unavailable (sometime 2 seconds and other times almost 10 seconds). Lets remember that Storage Default Mode is changed in leader inverter even if the follower inverter is the Storage inverter with the attached battery.

I hope that my analyis and pictures can help find the issue

### Expected behavior

- Real energy records witout spikes as in 2.3 version
- Shorter times of unavailability (or none) of sensors within the intergration when Storage Default Mode

### Diagnostic File

[image]
[image]
[image]

### Home Assistant Version

2023.8.4

### solaredge-modbus-multi Version

2.4.2

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

WillCodeForCats, 2023-08-27:
Nothing has changed in the integration with regard to how data is returned. The integration passes the data across the inverter gives it. That will always be the case.

In v2.4.0 the inverter command delay was removed because I thought it might be unnecessary after rewriting the update process to be more pure async, but it also might be the case that some inverters return bad data after sending them commands. So the inverter command delay feature was restored in v2.4.2 but the default is now 0 instead of 3, so f you were just using the default before you could try changing it to a value between 3-10 seconds (or more) and see if that helps. You didn't attach a diagnostic file so I don't know what your settings are.

FranciscoBigWings, 2023-08-27:
Hi,

Something must have changed or some change that affected somehow this; I cant say because you did the changes and were many buy I am sure this issue never happened in 2.3.X

- I attach diagnostic file.
- I tried configuration of the integration with Command delay of 6 seconds (what I had used before), 3 seconds and 0 seconds. I have to say that still, when I change Storage default mode, in all cases all sensors from the integration change unavailable and are restored. Maybe 0 seconds got them available faster than the others.
- I will see what happens with energy records tomorrow and the day after.
- When I add the inverter again to the energy configuration for the records of solar production, I still have the wrong records from that inverter but I cant find the record itself to delete or change to 0. Do you have a suggestion for this? The wrong records completely mess up my historic records and statistics.

Thanks
[config_entry-solaredge_modbus_multi-aa74d1e75f85eaa741ca4b8728e80c78.json (1).txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12448325/config_entry-solaredge_modbus_multi-aa74d1e75f85eaa741ca4b8728e80c78.json.1.txt)

WillCodeForCats, 2023-08-27:
Every change is here:

https://github.com/WillCodeForCats/solaredge-modbus-multi/compare/v2.3.4...v2.4.2

There are no changes to the sensor entity platform, except for fixing a bug in the battery energy cycle count feature. A new button was added and a new switch was added.

I don't have batteries and my inverters don't support commands, so my ability to help specifically with those things is very limited. I can't actually experience how sending commands affects a real inverter, and my simulator is always well-behaved when handling commands. My simulator lets me prove the right things are being sent and received, but it doesn't have all the "problems" like a real inverter or battery can.

The integration is designed to go unavailable if the inverter stops responding. The only suggestion I have after sending a command is to increase the inverter command delay higher. SolarEdge's own documents say the inverter may take 3-6 seconds to process a command so try 10 seconds, or 30. But again I can't offer any practical advice with this since I can't try it on my system.

You can also try manually installing the pymodbus 3.4.1 pinned branch:

https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/pymodbus_3_4_1.zip

Home Assistant 2023.8 uses pymodbus 3.3.1 but I would I expect to see pymodbus 3.4.1 in HA core soon since there was a recent PR (https://github.com/home-assistant/core/pull/97612) for that. If you don't want to try installing that branch manually you then can just wait until the next Home Assistant release that includes pymodbus 3.4.1 which will probably be in 2023.9 soon. I can't pin my integration to pymodbus 3.4.1 until after HA updates because pydmobus 3.3.1->3.4.1 contains a change that will break the built-in modbus integration.

FranciscoBigWings, 2023-08-28:
I have made many tests with the inverter command delay:
- If 0, when changing storage default mode, all sensors go unavailable instantly but also go available instantly (1 second or less)
- If I start increasing the delay, the sensors go unavailable around the same time of the delay but sometimes go available instantly and others not. The delay basically, postpones the time the sensors go unavailable but the always go unavailable.

Even if I think there is some kind of relationship between the wrong energy data and the unavailability of the sensors but it doesnt explain by itself why the bad records of energy measures. What do you think? I checked your changes and I couldnt read anything that would explain this error.

I dont feel confindent enough to install that manually... I am sorry

WillCodeForCats, 2023-08-28:
Turn on debug logging to catch it happening and upload the full log.

WillCodeForCats, 2023-08-28:
The only other idea I have at the moment is to go back to using the pymodbus sync client with workers instead of the pure async client that was implemented for release v2.4.0. There has been another random error that happens (PR #424) that could be related.

I can prepare a pre-release that does that and see if it helps.

FranciscoBigWings, 2023-08-28:
Great. Quote me when you do so and I will test itThanksFrancisco F. Mayoralas

WillCodeForCats, 2023-08-29:
@FranciscoBigWings

Try [Release v2.4.4-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.4-pre.1)

I'm trying a different arrangement with the locking and write timing, plus adding a check not to start reading if there is a write in progress.

FranciscoBigWings, 2023-08-30:
I installed it yesterday night with a delay of 3 seconds. So far so good.

I tried to change manually the state of Storage default mode and it sends it well and the sensor of the integration are always available. Not even 1 time went unavailable. I am still observing if for any reason a wrong value of energy data is recorded, because for the moment everthing is ok. I will update in some days or if it fails.

I am trying to delete an old record from the integration related to energy data. I see the point of data (marked in red) but if I try to find that data I cant find it to delete it or change it to 0kWh. Any suggestion?
[image]

WillCodeForCats, 2023-08-30:
Under Developer Tools -> Statistics you can adjust the incorrect statistic.

WillCodeForCats, 2023-08-30:
I don't see the struct errors at all in use, so it's probably related to commands only (since my inverters don't support commands).

I happened to see it once in my dev environment when I was spamming my simulator with commands to see if it would process them in order from the UI.

FranciscoBigWings, 2023-08-30:
Yes, that’s how I was used to do it. The problem is that the data doesn’t appear in Statistics to adjust the value to 0Francisco F. Mayoralas

FranciscoBigWings, 2023-09-03:
Hi,

2.4.4-pre1 was working pretty well. Yesterday I installed 2.4.4-pre2

Related to the spikes in energy dashboard, they didnt happen again since 2.4.4-pre2 but searching for a solution about my misterious spikes I found this conversation. Maybe it is useful for @WillCodeForCats 
https://community.home-assistant.io/t/data-spikes-in-the-energy-dashboard/469843

FranciscoBigWings, 2023-09-03:
I can confirm that in pre2 I have sometimes some issues that sensors go unavailable when trying to change the Storage Default mode or Storage charge limit

Log that I hope it helps you:

Logger: homeassistant.helpers.template_entity
Source: helpers/template_entity.py:362 
First occurred: 07:43:18 (2 occurrences) 
Last logged: 16:30:11

TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.solaredge_i1_ac_power') | float) + (states('sensor.solaredge_i2_ac_power') | float) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solaredge_i1_ac_power') | float) + (states('sensor.solaredge_i2_ac_power') | float) }}) renders=414>' for attribute '_attr_native_value' in entity 'sensor.solar_production'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.solaredge_i1_ac_power') | float) + (states('sensor.solaredge_i2_ac_power') | float) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solaredge_i1_ac_power') | float) + (states('sensor.solaredge_i2_ac_power') | float) }}) renders=16906>' for attribute '_attr_native_value' in entity 'sensor.solar_production'

Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:262 
Integration: SolarEdge Modbus Multi ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues)) 
First occurred: 16:30:11 (1 occurrences) 
Last logged: 16:30:11

Timeout fetching SolarEdge Coordinator data

WillCodeForCats, 2023-09-03:
Timeout is inverter timeout, not an integration issue.

WillCodeForCats, 2023-09-03:
For the template issues, I made some suggestions on #396 how to fix them.

WillCodeForCats, 2023-09-06:
Marking as resolved for Release v2.4.4

Template related stuff will be added to the wiki since those errors are due to templates not considering non-numeric states, but is not an integration issue.

closed 2023-09-06 by WillCodeForCats