WillCodeForCats/solaredge-modbus-multi #420: Failed to connect

NilsWildt, 2023-08-27:
### Describe the bug

Hi, 
after I had installed the hacs extension, everything was working well. Then  I had a scheduled maintenance-down of my HA. After that, when I restarted everything, I couldn't connect any more to the SolarEdge Coordinator anymore, see logs below -- Since then I tried already:

1. Connect to modbus via other script (perl indeed, it worked and it is working (from the same subnet, not the same container though) 
2. Restart HA, restart Router, check router firewall, deinstall solaredge-modbus-multi, reinstall other version etc. 
3. Check that nothing else connects to same modbus
4. Change modbus port
5. Change timeout in source

nothing above was successful. Anyone else had similar issues and/or has a hint what to try next? 

### Expected behavior

I hoped it would not fail to connect.

### Diagnostic File

[home-assistant_solaredge_modbus_multi_2023-08-27T12-01-27.525Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12447921/home-assistant_solaredge_modbus_multi_2023-08-27T12-01-27.525Z.log)

### Debug logs

```shell
home-assistant   | 2023-08-27 12:21:07.306 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
home-assistant   | 2023-08-27 12:28:54.284 WARNING (MainThread) [pymodbus.logging] Failed to connect 
home-assistant   | 2023-08-27 12:28:54.346 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdgeModbusTCP' for solaredge_modbus_multi integration not ready yet: Modbus/TCP connect to 192.168.178.125:1502 failed.; Retrying in background
home-assistant   | 2023-08-27 12:29:02.534 WARNING (MainThread) [pymodbus.logging] Failed to connect 
home-assistant   | 2023-08-27 12:29:11.729 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=False, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
home-assistant   | 2023-08-27 12:29:11.729 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
home-assistant   | 2023-08-27 12:29:14.732 WARNING (MainThread) [pymodbus.logging] Failed to connect 
home-assistant   | 2023-08-27 12:29:14.740 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.011 seconds (success: False)
home-assistant   | 2023-08-27 12:29:14.740 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdgeModbusTCP' for solaredge_modbus_multi integration not ready yet: Modbus/TCP connect to 192.168.178.125:1502 failed.; Retrying in background
home-assistant   | 2023-08-27 12:29:20.003 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=False, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
home-assistant   | 2023-08-27 12:29:20.003 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
home-assistant   | 2023-08-27 12:29:23.005 WARNING (MainThread) [pymodbus.logging] Failed to connect 
home-assistant   | 2023-08-27 12:29:23.008 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.005 seconds (success: False)
home-assistant   | 2023-08-27 12:29:33.242 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=False, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
home-assistant   | 2023-08-27 12:29:33.242 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
home-assistant   | 2023-08-27 12:29:36.244 WARNING (MainThread) [pymodbus.logging] Failed to connect 
home-assistant   | 2023-08-27 12:29:36.247 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.005 seconds (success: False)
home-assistant   | 2023-08-27 12:29:56.599 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=False, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
home-assistant   | 2023-08-27 12:29:56.600 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
home-assistant   | 2023-08-27 12:29:59.603 WARNING (MainThread) [pymodbus.logging] Failed to connect 
home-assistant   | 2023-08-27 12:29:59.606 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.006 seconds (success: False)
home-assistant   | 2023-08-27 12:30:39.914 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=False, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
home-assistant   | 2023-08-27 12:30:39.915 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
```

### Home Assistant Version

Home Assistant 2023.8.4

### solaredge-modbus-multi Version

v2.4.3-pre.7

### Installation Type

Container

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

WillCodeForCats, 2023-08-27:
I see people reporting connections problems and then they disappear.

I unfortunately don't have any suggestions to offer.

NilsWildt, 2023-08-27:
Thank you for you answer, good to know. I also didn't see anything weird in the code. 

But then I'll be waiting for the  to disappear and debug a little bit more.  Best chance is maybe to debug inside my container, just to make sure that's not the problem.

closed 2023-08-27 by NilsWildt

WillCodeForCats, 2023-08-27:
You can try manually installing the pymodbus 3.4.1 pinned branch:

https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/pymodbus_3_4_1.zip

Home Assistant currently has pymodbus 3.3.1 but I would I expect to see pymodbus 3.4.1 in HA 2023.9 since there was a recent PR in HA core for that.

I just call the pymodbus connection function, the integration doesn't really have anything to do with connection errors.