WillCodeForCats/solaredge-modbus-multi #427: No modbus connection anymore after update to 2.4.4

apf-doit, 2023-09-08:
### Describe the bug

I just updated to the latest version 2.4.4 and since then the integration can not communicate to the inverter anymore. It was just running fine before the update. I double check the TCP Modbus IP and Port on the inverter - nothing has changed, all enabled.

### Expected behavior

I expected the integration to just work like before the update.

### Diagnostic File

i don't have a diagnostic file.

### Debug logs

```shell
2023-09-08 21:00:32.708 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.10.40', 1502)
2023-09-08 21:00:43.459 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=True, detect_extras=True, keep_modbus_open=True, adv_storage_control=True, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
2023-09-08 21:00:43.460 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-09-08 21:00:43.464 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.10.40', 1502)
2023-09-08 21:00:43.467 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.008 seconds (success: False)
2023-09-08 21:00:43.468 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SE10K-RWB48' for solaredge_modbus_multi integration not ready yet: Modbus/TCP connect to 192.168.10.40:1502 failed.; Retrying in background
```

### Home Assistant Version

2023.8.2

### solaredge-modbus-multi Version

2.4.4

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2023-09-08:
Can't really help with connection errors; see other closed issues for more information.

WillCodeForCats, 2023-09-08:
Rebooting (or power cycling) the system Home Assistant runs on should be the first step with connection issues. This must be a true reboot of whatever OS is being used to clear any hung network connections, not a core-only restart. Second would be to power cycle the inverter with the modbus/tcp connection in case it is blocking longer than the 2 minute max it's supposed to have.

If any wireless is involved it's widely reported that wifi and solaredge modbus has various issues. Anyone is welcome to try and work with wifi of course, but I don't want to see issue reports about wifi. My inverters and HA system are all hardwired so I don't have wifi to test with.

Generally speaking, because I use my own integration, I would never actually make a release that fails to connect to my inverters so it's more likely connection failures are a local problem not an integration level problem. Since it's connecting fine on my own systems I have to close this as unable to reproduce.

If you are having network issues the discussions section or Home Assistant forums are more appropriate.

apf-doit, 2023-09-08:
I will close the issue. I updated HA to latest version and it works again.

closed 2023-09-08 by apf-doit