WillCodeForCats/solaredge-modbus-multi #429: Error after update to 2023.09.01

djfanatix, 2023-09-09:
### Describe the bug

I updated to HA 2023.09.01 and after the update solaredge modbus doesn’t work anymore.

see log below

### Expected behavior

Work

### Diagnostic File

`2023-09-09 07:30:40.515 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=True, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=True, sleep_after_write=0, battery_rating_adjust=0, 
2023-09-09 07:30:40.515 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-09-09 07:30:40.540 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: Big
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
    await self._async_init_solaredge()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 202, in _async_init_solaredge
    await new_inverter.init_device()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 727, in init_device
    inverter_data.registers, byteorder=Endian.Big
                                       ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
2023-09-09 07:30:40.543 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.028 seconds (success: False)
2023-09-09 07:30:40.543 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: Big; Retrying in background
2023-09-09 07:30:45.884 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=True, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=True, sleep_after_write=0, battery_rating_adjust=0, 
2023-09-09 07:30:45.885 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2023-09-09 07:30:45.912 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: Big
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
    await self._async_init_solaredge()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 202, in _async_init_solaredge
    await new_inverter.init_device()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 727, in init_device
    inverter_data.registers, byteorder=Endian.Big
                                       ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
2023-09-09 07:30:45.919 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.035 seconds (success: False)
`

### Home Assistant Version

2023.09.01

### solaredge-modbus-multi Version

Latest 2.4.4

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

dolenec, 2023-09-09:
Same for me..

nickischenker, 2023-09-09:
I can confirm the statement of dolenec. I have the same incident and came to the same conclusion.

citoki, 2023-09-09:
I also tried to use the latest beta version `2.4.5-pre.2` where PR #428 should solve it, as far as I see, but this hasn't worked.

Uhae, 2023-09-09:
Same problem here. Will try to roll back to 2023.9.0 
Ist there a „best“ way to downgrade?

dolenec, 2023-09-09:
simply use in CLI

ha core update --version=2023.9.0

citoki, 2023-09-09:
> Same problem here. Will try to roll back to 2023.9.0
> Ist there a „best“ way to downgrade?

HA >> Settings >> System >> Backup: choose the latest `core_2023.*` item

nickischenker, 2023-09-09:
That's it, and stay relaxed. Such a downloade lasts easily times ne hour or maybe longer.

nickischenker, 2023-09-09:
Exactly like this ^ and stay relaxed. Such a downgrade easily takes an hour or maybe longer.

Uhae, 2023-09-09:
Downgrade completed and all working again. Thanks for the help!

acuarty, 2023-09-09:
Same here. Restore of core2023.9.0. First time ever had to roll back, surprised it took so long. Maybe 45 mins or so.

PoelieVortex, 2023-09-09:
I had the same problem.
Everything is fine after a downgrade

purcell-lab, 2023-09-09:
I'm seeing a networking issue with _HA 2023.9.1_ and _pymodbus 3.5.1_ which causes networking to hang, when enabling _modbus-multi v2.4.5-pre.2_

```
mark@odroid:~$ telnet 192.168.86.75 1502
Trying 192.168.86.75...
Connected to 192.168.86.75.
Escape character is '^]'.
^[^]
telnet> quit
Connection closed.

# Enable modbus-multi custom integration
mark@odroid:~$ telnet 192.168.86.75 1502
Trying 192.168.86.75...
telnet: Unable to connect to remote host: Connection refused
```

home-assistant.log:
```logs
2023-09-09 20:19:04.153 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: Big
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
    await self._async_init_solaredge()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 202, in _async_init_solaredge
    await new_inverter.init_device()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 727, in init_device
    inverter_data.registers, byteorder=Endian.Big
                                       ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
2023-09-09 20:19:04.160 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: Big; Retrying in background
2023-09-09 20:19:09.233 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.86.75', 1502)
2023-09-09 20:19:19.626 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.86.75', 1502)
```
```
# Disable modbus-multi
mark@odroid:~$ telnet 192.168.86.75 1502
Trying 192.168.86.75...
telnet: Unable to connect to remote host: Connection refused
```

```
# Reboot host system
mark@odroid:~$ telnet 192.168.86.75 1502
Trying 192.168.86.75...
Connected to 192.168.86.75.
Escape character is '^]'.
```

mantaalex, 2023-09-09:
> I'm seeing a networking issue with _HA 2023.9.1_ and _pymodbus 3.5.1_ which causes networking to hang, when enabling _modbus-multi v2.4.5-pre.2_
> 
> ```
> mark@odroid:~$ telnet 192.168.86.75 1502
> Trying 192.168.86.75...
> Connected to 192.168.86.75.
> Escape character is '^]'.
> ^[^]
> telnet> quit
> Connection closed.
> 
> # Enable modbus-multi custom integration
> mark@odroid:~$ telnet 192.168.86.75 1502
> Trying 192.168.86.75...
> telnet: Unable to connect to remote host: Connection refused
> ```
> 
> home-assistant.log:
> 
> ```
> 2023-09-09 20:19:04.153 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: Big
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
>     self.data = await self._async_update_data()
>                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
>     return await self._refresh_modbus_data_with_retry(
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
>     raise ex
>   File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
>     return await self._hub.async_refresh_modbus_data()
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
>     await self._async_init_solaredge()
>   File "/config/custom_components/solaredge_modbus_multi/hub.py", line 202, in _async_init_solaredge
>     await new_inverter.init_device()
>   File "/config/custom_components/solaredge_modbus_multi/hub.py", line 727, in init_device
>     inverter_data.registers, byteorder=Endian.Big
>                                        ^^^^^^^^^^
>   File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
>     raise AttributeError(name) from None
> AttributeError: Big
> 2023-09-09 20:19:04.160 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: Big; Retrying in background
> 2023-09-09 20:19:09.233 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.86.75', 1502)
> 2023-09-09 20:19:19.626 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.86.75', 1502)
> ```
> 
> ```
> # Disable modbus-multi
> mark@odroid:~$ telnet 192.168.86.75 1502
> Trying 192.168.86.75...
> telnet: Unable to connect to remote host: Connection refused
> ```
> 
> ```
> # Reboot host system
> mark@odroid:~$ telnet 192.168.86.75 1502
> Trying 192.168.86.75...
> Connected to 192.168.86.75.
> Escape character is '^]'.
> ```

i'm having serious problems with solaredge modbus even with 2023.09.0 i guess it could be something like you said with networking

mantaalex, 2023-09-09:
its strange:

root@proxmox:~# nmap 192.168.2.154
Starting Nmap 7.80 ( https://nmap.org ) at 2023-09-09 12:59 CEST
Nmap scan report for 192.168.2.154
Host is up (0.0023s latency).
Not shown: 998 closed ports
PORT     STATE    SERVICE
80/tcp   filtered http
8080/tcp filtered http-proxy
MAC Address: 84:D6:C5:11:82:65 (SolarEdge Technologies)

Nmap done: 1 IP address (1 host up) scanned in 1.45 seconds
root@proxmox:~#

purcell-lab, 2023-09-09:
> i'm having serious problems with solaredge modbus even with 2023.09.0 i guess it could be something like you said with networking

I had to roll back to HA 9.0 and modus multi 2.4.5-pre.1 to get a working combination.

dolenec, 2023-09-09:
Current solution is to go back on 2023.8.4

shoei600, 2023-09-09:
Same for me too. Now rolled back and everything working fine.

citoki, 2023-09-09:
> I had to roll back to HA 9.0 and modus multi 2.4.5-pre.1 to get a working combination.

can confirm, also had issues and tried a lot combinations, including restarting my inverters. HA 9.0 with `2.4.5-pre.1` works. Older or newer version didn't work.

djfanatix, 2023-09-09:
I also tried downgrading Pymodbus to 3.4.1, but without success

mantaalex, 2023-09-09:
they just released a new beta for modbus 2.4.5-pre.2 should be with a fix for 9.1

doctawaald, 2023-09-09:
> > I had to roll back to HA 9.0 and modus multi 2.4.5-pre.1 to get a working combination.
> 
> can confirm, also had issues and tried a lot combinations, including restarting my inverters. HA 9.0 with `2.4.5-pre.1` works. Older or newer version didn't work.

Can confirm that this is the only version that works for me.
The newest 2.4.5-pre.2 did not fix the issue.

WillCodeForCats, 2023-09-09:
Pymodbus changed constant inheritance in 3.5.1. At the time HA 2023.9.1 was in process of release but it hadn't actually released by the time I went to sleep.

I'm updating it now and will release 2.4.5-pre.3 shortly.

WillCodeForCats, 2023-09-09:
[Release v2.4.5-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.5-pre.4)

closed 2023-09-09 by WillCodeForCats

WillCodeForCats, 2023-09-09:
[Release v2.4.5](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.5)

seanw18, 2023-09-09:
Thanks @WillCodeForCats, very quick turn around and it's working, cheers!

nickischenker, 2023-09-09:
I can confirm that this works

HA 2023.9.1 and Release v2.4.5

Clean and fast work that works. Thank you VERY! much.

Uhae, 2023-09-10:
Thank you @WillCodeForCats! Excellent work!

mantaalex, 2023-09-11:
thanks you for the fix