WillCodeForCats/solaredge-modbus-multi #430: Anomalous Reading Causing Large Spikes in Data

RobinXe, 2023-09-09:
**Describe the current behavior**
On occasion (seeming, but unconfirmed, relation to changing the inverter storage max charge value) the 'i1 AC Power' value has spiked to negative MW/GW orders of magnitude for 1-2 update cycles.

**What should be updated or changed?**
I believe this value should always be >= 0, at least for single-phase implementations. Would it be possible to please incorporate validation of the value to max(value, 0)?

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.9.1
 - solaredge-modbus-multi Version: 2.4.4

**Additional context**
I fully appreciate that this is very unlikely to be an issue with the integration, and more likely that it is simply reporting spurious data passed from the inverter. I don't wish to sound demanding, but am hopeful that it's an easy fix. My excuse for not fixing it myself is not fully understanding its implications for the ACPower class, and any 3-phase implementations.

WillCodeForCats, 2023-09-09:
Negative AC power is a legitimate reading for systems with batteries, so it cannot be assumed to be 0 or higher.

Upload a diagnostic file (or debug log) showing the problem at the time it appears so I can see the actual values.

WillCodeForCats, 2023-09-09:
Have you tried setting the Inverter Command Delay to a non-zero delay?

[image]

RobinXe, 2023-09-10:
> Negative AC power is a legitimate reading for systems with batteries, so it cannot be assumed to be 0 or higher.
> 
> Upload a diagnostic file (or debug log) showing the problem at the time it appears so I can see the actual values.

Ah yes, of course, I hadn't considered the case of charging the battery from the grid, as I haven't needed to yet. I'm sure I will, going into the winter! My apologies!

RobinXe, 2023-09-10:
> Have you tried setting the Inverter Command Delay to a non-zero delay?
> 

I have not, but I'll give it a go. Is that the intended application for the delay option then? Are inverters known to sometimes send bad data without it?

WillCodeForCats, 2023-09-10:
That's one of the uses for it, yes.

RobinXe, 2023-09-15:
I haven't had a recurrence of the issue, so I haven't played with the command delay, but I did increase my polling interval, which may or may not have had an effect.

Clearly this issue isn't a big enough problem, nor a simple enough fix, to warrant any further attention at this point. Thanks very much for taking the time!

closed 2023-09-15 by RobinXe

RobinXe, 2023-09-15:
Thanks for the further info. I'll definitely try playing with the various settings if the problem recurs. The only real effect is my data integrity and the auto-scaling of my pretty charts, so I can add my own validation between the output and my data store if it does become an intolerable issue.

I'm under no doubt that it's not an issue with the integration, and it's only my ignorance of other use-cases that led me to think it might be something the integration could address in a straightforward, one-size-fits-all, manner.

On re-reading my previous comment I realise that my 'clearly' could be taken as me being huffy at not having my individual problems made the center of the universe, and I'm certainly not! I probably should have written 'I don't believe that this is a big enough problem...'; my apologies if it came across as anything else! I'm very grateful for all your efforts in creating, and maintaining, this integration.