WillCodeForCats/solaredge-modbus-multi #439: HomeAssistant 2023.9.2 breaks integration

Spotlight-Kid, 2023-09-14:
### Describe the bug

I just updated from HA 2023.9.0 to 2023.9.2 and Modbus completely failed. I got an error immediately saying the configuration failed and then continual connection errors. Sorry, but I didn't take any screen shots or details but immediately rolled back the update and now everything works again.

### Expected behavior

Everything to continue working as before.

### Diagnostic File

Sorry, but I didn't get the file before restoring.

### Home Assistant Version

2023.9.2

### solaredge-modbus-multi Version

Whatever is the latest version. How do you find the number?

### Installation Type

Core

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

Spotlight-Kid, 2023-09-14:
I just tried up[dating to 2023.9.1 and exactly the same thing happened. I've attached the current diagnostics file but this is after restoring to 2023.9.0.
[config_entry-solaredge_modbus_multi-ebea85c5e5650c9818f021e8d73a4100.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12609068/config_entry-solaredge_modbus_multi-ebea85c5e5650c9818f021e8d73a4100.json.txt)

Spotlight-Kid, 2023-09-14:
I reinstalled 2023.9.1 with debug mode set on and here is the log file
[home-assistant_solaredge_modbus_multi_2023-09-14T13-33-34.528Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12609484/home-assistant_solaredge_modbus_multi_2023-09-14T13-33-34.528Z.log)

Spotlight-Kid, 2023-09-14:
The latest version available through HACS seems to be 2.4.4 though

Spotlight-Kid, 2023-09-14:
Does 2.4.5 only appear after upgrading to HA 2023.9.1?

WillCodeForCats, 2023-09-14:
> Does 2.4.5 only appear after upgrading to HA 2023.9.1?

Yes that's how HACS works.

https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084/438?u=willcodeforcats

Spotlight-Kid, 2023-09-14:
Okay. Thanks. Sorry for bothering you with my ineptitude!