WillCodeForCats/solaredge-modbus-multi #441: Storage charge limit to a new value

zh17home, 2023-09-16:
**Describe the current behavior**
A clear and concise description of what the feature should be.

Please update the storage charge limit to 8000.

**What should be updated or changed?**
A clear and concise description of the proposed new or updated behavior.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.9.2
 - solaredge-modbus-multi Version: last

**Additional context**
Add any other context about the request here.

After the upgrade to 2023.9.2, I saw the multi set to 6300 to charge the three battery's. I try to chante to 8000 but you was deny. The last value is 5000. Can you please change to 8000?

Pictures are attached.

zh17home, 2023-09-16:
Please close the topic is working well after a reload :)

WillCodeForCats, 2023-09-16:
The battery charge limit value is 50,000 (50kW).

closed 2023-09-16 by WillCodeForCats