WillCodeForCats/solaredge-modbus-multi #442: Home Assistant Update to 2023.9.1 o 2023.9.2 breaks Solaredge-Modbus-Multi

alexander-th, 2023-09-16:
### Describe the bug

My setup is as follows:

- Solaredge StoreEdge inverter with BYD battery and SolarEdge power meter. 
- Homeassistant version 2023.9.0 
- EVCC version 0.119.5 and/or 0.120.0 installed as Homeassistant Addon, EVCC connected to Solaredge inverter via modbus and acting as a modbus proxy
- Solaredge-Modbus-Multi installed via HACS connected to EVCC modbus proxy
- 3 Solaredge-Modbus-Multi devices configured: Inverter, battery, power meeter

=> Everything working perfect

When upgrading Homeassistant to 2023.9.1 or to 2023.9.2 Solaredge-Modbus-Multi stopps working and configured devices no longer show data. 

The protocol shows the following information:

Logger: custom_components.solaredge_modbus_multi
Source: custom_components/solaredge_modbus_multi/hub.py:727
Integration: SolarEdge Modbus Multi (documentation, issues)
First occurred: 19:58:41 (22 occurrences)
Last logged: 20:11:43

``
Unexpected error fetching SolarEdge Coordinator data: Big
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
    await self._async_init_solaredge()
  File "/config/custom_components/solaredge_modb
[config_entry-solaredge_modbus_multi-19d6a60580bbb5d6e3639b749b652145.json (1).txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12641139/config_entry-solaredge_modbus_multi-19d6a60580bbb5d6e3639b749b652145.json.1.txt)
us_multi/hub.py", line 202, in _async_init_solaredge
    await new_inverter.init_device()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 727, in init_device
    inverter_data.registers, byteorder=Endian.Big
                                       ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
``

Installing Homeassistant version 2023.9.0 from backup => everything works again

The configuraiton file when it is working is attached.

### Expected behavior

Update to 2023.9.1 or higher does not break Solaredge-Modbus-Multi

### Screenshots

[image]

### Diagnostic File

[home-assistant_2023-09-16T16-27-00.059Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/12641148/home-assistant_2023-09-16T16-27-00.059Z.log)

### Debug logs

```shell
I an not a computer expert but I am happy to help. If you like me to provide more detailed information please let me know, though I will need some detailed instructions :=)
```

### Home Assistant Version

2023.9.2

### solaredge-modbus-multi Version

2.4.4

### Installation Type

Supervised

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

alexander-th, 2023-09-16:
... and 5 minutes later I noticed that there is a new version 2.4.5. of Solaredge-modbus-Multi

Upgrade to Solaredge-modbus-Multi 2.4.5 => everything works

WillCodeForCats, 2023-09-16:
2.4.4 does not work with HA 2023.9.1

closed 2023-09-16 by WillCodeForCats