WillCodeForCats/solaredge-modbus-multi #454: SolarEdge Modbus Multi interfering with other modbus integrations

huub66, 2023-10-12:
### Describe the bug

Hi, facing somewhat strange issue, perhaps a noob question; when I am installing the SolarEdge Modbus Multi integration and reboot, another modbus integration (Alfen Modbus component, HACS integration as well) is failing Modbus communication and fails. The SolarEdge Modbus Multi integration is working properly. When I remove the SolarEdge Modbus Multi integration, this Alfen Modbus integration works properly and Modbus communication is OK. 
Both integrations work on port 502 but to different IP addresses, my assumption is that this could co-exist due to different IP integration. (Correct?)

### Expected behavior

Multiple integrations using Modbus should work in parallel

### Diagnostic File

Diagnostic file for SolarEdge Modbus Multi not relevant as this works properly, general modbus diagnostics on request if required

### Home Assistant Version

2023.10.1

### solaredge-modbus-multi Version

2.4.6

### Installation Type

Core

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2023-10-12:
What is the link to "Alfen Modbus"? I don't see that in HACS.

[image]

huub66, 2023-10-12:
> What is the link to "Alfen Modbus"? I don't see that in HACS.
> [image]

https://github.com/ThaStealth/alfen_modbus

WillCodeForCats, 2023-10-12:
In this file:

https://github.com/ThaStealth/alfen_modbus/blob/main/custom_components/alfen_modbus/manifest.json

An older, incompatible version of pymodbus (pymodbus==3.1.1) is set as a requirement for that integration.

You will need to open an issue with that integration to get their version requirements to a current version of pymodbus, or do not load `alfen_modbus` since it requires a conflicting modbus version.

WillCodeForCats, 2023-10-12:
This integration requires pymodbus 3.5.1 or newer and Home Assistant core currently uses 3.5.2.

huub66, 2023-10-12:
> In this file:
> 
> https://github.com/ThaStealth/alfen_modbus/blob/main/custom_components/alfen_modbus/manifest.json
> 
> An older, incompatible version of pymodbus (pymodbus==3.1.1) is set as a requirement for that integration.
> 
> You will need to open an issue with that integration to get their version requirements to a current version of pymodbus, or do not load `alfen_modbus` since it requires a pymodbus version that is simply too old.

OK clear thanks! Just out of curiosity, how would that old version break functionality if 2 integrations are using different versions of pymodbus at the same time?

WillCodeForCats, 2023-10-12:
Home Assistant can't have two versions of pymodbus installed at the same time. One of them will take priority when it reads and loads requirements.

closed 2023-10-12 by WillCodeForCats