WillCodeForCats/solaredge-modbus-multi #460: Handle invalid inverter ID at setup

WillCodeForCats, 2023-10-25:
### Describe the bug

Handle condition where an invalid inverter ID was configured when adding the integration.

### Expected behavior

Change to a more user friendly response.

### Diagnostic File

N/A

### Debug logs

```shell
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:290
Integration: SolarEdge Modbus Multi (documentation, issues)
First occurred: 10:22:27 AM (2 occurrences)
Last logged: 10:22:45 AM

Unexpected error fetching SolarEdge Coordinator data: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
    await self._async_init_solaredge()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 202, in _async_init_solaredge
    await new_inverter.init_device()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 732, in init_device
    inverter_data = await self.hub.modbus_read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 478, in modbus_read_holding_registers
    result = await self._client.read_holding_registers(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 207, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
```

### Home Assistant Version

2023.10.5

### solaredge-modbus-multi Version

2.4.7-pre.2

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2023-10-25:
This is not a critical problem (you can just delete and re-add the integration) but there are probably ways it can be better.

Hans-Solo-Dolo, 2025-03-08:
> This is not a critical problem (you can just delete and re-add the integration) but there are probably ways it can be better.

Thank you, sometimes the simplest solutions are the best ones. Had this same problem pop up randomly last night and trying to solve it, this did it for me.

Hans-Solo-Dolo, 2025-03-08:
> This is not a critical problem (you can just delete and re-add the integration) but there are probably ways it can be better.

Thank you, sometimes the simplest solutions are the best ones. Had this same problem pop up randomly last night and trying to solve it, this did it for me. I noticed as soon as I enabled "auto-detect new entities" the same error came back

closed 2026-08-16 by WillCodeForCats