WillCodeForCats/solaredge-modbus-multi #461: Unable to set Site Limit with 2.4.7-pre - Type Error

purcell-lab, 2023-10-30:
### Describe the bug

https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084/509?u=markpurcell

With version 2.4.7-pre, I am unable to set the Site Limit entity.

[image]

### Expected behavior

With version 2.4.6 and earlier I can set the Site Limit entity.

### Screenshots

[image]

[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-0e4f1fd185e47f73b9a07d4577da5b46.json (1).txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/13199676/config_entry-solaredge_modbus_multi-0e4f1fd185e47f73b9a07d4577da5b46.json.1.txt)

### Debug logs

```shell
2023-10-30 10:31:55.112 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 233, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 389, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 172, in async_update_listeners
    update_callback()
  File "/config/custom_components/solaredge_modbus_multi/number.py", line 101, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 743, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 843, in _async_write_ha_state
    state, attr = self._async_generate_attributes()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 783, in _async_generate_attributes
    available = self.available  # only call self.available once per update cycle
                ^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/number.py", line 534, in available
    or self._platform.decoded_model["I_Power_Limit"] > 100
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'str' and 'int'
```

### Home Assistant Version

2023.11.0b2

### solaredge-modbus-multi Version

2.4.7-pre

### Installation Type

Supervised

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

I use site limit control daily to manage negative prices and it is a great feature:

When I have a negative feed in price, I set Export Control mode and site limit to 0W, which prevents my system from exporting power to the grid, but it varies solar production to cover my household load.

When I have a negative general price, I set Production Control mode and site limit to 0W, which switches off all solar production and my house load runs 100% from the grid.

[image]

WillCodeForCats, 2023-10-30:
Looks like just a typing mismatch, I'll review everything to make sure data types are consistent.

WillCodeForCats, 2023-11-01:
In the meantime you can just use 2.4.7-pre.1 if you want the pymodbus version bump without going to HA 2023.11.0. I never run the .0 releases personally since I'd rather wait for someone else to encounter major release issues first that get fixed in the .1 or .2 release. I do load betas and .0 on my dev system to make sure the integration still loads, just not on my live home system.

2.4.7-pre.2 doesn't contain any improvements over pre.1, it's just a bunch of refactoring to move the property where validation checks are happening.

If you install HA 2023.11.0 or newer then you will get the pymodbus version bump from HA core. Eventually when 2.4.7 goes final the min HA version will be bumped to 2023.11.0.

WillCodeForCats, 2023-11-12:
I had added additional debug logging to log the object type to see if I could find this issue, but I haven't been able to see it personally so I'm not exactly sure how or why it happened.

What I could probably do is cast all reads to their expected type (int, float) but otherwise I don't have an explanation for how `I_Power_Limit` became a string.

WillCodeForCats, 2023-11-12:
I will make the object type in debug logs a permanent addition (PR #465).

WillCodeForCats, 2023-11-12:
Actually I think what I'll try for now is to catch TypeError in the available property.

purcell-lab, 2023-11-14:
Thanks, this issue is resolved for me with pre3.

purcell-lab, 2023-11-15:
Looks like this issue is still present with pre3 ;-(

WillCodeForCats, 2023-11-15:
I can't duplicate this, what is the full error message?

purcell-lab, 2023-11-15:
```logs
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/solaredge_modbus_multi/helpers.py:17
Integration: SolarEdge Modbus Multi (documentation, issues)
First occurred: 2:14:24 PM (1 occurrences)
Last logged: 2:14:24 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 233, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 389, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 172, in async_update_listeners
    update_callback()
  File "/config/custom_components/solaredge_modbus_multi/number.py", line 101, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 743, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 843, in _async_write_ha_state
    state, attr = self._async_generate_attributes()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 783, in _async_generate_attributes
    available = self.available  # only call self.available once per update cycle
                ^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/number.py", line 586, in available
    float_to_hex(self._platform.decoded_model["I_CosPhi"])
  File "/config/custom_components/solaredge_modbus_multi/helpers.py", line 17, in float_to_hex
    return hex(struct.unpack("<I", struct.pack("<f", f))[0])
                                   ^^^^^^^^^^^^^^^^^^^^
struct.error: required argument is not a float
```

WillCodeForCats, 2023-11-15:
Yeah that's something else different.

What that means is `I_CosPhi 0x3f800000 <class 'float'>` is not a float type, but in the debug logs it's clearly a float type so it was decoded as something else at that moment in time but it's a temporary thing.

purcell-lab, 2023-11-15:
> Yeah that's something else different.
> 
> What that means is `I_CosPhi 0x3f800000 <class 'float'>` is not a float type, but in the debug logs it's clearly a float type so it was decoded as something else at that moment in time but it's a temporary thing.

That was the last error message, but I agree it is probable unrelated.

Just rolling back to pre1 now.

WillCodeForCats, 2023-11-15:
It's related but what's happening is the debug log was not exactly at the same time as the error. I_CosPhi was something other than class float to trigger the error.

In any case I can just raise ValueError from struct.error in the helper function to deal with it.

WillCodeForCats, 2023-11-15:
Oops I meant TypeError, adding in PR #468 and will make new pre release as soon as checks are done.

thargy, 2023-11-17:
I also can't set Site limit in pre-3 or pre-4.

WillCodeForCats, 2023-11-17:
> I also can't set Site limit in pre-3 or pre-4.

There are two site limit entities. Which one?

Upload your diagnostic file.

thargy, 2023-11-17:
I can't change `number.solaredge_i1_site_limit`, even when  `select.solaredge_i1_limit_control_mode` is set to `Export Control (Export/Import Meter)`.

thargy, 2023-11-17:
Some logs whilst I changed from 'Production' to 'Export Control'. Note the `Site Limit` remained unavailable.

```
2023-11-17 16:08:18.959 ERROR (MainThread) [custom_components.solaredge_modbus_multi.hub] Connection failed: Modbus Error: [Connection] Connection lost during request
2023-11-17 16:08:18.959 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139639052123328] Connection to inverter ID 1 failed.
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 525, in write_registers
    result = await self._client.write_registers(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 199, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 106, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/solaredge_modbus_multi/select.py", line 340, in async_select_option
    await self._platform.write_registers(address=57344, payload=set_bits)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 1231, in write_registers
    await self.hub.write_registers(self.inverter_unit_id, address, payload)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 551, in write_registers
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Connection to inverter ID 1 failed.
2023-11-17 16:11:05.713 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/protocol.py", line 1302, in close_connection
    await self.transfer_data_task
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/protocol.py", line 959, in transfer_data
    message = await self.read_message()
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/protocol.py", line 1029, in read_message
    frame = await self.read_data_frame(max_size=self.max_size)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/protocol.py", line 1104, in read_data_frame
    frame = await self.read_frame(max_size)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/protocol.py", line 1161, in read_frame
    frame = await Frame.read(
            ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/framing.py", line 68, in read
    data = await reader(2)
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 732, in readexactly
    await self._wait_for_data('readexactly')
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 525, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError
[... 510 lines cut ...]
2023-11-17 16:12:40.739 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Import_Energy_WH 0xe06 <class 'int'>
2023-11-17 16:12:40.739 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Energy_Max 0x46179000 <class 'float'>
2023-11-17 16:12:40.739 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Energy_Available 0x461bb333 <class 'float'>
2023-11-17 16:12:40.739 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_SOH 0x42c80000 <class 'float'>
2023-11-17 16:12:40.739 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_SOE 0x42096548 <class 'float'>
2023-11-17 16:12:40.740 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Status 0x4 <class 'int'>
2023-11-17 16:12:40.740 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Status_Vendor 0x2 <class 'int'>
2023-11-17 16:12:40.740 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Event_Log1 0x0 <class 'int'>
[... cut ...]
2023-11-17 16:12:40.741 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Event_Log_Vendor6 0x0 <class 'int'>
2023-11-17 16:12:40.741 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Event_Log_Vendor7 0x0 <class 'int'>
2023-11-17 16:12:40.741 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: B_Event_Log_Vendor8 0x0 <class 'int'>
2023-11-17 16:12:40.741 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 1.553 seconds (success: True)
```

WillCodeForCats, 2023-11-17:
This is a different issue.

For value:
`I2: E_Site_Limit 0xff7fffff <class 'float'>`
The behavior is correct and that value means the site limit setting is unavailable to be changed.

It can't be changed because limit control mode has all bits set to zero:
`I2: E_Lim_Ctl_Mode 0x0 <class 'int'>`
This means limit control is disabled.

When you tried to set limit control mode the inverter dropped the connection before completing the command:
`pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request`

So becasue it was not enabled it is not available.

WillCodeForCats, 2023-11-21:
[pre.6](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.7-pre.6) will probably be the final one before release unless something else comes up related to it.

purcell-lab, 2023-11-21:
With pre.6 Site Limit remains unavailable.

The last version that worked correctly was pre.1 :-(

Happy to provide more debug logs.

```logs
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Brisbane",
    "os_name": "Linux",
    "os_version": "5.14.0-odroid-arm64",
    "supervisor": "2023.11.3",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "24.0.5",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "huawei_solar": {
      "version": "1.2.4",
      "requirements": [
        "huawei-solar>=1.2.0,<2.0.0"
      ]
    },
    "googlewifi": {
      "version": "0.1.34",
      "requirements": [
        "googlewifi==0.0.21"
      ]
    },
    "meross_lan": {
      "version": "4.4.0-alpha.3",
      "requirements": []
    },
    "tesla_custom": {
      "version": "3.19.3",
      "requirements": [
        "teslajsonpy==3.9.8"
      ]
    },
    "dyson_cloud": {
      "version": "0.15.0",
      "requirements": [
        "libdyson==0.8.7"
      ]
    },
    "solaredge_modbus_multi": {
      "version": "2.4.7-pre.6",
      "requirements": [
        "pymodbus>=3.5.4"
      ]
    },
    "garmin_connect": {
      "version": "0.2.17",
      "requirements": [
        "garminconnect==0.2.3",
        "tzlocal"
      ]
    },
    "battery_consumption": {
      "version": "1.0.4",
[... 487 lines cut ...]
        "M_VAh_Imported_C": "0x0",
        "M_VAh_SF": "-0x8000",
        "M_varh_Import_Q1": "0x0",
        "M_varh_Import_Q1_A": "0x0",
        "M_varh_Import_Q1_B": "0x0",
        "M_varh_Import_Q1_C": "0x0",
        "M_varh_Import_Q2": "0x0",
        "M_varh_Import_Q2_A": "0x0",
        "M_varh_Import_Q2_B": "0x0",
        "M_varh_Import_Q2_C": "0x0",
        "M_varh_Export_Q3": "0x0",
        "M_varh_Export_Q3_A": "0x0",
        "M_varh_Export_Q3_B": "0x0",
        "M_varh_Export_Q3_C": "0x0",
        "M_varh_Export_Q4": "0x0",
        "M_varh_Export_Q4_A": "0x0",
        "M_varh_Export_Q4_B": "0x0",
        "M_varh_Export_Q4_C": "0x0",
        "M_varh_SF": "-0x8000",
        "M_Events": "0x0"
      }
    }
  }
}
```

WillCodeForCats, 2023-11-21:
I can try just making site limit values less than 0 display as 0 instead of being unavailable. In your logs site limit value is `0xff7fffff` which is floating point number `-3.40282346639e+38`

WillCodeForCats, 2023-11-21:
[pre.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.7-pre.7) overrides site limit values <0 returned by the inverter and treats them as 0.

purcell-lab, 2023-11-21:
Thanks, pre.7 has restored my access to Site Limit, when I enabled a Limit Control Mode.

[Screen recording 2023-11-21 15.18.13.webm](https://github.com/WillCodeForCats/solaredge-modbus-multi/assets/79175134/a16858bd-d06e-4369-9e42-95ab302381f8)

closed 2023-11-21 by purcell-lab