WillCodeForCats/solaredge-modbus-multi #463: SolarEdge_i1_status

MarcBotman, 2023-11-08:
### Describe the bug

On my installation, SolarEdge_i1_status appears as "I_STATUS_SLEEPING" or whatever it is...
It used to be a number 1..8....

### Expected behavior

A number 1..8 or a text

### Screenshots

[image]

### Diagnostic File

xxxxx

### Home Assistant Version

Core 2023.11.1 Supervisor 2023.10.1 Operating System 11.1 Frontend 20231030.1

### solaredge-modbus-multi Version

2.4.6

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2023-11-08:
Use the entity attributes if you want the number.

WillCodeForCats, 2023-11-08:
See attribute `status_value`:

[image]

[image]

closed 2023-11-08 by WillCodeForCats

MarcBotman, 2023-11-08:
Hi,
Did not find that yet! So sorry to have troubled you!
Will be more careful next time.
I just love your integration, cannot live without it!
Kind regards
Marc

WillCodeForCats, 2023-11-08:
If you want to see attribute values on status cards instead of the default value there is an example in the wiki:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Entity-Status-Cards-from-Attributes