WillCodeForCats/solaredge-modbus-multi #475: Infinite "Upgrading Inverter" with reboots until ethernet unplugged

CrazyCoder, 2023-11-17:
### Describe the bug

Device info
SE5000
by SolarEdge
Firmware: 0003.2543

https://www.solaredge.com/us/support/firmware-release-notes/inverters-with-display
Single Phase Inverter with HD-Wave Technology

I noticed HA integration going up and down for 2+ hours and inverter not working.

Went outside and noticed "Upgrading inverter" message on the screen. It was going from 0% to 35–37% and then rebooting to start all over again.

As soon as I unplugged Ethernet cable from the router, the upgrade was completed to 100% and the inverter started working again.

This might be just a coincidence or a temporary issue, but I wanted to document it in case someone else will report the same.
Or, there might be something that affects the upgrade process from this integration.

### Expected behavior

Inverter upgrade completes without rebooting at 35%.

Normally, it starts producing at 7am this time of year, but that day it didn't until the successful upgrade.

### Screenshots

[image]

### Diagnostic File

Didn't capture one :(

### Home Assistant Version

2023.11.2

### solaredge-modbus-multi Version

2.4.6

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2023-11-17:
Sounds like something to contact SolarEdge support about.

CrazyCoder, 2023-11-17:
Will do if it happens again. Hope it was a one-time glitch.

closed 2023-11-28 by WillCodeForCats