WillCodeForCats/solaredge-modbus-multi #478: Meter data unreliable after firmware update

moedi21, 2023-11-27:
### Describe the bug

Recently, my inverter (Home Hub 3 phase, SE8K-RWB48BFN4) has been upgraded to firmware 4.19.36. Since then, the readings of the imported/exported energy counter in the connected backup interface (SE-RGMTR-1D-240C-A) are missing for most part of the day. Also, the energy counters are partly downcounting.
After the update, I saw also a jump in the values, from about 290 kWh to 1311 kWh.

From the log, repeated frequently:
`2023-11-27 06:57:54.380 WARNING (MainThread) [homeassistant.components.utility_meter.sensor] Total Import received an invalid new state from sensor.solaredgemodbus_m1_imported_kwh : unknown`

Additional issue in the log:
`2023-11-27 11:37:01.506 WARNING (MainThread) [homeassistant.components.sensor.recorder] Entity sensor.solaredgemodbus_m1_exported_kwh from integration solaredge_modbus_multi has state class total_increasing, but its state is not strictly increasing. Triggered by state 1311.289 (1311.289) with last_updated set to 2023-11-27T10:22:17.346956+00:00. Please create a bug report at https://github.com/WillCodeForCats/solaredge-modbus-multi/issues`

### Expected behavior

Expected: Energy counters should be available without interruption, and also strictly upcounting

### Screenshots

[image]
[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-6e741e04c7c2d26d5b83e1f2129d8572.json (1).txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/13478915/config_entry-solaredge_modbus_multi-6e741e04c7c2d26d5b83e1f2129d8572.json.1.txt)

### Home Assistant Version

2023.11.3

### solaredge-modbus-multi Version

2.4.6

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

I have tried in parallel to read the data with pure Python and the 'solaredge_modbus' package available in pip. There I get continous readings, but also partly downcounting:
```{'import_energy_active': 1017520}
{'import_energy_active': 1017522}
{'import_energy_active': 1017526}
{'import_energy_active': 1017531}
{'import_energy_active': 1017531}
{'import_energy_active': 1017531}
{'import_energy_active': 1017454}
{'import_energy_active': 1017457}
{'import_energy_active': 1017462}
{'import_energy_active': 1017462}
{'import_energy_active': 1017470}
{'import_energy_active': 1017470}
{'import_energy_active': 1017457}
{'import_energy_active': 1017465}
{'import_energy_active': 1017473}

WillCodeForCats, 2023-11-27:
Contact solaredge support, it shouldn't go backwards.

WillCodeForCats, 2023-11-28:
I'm closing this because it's not an integration bug: if the change that caused the problem was the inverter firmware being updated, then the bug is in the inverter firmware not the integration

Contact solaredge and ask them to downgrade your firmware to an older version that doesn't have this problem.

moedi21, 2023-11-28:
Partly agree - the downcounting is probably caused by the firmware update. I have raised a ticket with SolarEdge support.
However, I also do not get continous readings from your integration, whereas I can get them using solaredge_modbus Python module. Not sure, why.

Anyways - thank you for your great work!

WillCodeForCats, 2023-11-28:
The integration rejects values that are less than the reading before it because Home Assistant expects total increasing.

WillCodeForCats, 2023-11-28:
I will add debug log entries for the error conditions as part of my sensor entity check refactoring that I'm already working on. You can see them anyway if you turn on debug logging, but I can also have it log why it failed a check.

moedi21, 2023-11-28:
Thank you for the explanation, makes sense.

aakso, 2023-11-29:
Just chiming in that this happened to me too last night. Though, according to the history, the issue started before the version was upgraded. My meter is the same inline meter in a home backup interface (SE-RGMTR-1D-240C-A)

Nonetheless, the values don't seem accurate at all anymore. I have a P1 device hooked directly to my meter so it's easy to compare and it almost looks like Solaredge has messed up and are displaying some completely different register values in the meter offsets relating to Wh values. The instant power register values are correct.

aakso, 2023-11-29:
@moedi21 if you are able, can you give an update if/when you get some response? I'm going to approach our representative if this issue can be seen in the mobile app as well.

We already have an outstanding issue open about the inline meter reporting two times the power and energy readings compared to reality. This issue, however, magically fixed itself.

WillCodeForCats, 2023-11-29:
Reopening for visibility.

WillCodeForCats, 2023-11-29:
> The integration rejects values that are less than the reading before it because Home Assistant expects total increasing.

PR #482 will add logging for this condition.

requiemmg, 2023-12-03:
Interestingly, I do have the same invertert as the thread opener @moedi21 and I do not experience any problems; just checked and it seems I have the same firmware version; not aware when it was updated.

[image]
[image]
[image]

I use the Modbus-Proxy addon (as my Wallbox depends on this data, too) and I only have the twice daily "connection refused" errors being logged during the inverters' reset periods.

aakso, 2023-12-03:
@philjohn @requiemmg Interesting, do you guys have the same meter SE-RGMTR-1D-240C-A?

requiemmg, 2023-12-03:
Yes, see Screenshots from both of us

aakso, 2023-12-03:
@requiemmg, oh sorry, missed that.

This is really odd. The difference is striking in my case. I'm exporting the data from Home Assistant to an external DB:
[image]

philjohn, 2023-12-03:
> @philjohn @requiemmg Interesting, do you guys have the same meter SE-RGMTR-1D-240C-A?
> 
> (This is the inline meter that is din rail mounted that apparently is contained in the Home Backup Interface, the one with CT clamps is different)

No, not the din rail one, it's WND-3Y-400-MB which is connected via modbus and uses a CT clamp.

moedi21, 2023-12-04:
I did not receive a response from SE support yet...
Unfortunately I do not have historical data from before the firmware upgrade, so I cannot check if the device attribute changed due to the update.

aakso, 2023-12-04:
@markuswesterberg interestingly, for me it went the other way. Looking at my exported archived data:
```
+-------------------+---------------------------------+
|state              |last_time                        |
+-------------------+---------------------------------+
|SE-RGMTR-1D-240C-A |2023-12-02 10:42:07.121971 +00:00|
|MTR-240-3PC1-D-A-MW|2023-12-04 20:22:21.292327 +00:00|
+-------------------+---------------------------------+
```

Looking closer, the value seems to randomly oscillate between the two with an interval of some days or weeks. Funky stuff.

requiemmg, 2023-12-04:
Also checked, never changed in my available history.still no Problems

aakso, 2023-12-04:
For those who have the inline meter (working or not), what version does your unit report (entity `sensor.solaredge_m1_version`)? Mine:
```
+------+---------------------------------+---------------------------------+
|state |first_time                       |last_time                        |
+------+---------------------------------+---------------------------------+
|1.0.2 |2023-11-16 14:32:31.281945 +00:00|2023-11-29 06:14:46.206316 +00:00|
|1.0.15|2023-12-02 10:52:58.423723 +00:00|2023-12-04 20:32:21.384697 +00:00|
+------+---------------------------------+---------------------------------+
```

For us, the problem started with the version 1.0.15 (though ver 1.0.2 was also faulty, reporting twice the consumption). Looking at Solaredge's release notes, this 1.0.15 doesn't seem to be "supported" by the latest inverter version, hmm.

WillCodeForCats, 2023-12-04:
My meter has a different model number printed on the case than what it shows through the integration, but it's always given the correct meter values (I have a second separate consumption meter to validate readings when I was first developing the integration).

moedi21, 2023-12-05:
@aakso funny things are happening...
So far, the meter version was reported as 0. But as we have heavy snow right now, I switched the inverter and also the backup interface off. After switching back on today, suddenly the meter name changed to the same values as you have, and also the version is now reported. Now, I have
```
Name: MTR-240-3PC1-D-A-MW
Version: 1.0.15

DragonQ, 2023-12-05:
> @aakso funny things are happening... So far, the meter version was reported as 0. But as we have heavy snow right now, I switched the inverter and also the backup interface off. After switching back [...]

My meter went from version 0 to version 31.

markuswesterberg, 2023-12-08:
Anyone had a new go with SolarEdge support about this? It's frustrating that all my HA setups dependent on the imported data are now not working because of this :(

DragonQ, 2023-12-08:
> Anyone had a new go with SolarEdge support about this? It's frustrating that all my HA setups dependent on the imported data are now not working because of this :(

I have a ticket open but was told it could take up to 5 working days to get a response or for my downgrade to be approved, so I have to chase on Monday 11th. Obviously it'd be better if they fixed the issue but given I don't need any of the new features in the new firmware anyway a downgrade is fine for me.

oimara79, 2023-12-08:
> Anyone had a new go with SolarEdge support about this? It's frustrating that all my HA setups dependent on the imported data are now not working because of this :(

I contacted Solaredge via my installer. Now it seems that they are sending a new meter. But I had already another ticket open, because the meter started with 800kWh+ on install.

For now I use `sensor.solar_exported_power_w` that integrates over `sensor.solaredge_m1_ac_power` to display my energy flow.
I got this from the great works of: https://github.com/Remko76/hasolarcfg

victorwelling, 2023-12-09:
Really glad I found this topic. Don't have an answer, but I'll add a datapoint: I too have a SolarEdge 3-phase Home Hub inverter (SE8K-RWB48BFN4, firmware v4.19.36), with a Home Backup Interface (meter reported as MTR-240-3PC1-D-A-MW, firmware v1.0.15).

It was installed November 15th this year, and has been reporting incorrect import/export values since it went online. I first noticed it in the mySolarEdge app when comparing the numbers to those reported by my energy provider (Tibber). When comparing the two, I found my SolarEdge meter underreports by ~30% on average.

The weird thing is, all the other data is reported by correctly and continuously. Through my installer, I have a case with SolarEdge, but they (SolarEdge) are yet to provide a response.

moedi21, 2023-12-09:
@oimara79 This is also my workaround right now. The power values seem to be correct. At least the values make somehow sense, even if I cannot prove it due to lack of references.

I still did not get any response from SE.

DragonQ, 2023-12-14:
Since the downgrade to 4.18.32 the modbus connection on my inverter has been stable for 36 hours, so I think that sorted it. I would advise people ask for a downgrade and avoid 4.19 firmwares for a while.

requiemmg, 2023-12-16:
I wonder why some people have issued and others not. Still on 4.19.36.

My meters firmware value changed from 0 to 78 in the last days. Still getting correct readings....

chofstaetter, 2023-12-17:
I have a 9.2 kWh battery and a "BI-NEUNU3P-01" backup interface installed in addition to my existing SE10K-RWB48BFN4 (FW: 0004.0019.0036) for two days. The SolarEdge Modbus Multi Integration immediately recognized the two new devices, the BI as "MTR-240-3PC1-D-A-MW" with FW: 1.0.15 and the battery as "SolarEdge Home Battery 9.2kWh" FW: 48V DCDC 3.2.9. 
The values from the RWB48 and the batteries look good, but the BI provides partially no or incorrect values: 
- AC Voltage AB, BC, CA and LL are always on "0.00 V",
- All 4 "Exported x kWh" and all "Imported x kWh" are on "Unknown" but sometimes provide single nonsensical values. 
Could it be that the backup interface is simply not supported correctly yet, since the values of the battery and the inverter are correct, or should I really go to an older FW?
[image]
[image]

moedi21, 2023-12-18:
@chofstaetter I have a similar setup as you, and I can just tell you that everything worked ok before the firmware upgrade...

moedi21, 2023-12-18:
Just a brief update - I received a mail regarding my support ticket with SolarEdge. It is now forwarded to 2nd level support...

chofstaetter, 2023-12-18:
> @chofstaetterIch habe ein ähnliches Setup wie Sie und kann Ihnen nur sagen, dass vor dem Firmware-Upgrade alles gut funktioniert hat ...

Perfect, thanks for the information, then I'll contact SE Support and ask them to downgrade the inverter.

WillCodeForCats, 2023-12-19:
> > The integration rejects values that are less than the reading before it because Home Assistant expects total increasing.
> 
> PR #482 will add logging for this condition.

This PR is part of [Release v2.4.8-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.8-pre.1)

victorwelling, 2023-12-22:
Just got an update they're working on a fix. No ETA unfortunately.