WillCodeForCats/solaredge-modbus-multi #479: wrong voltage in Meter

on7nh, 2023-11-28:
### Describe the bug

i see AC voltage AB like -255.61v... when i debus is see valie FAA or 401.0 Volts what scjoud be normal

### Expected behavior

wrong calculation ?

### Screenshots

[image]
[image]

### Diagnostic File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Brussels",
    "os_name": "Linux",
    "os_version": "6.1.59",
    "supervisor": "2023.11.3",
    "host_os": "Home Assistant OS 11.1",
    "docker_version": "24.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "victron": {
      "version": "0.0.11",
      "requirements": [
        "pymodbus>=3.5.1"
      ]
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "solaredge_modbus_multi": {
      "version": "2.4.6",
      "requirements": [
        "pymodbus>=3.5.1"
      ]
    },
    "entsoe": {
      "version": "0.0.1",
      "requirements": [
        "entsoe-py==0.5.8"
      ]
    },
    "buienalarm": {
      "version": "1.9",
      "requirements": [
        "pybuienalarm==0.1.1"
      ]
    },
    "nordpool": {
      "version": "0.0.14",
      "requirements": [
        "nordpool>=0.2",
        "backoff"
[... 655 lines cut ...]
    }
  }
}

### Home Assistant Version

2023.11.1

### solaredge-modbus-multi Version

2.4.6

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

.

WillCodeForCats, 2023-11-28:
The screenshots aren't from the same devices, that's why they don't match.

on7nh, 2023-11-28:
the screenshots are a minute later.

if you look at

"AC_Voltage_AN": "0x8df", is this 2271 or 227.1 volts..this corresponds approximately to 226.36

but if you take "AC_Voltage_AB": "0xfaa", it should be 401 volts...
include some variation in time... but this can never be -255.64 volts...
this number is Negative...

WillCodeForCats, 2024-01-05:
Closing because I can't see or duplicate the issue in the integration.

closed 2024-01-05 by WillCodeForCats