WillCodeForCats/solaredge-modbus-multi #486: DC Isolation Fault Diagnostic Data

ryanm101, 2023-12-28:
[application_note_isolation_fault_troubleshooting.pdf](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/13785296/application_note_isolation_fault_troubleshooting.pdf)

Per attached doc is there a way to get this info from modbus? I'll likely have this fault for the next week or so depending on how long it takes for me to find it / my installer to get someone out.

WillCodeForCats, 2023-12-28:
All known inverter sensors with a modbus register are already implemented.

It's possible there are undocumented registers, but someone would need to find them. For more information see: https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/285

WillCodeForCats, 2023-12-31:
You could just try scanning every possible register and see if anything responds outside of the known ranges.