WillCodeForCats/solaredge-modbus-multi #488: pymodbus is not installed, or pymodbus version is not supported

darkm20, 2023-12-28:
### Describe the bug

While adding the integration (after HACS installation) I get this error in the log:
pymodbus is not installed, or pymodbus version is not supported

### Expected behavior

Start the integration and ask for inverted config

### Diagnostic File

2023-12-28 17:47:33.964 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration solaredge_modbus_multi: pymodbus is not installed, or pymodbus version is not supported

### Home Assistant Version

2023.12.3

### solaredge-modbus-multi Version

latest

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2023-12-28:
Usually this means you are running another custom integration that is locked on an older unsupported version of pymodbus. Upload your full HA log or look for the lines at startup that say "We found a custom integration X".

darkm20, 2023-12-28:
You are totally right. I still had my previous modbus integration installed. Congratulations for the code and features. You can close this issue.

closed 2023-12-28 by darkm20