WillCodeForCats/solaredge-modbus-multi #495: Support for Max-Peak Power sensor of battery

wwerther, 2024-01-08:
**Describe the feature**
It would be great if you could add an additional sensor to the Solaredge Battery that displays the current max charge peak power of the battery. 
I discovered that this value is closely related to the actual temperature of the battery and can be (if it is cold outside) a lot under the normal maximum charge power.
I would like to use that value to check if it may make sense to heat the battery using an electrical heater instead of pumping the power into the grid.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2024.1.2
 - solaredge-modbus-multi Version: 1.10.1

**Additional context**

The IO-Broker implementation supports this feature. Relevant Modbus Address can be seen in this file
https://github.com/joergspoerl/ioBroker.solaredge-modbus/blob/master/src/solaredge/solaredgeModel.ts
line 442

I think it needs to be added to sensors.py and hub.py, but honestly I do not really understand yet how this is all related together

WillCodeForCats, 2024-01-08:
Do you mean these values?

[image]

WillCodeForCats, 2024-01-08:
Device attributes for `batt_charge_peak` and `batt_discharge_peak` will only be displayed if the modbus data returns a value greater than 0.

Turn on the debug logs and look for entries `B_MaxDischargePeakPower` and `B_MaxChargePeakPower` to see if they are non-zero values.

WillCodeForCats, 2024-01-09:
Do you have data from ioBroker that shows these are dynamic values?

The documentation calls them “battery nameplate information” which means to me that nameplate values should never change as if they are printed or attached to the device itself, and why I chose to implement them as attributes of the device.

wwerther, 2024-01-10:
Wow. This is a quick reaction :-)
Yes, this value seems to be dynamic and dependent to the temperature of the battery. In my case the battery is located in the garage and therefore it could get colder than inside the house.
Please find attached a screenshot from IOBroker that shows the variation on that value:
[image]
I will also turn on the debug logs to see if I can find the values there. HomeAssistant is new for me (used IOBroker up to now)...

wwerther, 2024-01-10:
I found it. I think this is the relevant part from homeassistant-debug-log:

      "inverter_unit_id": 1,
      "common": {
        "B_Manufacturer": "SolarEdge",
        "B_Model": "SolarEdge Home Battery 4.6kWh",
        "B_Version": "48V DCDC 3.2.9",
        "B_SerialNumber": "**REDACTED**",
        "B_Device_Address": 112,
        "B_RatedEnergy": 4600.0,
        "B_MaxChargePower": 2825.0,
        "B_MaxDischargePower": 4096.0,
        "B_MaxChargePeakPower": 1400.0,
        "B_MaxDischargePeakPower": 4250.0
      },

WillCodeForCats, 2024-01-10:
Thanks, I don't actually have batteries personally (just inverters and meters) so I rely on feedback for how batteries actually behave. I can program things around what the specs say but they are often different from reality.

Do any of the these other "nameplate" values change over time?

* rated_energy
* charge_continuous_power
* discharge_continuous_power
* discharge_peak_power
* charge_peak_power

wwerther, 2024-01-10:
Hi!

I checked the history from  IOBroker-setup.

battery.charge_continues_power and  discharge_continuous_power is not changing (since July). Same is true for rated_energy, but maybe that goes with the SOH value. We will see in some years.

charge_peak_power is changing (see screenshot above :-)

discharge_peak_power is not recorded at the moment, but it is always higher than the discharge_continues_power value and at least I have the feeling it is constant over the time.

Maybe Discharging a battery is not as temperature dependent than charging a battery.

Regards
Walter

WillCodeForCats, 2024-01-10:
Okay so I'm seeing is that it's possible for the `peak` values to change.

I will get a pre-release out shortly to move those values over to sensors instead of device attributes.

WillCodeForCats, 2024-01-10:
Please try [Release v2.4.9-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.9-pre.1)

closed 2024-01-10 by WillCodeForCats

wwerther, 2024-01-11:
Hurray :-) Working fine! 

[image]

And it is true. Also the discharge peak is changing. But not as much as the charge peak. 

Thank you very much for the prompt response and implementation. I wasn't expecting that. Amazing!