WillCodeForCats/solaredge-modbus-multi #497: Implementation of export controls

ChristophCaina, 2024-01-09:
**Describe the feature**
The Solaredge Modbus Integration provides controlls for the following settings (Screenshot below)

* Export Control Mode
* Export Control Limit Mode
* Export Control Site Limit

This setting allows to adjust the export limit to the grid, which was a regulation by law in germany a few years ago...

--> With the SolarEdge Modbus integration, the Active Power Limit is a dynamic entity which will be set by the inverter itself

**Screenshots**
[image]

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2024.1.2
 - solaredge-modbus-multi Version: 2.4.7

**Additional context**
Add any other context about the request here.

WillCodeForCats, 2024-01-09:
All export controls and storage controls are available.

Can you upload your diagnostic file?

WillCodeForCats, 2024-01-09:
Also make sure you turn on Power Control Options after installing the integration. Commands are disabled by default.

ChristophCaina, 2024-01-09:
Thanks.
That was what I was looking for :)

closed 2024-01-09 by ChristophCaina