WillCodeForCats/solaredge-modbus-multi #516: DC Current is unavailable

bouwew, 2024-01-23:
### Describe the bug

DC Current is shown as unavailable.

### Expected behavior

Show as not unavailable.

### Screenshots

[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-083686173b9116e785360ef33c3dd0d2.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14028704/config_entry-solaredge_modbus_multi-083686173b9116e785360ef33c3dd0d2.json.txt)

### Home Assistant Version

2024.1.5

### solaredge-modbus-multi Version

2.4.8

### Installation Type

Supervised

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-01-23:
From your diagnostic file:

```
"I_DC_Current": "0xffff",
"I_DC_Current_SF": "-0x8000",
```

This means your inverter does not have this data. Try rebooting or contact SolarEdge support.

bouwew, 2024-01-23:
I'm coming from binsentsu's integration, that one does show the DC Current.

bouwew, 2024-01-23:
[image]

WillCodeForCats, 2024-01-23:
Wait to see if they change once the inverter is producing from solar (when DC Power and DC Voltage are not zero).

These entries from the diagnostic file are the ones you pictured:

```
        "I_DC_Current": "0xffff",
        "I_DC_Current_SF": "-0x8000",
        "I_DC_Voltage": "0x0",
        "I_DC_Voltage_SF": "-0x1",
        "I_DC_Power": "0x0",
        "I_DC_Power_SF": "0x0",
```

Voltage and power show zero, current has the "unavailable" state values.

WillCodeForCats, 2024-01-25:
Were you able to check the DC current values when the solar panels were producing?

bouwew, 2024-01-25:
Yes, just before I left for work this morning, I noticed that binsentsu's integration showed non-zero DC-values.
So I removed binsentsu's integration and added yours.
When looking at the DC-current sensor, the value was 0, no longer unavailable.

I will have a look again later today when I'm home again.

Also, I'm curious to see what's displayed when it goes dark outside.

bouwew, 2024-01-25:
Just now (18:49 local NL-time) the value shows 0.
In diagnostics:
```
        "I_DC_Current": "0x0",
        "I_DC_Current_SF": "0x0",
```

bouwew, 2024-01-28:
[image]

WillCodeForCats, 2024-02-19:
Looks OK; closing issue.

closed 2024-02-19 by WillCodeForCats