WillCodeForCats/solaredge-modbus-multi #519: Change IP missing

4n4n4s, 2024-01-24:
### Describe the bug

Hi somehow after setting it up you can't change the IP address anymore of an existing hub.

Not sure what happens when I add an additional hub. Will it duplicate the entities? 

### Expected behavior

Add possibility to update IP address.

### Diagnostic File

-

### Home Assistant Version

2024.1.5

### solaredge-modbus-multi Version

2.4.8

### Installation Type

Container

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-01-24:
If it can't communicate a repair issue will be raised that asks you to verify your information. The IP can be changed there.

4n4n4s, 2024-01-24:
Perfect thanks a lot for that quick answer.

closed 2024-01-24 by 4n4n4s