WillCodeForCats/solaredge-modbus-multi #526: Cannot add second inverter

mtbsteve, 2024-02-05:
### Describe the bug

I do have a setup of 2 inverters, a SE battery and the SE Backup interface. The primary inverter, a SE-10K-RWB48 has the device id 1. The second inverter, a SE-5K, which is connected as as a slave, has the device id 9. I tried all possible configuration entries,  but this integration only finds the RWB48 inverter along with the battery and the meter, but not the slave inverter. The device ids were set by the installer company and I dont want to change them. Is there a way how I can run all inverters with this integration?

### Expected behavior

When entering the number of inverters and starting device id in the config menu, all connected devices should be found.

### Diagnostic File

--

### Home Assistant Version

2024.1.6

### solaredge-modbus-multi Version

2.4.9

### Installation Type

Core

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

deqw, 2024-02-05:
You say your configuration follows  https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

But there is says: **Important: If you have more than one inverter, the inverters must have sequential unit IDs (i.e. 1, 2, 3, ...).**

WillCodeForCats, 2024-02-05:
> The device ids were set by the installer company and I dont want to change them.

If you don't want to use a compatible config for right now, then you'll need to wait then until some future version allows non-sequential inverter IDs.

mtbsteve, 2024-02-05:
@WillCodeForCats non-sequential device ids is a valid Solaredge configuration. As long as my installation is under warranty, I cannot make changes to the setup w/o the risk of loosing warranty coverage from my installer. I hope you can understand this and I would really appreciate if this can be fixed. Thx.

mtbsteve, 2024-02-05:
> Non sequential ID support is already on the project idea list, but there is no timeline for it.

Appreciated :+1:

mtbsteve, 2024-02-06:
A quick and dirty hack to support inverters with non-sequential IDs as interim fix in the file hub.py line 205ff.
A better solution could be to input a comma-separated list of inverter ids in the config menu instead of number_of_inverters and then work down the list in the async_init_solaredge routine.
```
for inverter_index in range(self._number_of_inverters):                                                                               
        # quick hack with hardcoded ids to enable non-sequential inverter device ids                                                          
            if (inverter_index == 0):                                                                                                         
                inverter_unit_id = 1     # add inverter 1 with device id 1                                                                                                    
            elif (inverter_index == 1):                                                                                                       
                inverter_unit_id = 9    # add inverter 2 with device id 9                                                                                                      
        #    inverter_unit_id = inverter_index + self._start_device_id                                                                        
```

WillCodeForCats, 2024-02-06:
Yes I already have a solution, it's just not uploaded.

WillCodeForCats, 2024-02-06:
It will be part of #460 but there are other enhancements I want to get released first. Changing how it handles inverter IDs is is not backwards compatible, so once existing users upgrade it will force a config schema migration internally where they won't be able to downgrade to a previous version. This will be in the 2.5.x releases, it won't be part of the current 2.4.x line.

Menz01, 2024-03-27:
> It will be part of #460 but there are other enhancements I want to get released first. Changing how it handles inverter IDs is is not backwards compatible, so once existing users upgrade it will [...]

Has this been implemented? I have 2 inverters and I could never get this to find my second inverter so I was thinking it may not be a sequential id?

Menz01, 2024-06-24:
> A quick and dirty hack to support inverters with non-sequential IDs as interim fix in the file hub.py line 205ff. A better solution could be to input a comma-separated list of inverter ids in the [...]

does this still work? i think i am having the same issue... i have 2 inverters and it fails whenever i set the integration to 2

Menz01, 2024-07-06:
@WillCodeForCats Hello... i was just curious if this is still on your radar to do?

WillCodeForCats, 2024-07-06:
Yes, I've made a few iterations but wasn't happy with them or how the code looked.

Progress is in #604 and the branch.

WillCodeForCats, 2024-08-23:
[Release v3.0.0-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.0.0-pre.1) is now available for testing.

Menz01, 2024-08-23:
i grabbed it and installed it and at least for me its not working. i set it for 2 inverters and errored retried 3 times and failed. oh well... i guess i have something else wrong.

WillCodeForCats, 2024-08-23:
I need details. Log messages, screenshot.

It works fine for me on my system.

Menz01, 2024-08-23:
i attached the debug log and let me clarify that i am not 100% sure its not something on my system. the integration works fine when i set it for 1 inverter, but will not work when i set it for 2 and i have 2 inverters. so perhaps something in the 2nd one is still not right? 

[home-assistant_solaredge_modbus_multi_2024-08-23T20-23-35.039Z.log](https://github.com/user-attachments/files/16733325/home-assistant_solaredge_modbus_multi_2024-08-23T20-23-35.039Z.log)

WillCodeForCats, 2024-08-23:
In the new version you would configure, for example, inverter ID list `1,2` since there the number of inverters setting was removed. Do you know what IDs your inverters have? Upgrading old configurations would convert the count into a list for this version.

Go to Reconfigure in the menu and you can set the IDs there:

[image]

[image]

WillCodeForCats, 2024-08-23:
You should also remove the `solaredge_modbus` integration, it is not compatible with current versions of Home Assistant and pymodbus.

Menz01, 2024-08-23:
so i need to know the inverter ID numbers? well i do not know them and will probably have to jump through hoops to find them from my service providers :(

Menz01, 2024-08-23:
so i set it up like in your pics and still fails :( 

[image]

[home-assistant_solaredge_modbus_multi_2024-08-23T20-51-44.508Z.log](https://github.com/user-attachments/files/16733567/home-assistant_solaredge_modbus_multi_2024-08-23T20-51-44.508Z.log)

WillCodeForCats, 2024-08-23:
For now, yes, you need to know the inverter IDs. If the first one always worked then we already know the first ID is `1`. If `1,2` doesn't work then the second inverter is not using ID 2.

You can try updating the ID list to find it, such as `1,3` then `1,4` then `1,5` and so on to try and find it.

Sometimes installers might leave follower inverters set to Unit ID 1. You're supposed to change it but it still works this way with the monitoring platform, but will not work for modbus. My inverters were like this after first install and I had to go in and change my second ID.

mtbsteve, 2024-08-25:
@WillCodeForCats thank you! The new non sequential device id feature works perfectly fine for me.
For anyone who doesnt know the device id of their inverters may look into the following procedure:

1. Connect to the inverter access point like you would for a normal network. Scan the QR code on the side of your inverter.
2. Open up a browser and go to http://172.16.0.1/ then click on Device Manager
3. there you can see the current device id of your inverter.

Menz01, 2024-08-25:
so i was able to get into my follower inverter and it was set to 1 so i changed it to 2. i do not know why, but i could not get into my main inverter. i used the same steps but my phone would connect to the wifi but the page would not open. i remember the tech who turned on the modbus said he followed the guide i gave him so, that and this integration worked with "1" i just assumed it was setup correctly. I guess it was as i changed this integration to "1,2" and now i see 2 inverters in HA without errors.

thank you for all the help

closed 2024-08-28 by WillCodeForCats