WillCodeForCats/solaredge-modbus-multi #527: Support for Battery Backup Reserve Setting

mtbsteve, 2024-02-05:
**Describe the feature**
Feature request to support the backup reserve setting of the SE Battery in conjunction with the SE Backup Interface. 
The modbus register address is 97353 and returns the set backup reserve in percent. 
This register is also writable, so you can preset the backup reserve.

**Screenshots**
If applicable, add screenshots or edits of the proposed feature that help describe it.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2024.1.6
 - solaredge-modbus-multi Version: 2.4.9

**Additional context**
Add any other context about the request here.

WillCodeForCats, 2024-02-05:
Do you know the register type and size?

mtbsteve, 2024-02-05:
In my ioBroker modbus implementation I have it as Float, Length 2, unit percent

WillCodeForCats, 2024-02-06:
Are there other registers in this range? How did you find this one?

If there are other related registers in a block with this one I prefer to make sure we are reading a range of addresses in a single request and not single addresses across multiple requests.

mtbsteve, 2024-02-06:
> Are there other registers in this range? How did you find this one?
> 
> If there are other related registers in a block with this one I prefer to make sure we are reading a range of addresses in a single request and not single addresses across multiple requests.

I used this document:
https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/
See page 18, StoreEdge control register settings. 
As per SolarEdge support, the listed register mappings in that document are no longer valid for the new RWB inverters along with the 48V Solaredge batteries. Accordingly, most of the storage control registers did not work for me with the new Battery systems. However the address E008/F708 for the manual Backup Reserve setting works.

WillCodeForCats, 2024-02-06:
I've already implemented the registers in that document.

WillCodeForCats, 2024-02-06:
Upload your diagnostic file if you aren't seeing the backup reserve entity.

mtbsteve, 2024-02-06:
@WillCodeForCats thanks, yes I found the the backup_reserve settings in the meantime. It reads the defined reserve value correctly but I cannot properly change the setting. If I change the slider value, the result  will not be updated in the solaredge app. AFAIK the backup reserve setting also requires the specification of a duration period - at least when I change the reserve setting in the app. 

[image]

```
2024-02-06 17:05:07.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40                                                                                     
2024-02-06 17:05:08.077 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 4 4                                                                                       
2024-02-06 17:05:08.135 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 2 2                                                                                       
2024-02-06 17:05:08.135 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_DID 0x67 <class 'int'>                                                                                     
2024-02-06 17:05:08.135 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_Length 0x32 <class 'int'>                                                                                  
2024-02-06 17:05:08.136 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current 0xb8 <class 'int'>                                                                                        
2024-02-06 17:05:08.136 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_A 0x40 <class 'int'>                                                                                      
2024-02-06 17:05:08.136 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_B 0x3e <class 'int'>                                                                                      
2024-02-06 17:05:08.136 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_C 0x39 <class 'int'>                                                                                      
2024-02-06 17:05:08.137 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_SF -0x2 <class 'int'>                                                                                     
2024-02-06 17:05:08.137 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_AB 0x1013 <class 'int'>                                                                                   
2024-02-06 17:05:08.137 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_BC 0x1015 <class 'int'>                                                                                   
2024-02-06 17:05:08.137 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_CA 0x1015 <class 'int'>                                                                                   
2024-02-06 17:05:08.137 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_AN 0x949 <class 'int'>                                                                                    
2024-02-06 17:05:08.138 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_BN 0x94a <class 'int'>                                                                                    
2024-02-06 17:05:08.138 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_CN 0x947 <class 'int'>                                                                                    
2024-02-06 17:05:08.138 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_SF -0x1 <class 'int'>                                                                                     
2024-02-06 17:05:08.138 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Power 0xfd9 <class 'int'>                                                                                         
2024-02-06 17:05:08.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Power_SF -0x1 <class 'int'>                                                                                       
2024-02-06 17:05:08.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Frequency 0x138a <class 'int'>                                                                                    
2024-02-06 17:05:08.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Frequency_SF -0x2 <class 'int'>                                                                                   
2024-02-06 17:05:08.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_VA 0x1118 <class 'int'>                                                                                           
2024-02-06 17:05:08.139 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_VA_SF -0x1 <class 'int'>                                                                                          
2024-02-06 17:05:08.140 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_var -0x400b <class 'int'>                                                                                        [... cut ...]
2024-02-06 17:05:09.474 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor7 0x0 <class 'int'>                                                                              
2024-02-06 17:05:09.474 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor8 0x0 <class 'int'>                                                                              
2024-02-06 17:05:09.474 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 1.572 seconds (success: True)                                                
- home-assistant.log 5066/5066 100%
```

WillCodeForCats, 2024-02-06:
Modbus does not have a duration associated with the backup reserve setting as far as I am aware. There is a command timeout value, but that is related to the default mode and command mode settings.

The are additional settings you won't see in 2.4.9 in the Power Settings grouping in my example that will be available 2.4.10 or now in the pre-releases.

WillCodeForCats, 2024-02-06:
I also recall someone saying that modbus commands need to be sent to whichever inverter is the lead unit, but I don't remember where that discussion was.

WillCodeForCats, 2024-02-07:
Were you able to try changing the mode to Remote Control?

mtbsteve, 2024-02-08:
> Were you able to try changing the mode to Remote Control?

The behavior is somehow a bit strange. When I set the backup_reserve in the ioBroker modbus interface by writing the target percentage value into the E008/F708 register the value is correctly displayed in the app and, if the reserve is greater than the previous value, the battery will be automatically charged from the grid if there is no/insufficient PV production.
When I set the reserve value from within this adapter, the target percentage value is set correctly in the register, BUT its not updated in the app. I still need to test if the automatic charging from the grid is initiated once the new reserve setting is grater than the current value.

WillCodeForCats, 2024-02-08:
The integration is just writing modbus values, I don't see how it would be any different. There's enough users using commands that report it's fine that I don't have any reason to suspect there is a problem.

But my inverters do not accept commands so I can't help much beyond actual integrations bugs, since I can't test or use any commands on real inverter hardware or give any advice beyond what I can recall other people have said.

cyberkryten, 2024-02-12:
@mtbsteve I think this is due to a delay of the inverter reporting those figures back to the monitoring platform which is used to generate figures for the App.  

I also thought it wasn't working at first, but after some experimentation, the App 'catches up' if you leave it a while, so it's worth trying again and seeing what values show in the App 15min later.

I've noticed some differences between the values shown in the App and on the website even before I started changing them via Modbus so I think it's an issue on the monitoring side and not an issue with setting them locally via Modbus.

mtbsteve, 2024-02-14:
@cyberkryten its not updated in the app regardless how long I wait. Seems like SE changed something recently in their data structures. Not even when deleting the app and reinstalling it will show the reserve setting when I set it through the modbus register.
@WillCodeForCats I am closing this issue for now - in case you run over the root cause for this app behaviour let me know - Thanks for all your great work!

closed 2024-02-14 by mtbsteve