WillCodeForCats/solaredge-modbus-multi #528: State, device and unit of measurement classes not correct

markuswesterberg, 2024-02-07:
### Describe the bug

Hi. I'm getting error after update:

State classes is not correct
Translation Error: The intl string context variable "state_classes" was not provided to the string "Device classes {state_classes} on source entities {source_entities} needs to be same for sensor group {entity_id}.

Please correct the state classes on the source entities and reload the group sensor to fix this issue."

Device classes is not correct
Translation Error: The intl string context variable "device_classes" was not provided to the string "Device classes {device_classes} on source entities {source_entities} needs to be same for sensor group {entity_id}.

Please correct the device classes on the source entities and reload the group sensor to fix this issue."

Unit of measurements is not correct
Unit of measurements Öre/kWh, kWh of input sensors sensor.nordpool_kwh_se3_sek_0_10_0, sensor.importerade_kwh_h are not compatible using no device class of sensor group sensor.kostnad_el_timpris_ore_h.

Please correct the unit of measurements on the source entities or set a proper device class on the sensor group and reload the group sensor to fix this issue.

### Expected behavior

?

### Diagnostic File

?

### Home Assistant Version

2024.2.0

### solaredge-modbus-multi Version

Latest

### Installation Type

Supervised

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-02-07:
I'm not really sure what this means, since I don't see it in my language (English). Maybe someone who is more familiar with translations will be able to help.

WillCodeForCats, 2024-02-07:
Are you sure this is about my integration? I don't recognize the sensor names.

Please provide the required diagnostic file.

WillCodeForCats, 2024-02-09:
Closing because can't duplicate and requested information not provided.

markuswesterberg, 2024-02-09:
Sorry, I may have turned to you too soon. I have restored the back-up I had
and will try to find the issue myself.

markuswesterberg, 2024-02-10:
Still not working as expected. Being asked to reconfigure and entities become unavailable. Happens frequently. What do you need to analyze the issue? Home assistant log?

WillCodeForCats, 2024-02-10:
Start with the info I asked for.

markuswesterberg, 2024-02-10:
OK, so now I'm on HA core 2024.2.1. Attaching the debug log file here and also some screenshots. Let me know if you need something else? :)
[Home Asiistant log file debugging Modbus Multi.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14230444/Home.Asiistant.log.file.debugging.Modbus.Multi.txt)
[image]
[image]
[image]
[image]

markuswesterberg, 2024-02-10:
Sorry, wrong debug 
[home-assistant_solaredge_modbus_multi_2024-02-10T20-41-00.821Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14230447/home-assistant_solaredge_modbus_multi_2024-02-10T20-41-00.821Z.log)
log file I understand now attached before. Now available here instead the right one.

WillCodeForCats, 2024-02-10:
Look I'm thing not to get angry, but you are not listening to me.

UPLOAD THE DIAGNOSTIC FILE.

markuswesterberg, 2024-02-10:
I'm sorry :( Did I attach the right file now...? Or do you mean some other file? let me know how to generate the correct file if it's still the wrong file.

markuswesterberg, 2024-02-10:
Again, so sorry.I think you must have meant maybe the file attached here now:

[config_entry-solaredge_modbus_multi-fb9d2fa1828c4b72197701dd08beecbd.json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14230457/config_entry-solaredge_modbus_multi-fb9d2fa1828c4b72197701dd08beecbd.json)

WillCodeForCats, 2024-02-10:
First you need to remove `solaredge_modbus` integration. It does not use the right version of pymodbus and can cause problems.

markuswesterberg, 2024-02-10:
OK. I have removed it now and restarted HA. I was then (again, 5th time today I think) asked to reconfigure Modbus Multi. As usual after reconfiguration data is now available again. I assume it will become unavailable again in an hour or so but I'll let you know what happens. Let me know how to proceed. Appreciate your patience a lot :)

WillCodeForCats, 2024-02-10:
Unavailable is normal if there is a network connection issue and it can't connect to the inverter.

Starting in [Release v2.4.10-pre.5](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.10-pre.5) repair issues raised from setup failure will be automatically cleared after the first successful setup.

markuswesterberg, 2024-02-10:
There is no network issue I'm pretty sure because the SolarEdge App works well. Also, never had this issue before. Started when I updated HA and Modbus Multi. Do you mean that the issue I'm seeing will be resolved by the new releases of the Modbus Multi software? Should I update already now to the Release v2.4.10-pre.5?

WillCodeForCats, 2024-02-10:
You have long delays between reads. Each of those is 3 seconds apart, which is slow:

```
2024-02-10 21:40:13.726 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40
2024-02-10 21:40:16.744 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 4 4
2024-02-10 21:40:19.981 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 2 2
2024-02-10 21:40:23.024 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 4 4
2024-02-10 21:40:26.072 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 2 2
```

Something is causing a slowdown.

On my system these are fast:

```
2024-02-10 13:21:39.609 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40
2024-02-10 13:21:39.612 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 4 4
2024-02-10 13:21:39.650 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 86 86
2024-02-10 13:21:39.688 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 84 84
2024-02-10 13:21:39.691 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 4 4
2024-02-10 13:21:39.692 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 2 2
```

markuswesterberg, 2024-02-10:
Is it because I have set the configuration polling frequency to 5 seconds?
Or is it that my HA is on a server that is connected over WiFi and not
Ethernet? Either way, nothing changed and used to work as a charm before
updates :(

WillCodeForCats, 2024-02-10:
Turn on pymodbus debugging to see if there's a reason the reads are slow.

In configuration.yaml:

```
logger:
    logs:
        pymodbus: debug
```

markuswesterberg, 2024-02-10:
OK, done. And I have restarted HA also now.

WillCodeForCats, 2024-02-11:
Try [Release v2.4.11-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.1)

markuswesterberg, 2024-02-13:
OK. Will do. But what about the pymodbus debug logging? It's running now,
how do I get the file from it to send to you?

markuswesterberg, 2024-02-14:
Hi again. I decided to install the new version instead, 2.4.10. Didn't
understand how to install the pre versions :S But 2.4.10 did not resolve
the issue. Still have to reconfigure from time to time. Can you please help
me find the cause?

WillCodeForCats, 2024-02-16:
Starting in [Release 2.4.11-pre.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.7) the requirement for Keep Modbus Connection Open option has been removed for polling intervals less than 10 seconds.

Several other reports have this option causing connection problems that are fixed by turning this option off.