WillCodeForCats/solaredge-modbus-multi #535: Modbus connection randomly disconnects

mtbsteve, 2024-02-15:
### Describe the bug

I am experiencing random disconnects of the modbus connection. So about 1-2 times per day the reporting of values stops w/o an error message. I then need to manually reload the adapter in order to get the connection restarted. Also with other adapters connections sometimes break either due to network issues, or cloud service problems or whatever, but all other adapters I am using do an automatic reconnect in such a case. 

### Expected behavior

Please add an automatic reconnect service in case the modbus connection gets lost.

### Diagnostic File

[config_entry-solaredge_modbus_multi-c10193190040be4f930489a1a6dfb6fb.json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14296642/config_entry-solaredge_modbus_multi-c10193190040be4f930489a1a6dfb6fb.json)

### Debug logs

```shell
024-02-15 12:33:25.035 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 1.709 seconds (success: True)
2024-02-15 12:33:39.326 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 19.2
2024-02-15 12:33:45.470 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40
2024-02-15 12:33:48.529 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 4 4
2024-02-15 12:33:51.658 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 86 86
2024-02-15 12:33:54.759 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 84 84
2024-02-15 12:33:54.761 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_DID 0x67 <class 'int'>
2024-02-15 12:33:54.762 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_Length 0x32 <class 'int'>
2024-02-15 12:33:54.762 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current 0x31f <class 'int'>
2024-02-15 12:33:54.762 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_A 0x109 <class 'int'>
2024-02-15 12:33:54.763 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_B 0x10c <class 'int'>
2024-02-15 12:33:54.763 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_C 0x109 <class 'int'>
2024-02-15 12:33:54.763 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_SF -0x2 <class 'int'>
2024-02-15 12:33:54.764 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_AB 0x1019 <class 'int'>
2024-02-15 12:33:54.764 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_BC 0x101f <class 'int'>
2024-02-15 12:33:54.764 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_CA 0x1021 <class 'int'>
2024-02-15 12:33:54.766 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_AN 0x949 <class 'int'>
2024-02-15 12:33:54.766 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_BN 0x952 <class 'int'>
2024-02-15 12:33:54.767 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_CN 0x951 <class 'int'>
2024-02-15 12:33:54.767 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_SF -0x1 <class 'int'>
2024-02-15 12:33:54.767 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Power 0x497f <class 'int'>
2024-02-15 12:33:54.767 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Power_SF -0x1 <class 'int'>
2024-02-15 12:33:54.767 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Frequency 0x1387 <class 'int'>
2024-02-15 12:33:54.768 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Frequency_SF -0x2 <class 'int'>
2024-02-15 12:33:54.768 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_VA 0x4a63 <class 'int'>
2024-02-15 12:33:54.768 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_VA_SF -0x1 <class 'int'>
2024-02-15 12:33:54.768 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_var -0x7297 <class 'int'>
2024-02-15 12:33:54.768 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_var_SF -0x2 <class 'int'>
2024-02-15 12:33:54.768 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_PF -0x2698 <class 'int'>
2024-02-15 12:33:54.769 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_PF_SF -0x2 <class 'int'>
2024-02-15 12:33:54.769 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Energy_WH 0xc36ea <class 'int'>
2024-02-15 12:33:54.769 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Energy_WH_SF 0x0 <class 'int'>
2024-02-15 12:33:54.769 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_DC_Current 0x5b01 <class 'int'>
2024-02-15 12:33:54.769 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_DC_Current_SF -0x4 <class 'int'>
2024-02-15 12:33:54.769 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_DC_Voltage 0x2007 <class 'int'>
2024-02-15 12:33:54.770 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_DC_Voltage_SF -0x1 <class 'int'>
2024-02-15 12:33:54.770 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_DC_Power 0x4a9e <class 'int'>
2024-02-15 12:33:54.770 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_DC_Power_SF -0x1 <class 'int'>
2024-02-15 12:33:54.770 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Temp_Cab -0x8000 <class 'int'>
2024-02-15 12:33:54.770 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Temp_Sink 0x1092 <class 'int'>
2024-02-15 12:33:54.770 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Temp_Trns -0x8000 <class 'int'>
2024-02-15 12:33:54.771 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Temp_Other -0x8000 <class 'int'>
2024-02-15 12:33:54.771 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Temp_SF -0x2 <class 'int'>
2024-02-15 12:33:54.771 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Status 0x4 <class 'int'>
2024-02-15 12:33:54.771 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Status_Vendor 0x0 <class 'int'>
2024-02-15 12:33:54.771 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_RRCR 0x0 <class 'int'>
2024-02-15 12:33:54.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_Power_Limit 0x64 <class 'int'>
2024-02-15 12:33:54.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: I_CosPhi 0x3f800000 <class 'float'>
2024-02-15 12:33:54.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: CommitPwrCtlSettings 0x0 <class 'int'>
2024-02-15 12:33:54.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: RestorePwrCtlDefaults 0x0 <class 'int'>
2024-02-15 12:33:54.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: PwrFrqDeratingConfig 0x7 <class 'int'>
2024-02-15 12:33:54.772 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: ReactivePwrConfig 0x0 <class 'int'>
2024-02-15 12:33:54.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: ReactPwrIterTime 0xc8 <class 'int'>
2024-02-15 12:33:54.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: ActivePwrGrad 0x1 <class 'int'>
2024-02-15 12:33:54.773 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: FixedCosPhiPhase 0x3f800000 <class 'float'>
[... 86 lines cut ...]
2024-02-15 12:33:57.922 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0xffff <class 'int'>
2024-02-15 12:33:57.922 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: charge_limit 0x459c4000 <class 'float'>
2024-02-15 12:33:57.923 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: discharge_limit 0x459c4000 <class 'float'>
2024-02-15 12:33:58.528 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Timeout fetching SolarEdge Coordinator data
[... cut ...]
### Home Assistant Version

2024.2.1

### solaredge-modbus-multi Version

2.4.10

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

Siggy101, 2024-02-15:
I have also started getting this same behaviour today.  A reboot of HA brings it back online but only for a short time (maybe 15 minutes).  
I have this error in my log:

Log Details (ERROR)
This error originated from a custom integration.
Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:319
Integration: SolarEdge Modbus Multi ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues))
First occurred: 15:32:50 (1 occurrences)
Last logged: 15:32:50

Timeout fetching SolarEdge Coordinator data

Siggy101, 2024-02-15:
I can confirm that since rolling back to 2.4.9, the Modbus connection is now stable once more.
For info, here is the historical graph showing the disconnects since updating to 2.4.10 last night:

[image]

[image]

WillCodeForCats, 2024-02-15:
Please try [Release v2.4.11-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.2)

Siggy101, 2024-02-15:
Thanks. I'll install now and see how it goes tonight

Siggy101, 2024-02-15:
Can you release it as a beta?  I cannot seem to install it via HACS...

Siggy101, 2024-02-15:
Thank you. I have got it installed now.
Will leave it overnight and report back tomorrow.
Thanks for the speedy release. Let's hope it fixes the issue.

Siggy101, 2024-02-15:
I'm afraid it didn't fix it.
It lasted about 25 mins and has then disconnected again.

[image]

Siggy101, 2024-02-15:
Here's the log error:

WillCodeForCats, 2024-02-15:
[Release v2.4.11-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.3) removes the current limit entity and the matching modbus read.

WillCodeForCats, 2024-02-15:
If pre.3 doesn't change anything then next I'll try removing polling reads for the commit/restore functions.

Siggy101, 2024-02-15:
OK, installing now.
I'll report back in the next hour if it stops working, otherwise I'll report back that it is working in about 13 hours.

WillCodeForCats, 2024-02-15:
I went ahead and made [Release v2.4.11-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.4) so you can try it immediately if needed.

Siggy101, 2024-02-15:
Pre4 installed.
Will keep you posted.

Siggy101, 2024-02-15:
Pre4 has also done the same thing.

WillCodeForCats, 2024-02-15:
[Release v2.4.11-pre.5](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.5) removes the power reduce entity and associated read.

Siggy101, 2024-02-15:
Thanks.
Installed and testing

WillCodeForCats, 2024-02-15:
Upload your diagnostic file too.

Siggy101, 2024-02-15:
[config_entry-solaredge_modbus_multi-c62b22ee6af38c36522d4c8dca9025ea.json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14302655/config_entry-solaredge_modbus_multi-c62b22ee6af38c36522d4c8dca9025ea.json)

WillCodeForCats, 2024-02-15:
@mtbsteve 
> Please add an automatic reconnect service in case the modbus connection gets lost.

FYI this should already be addressed specifically for timeouts in [Release v2.4.11-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.1).

The real issue is the long delays when communicating with the inverter. If you look at the `Registers received requested : X Y` there are 3 seconds between them, which is not normal, but also I don't know why that is happening or what I can do about it.

Also try turning off the "Keep Modbus Open" option as someone on the HA forums reported disabling this helped their similar issue.

Siggy101, 2024-02-15:
I just just turned off 'keep modbus connection open".
Will see how it looks tomorrow

WillCodeForCats, 2024-02-15:
You have a 5 second polling interval, the integration currently forces that option for less than 10 seconds. I will need to make another pre-release to remove that condition.

WillCodeForCats, 2024-02-15:
[Release v2.4.11-pre.6](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.6) removes internal requirement for keep modbus open if polling interval is less than 10 seconds.

mtbsteve, 2024-02-15:
thanks! I am still on Pre4 which runs stable for the last couple of hours. I have Keep Modbus Open set and the polling interval set for 30sec. Will get back tomorrow morning.

WillCodeForCats, 2024-02-15:
I do not recommend using keep open unless there is a specific problem it solves.

Siggy101, 2024-02-16:
Pre6 with "keep modbus open" disabled and Pilling at 5s has worked perfectly overnight.

[image]

It looks like my issue was the "keep modbus open" then.

Thank you for the fix. Is it possible to remove the requirement to keep modbus open under 10s polling in your next release, but with all the other changes you were planning?

mtbsteve, 2024-02-16:
> I do not recommend using keep open unless there is a specific problem it solves.

Thanks. I disabled Keep Open and with Pre4 the connection kept stable overnight. Keeping my fingers crossed :-)

liquidv, 2024-02-16:
I don't think this is a modbus issue since I've been experiencing the same issue for the whole of last year.
When I lose connection to the Modbus it's because the Inverter/Battery resets according to Solaredge which happens twice a day. This can also be seen in the SE monitoring portal as you see only power being drawn from the grid while both Solar and battery goes idle for 1 minute, before restablishing comms again.

WillCodeForCats, 2024-02-16:
[Release 2.4.11-pre.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.11-pre.7) goes back to pre.1, but the keep modbus open requirement for polling intervals under 10 seconds removed so it will now always follow the option setting for any polling interval.

Siggy101, 2024-02-16:
Thanks.
I have installed pre7 and left my polling interval at 5s, with the keep alive unticked.  Has been working ok for 30 mins so will leave it overnight.

Incidentally, Pre6 was perfect all day. Not a single drop. I did see some errors in the log though...

[image]

The errors are still happening with pre7, but no drops observed....

mtbsteve, 2024-02-17:
@WillCodeForCats after 36 hours with Pre4, there are no more errors or disconnects in my setup.
Thanks a lot for your speedy resolution!

barrown, 2024-02-17:
For another datapoint: I was having the same original issue. After reading this thread I installed pre7 and unticked keep alive. Polling interval = 10s. I've experienced no permanent drop-outs overnight. So it looks fixed?

I did see this in the logs:
Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:345
Integration: SolarEdge Modbus Multi (documentation, issues)
First occurred: 05:11:00 (1 occurrences)
Last logged: 05:11:00
Error fetching SolarEdge Coordinator data: Modbus/TCP connect to 192.168.0.222:1502 failed.

But it's not a problem as it came back 10 seconds later.

As mtbsteve said, thanks for your speedy resolution, WillCodeForCats.

markuswesterberg, 2024-02-17:
> It is.
> 
> <img alt="Screenshot 2024-02-15 at 09 15 44" width="638" src="https://private-user-images.githubusercontent.com/48533968/305162701-da215aa1-076f-41d6-926b-36c413d55d1a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxNjgwMDUsIm5iZiI6MTcwODE2NzcwNSwicGF0aCI6Ii80ODUzMzk2OC8zMDUxNjI3MDEtZGEyMTVhYTEtMDc2Zi00MWQ2LTkyNmItMzZjNDEzZDU1ZDFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE3VDExMDE0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViYTQzZjc4Y2U1NmJjZGM5ODAwNzU3ZDhiNGE5MjhmMGRhMmRhNDliM2EyNmYxZGUwZDY5NTE0M2I4MTQ5YTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.cm6elVPUlfhptzAtIVvAuO3UmsesK4KSJcVM3nab83Q">

Can someone please explain to me how to install pre-releases? Do I have to enable someting in HA to see beta/pre-release versions in updates?

WillCodeForCats, 2024-02-17:
The connection errors are something I can't fix, unfortunately. I can only try to work around them. If the inverter doesn't accept a connection, it doesn't. The integration can detect that and will run through some retry attempts before it gives up. It's currently set to retry up to 5 times.

From the feedback it sounds like it's safe to take this into a final release so I'll do that shortly.

WillCodeForCats, 2024-02-17:
> Can someone please explain to me how to install pre-releases? Do I have to enable someting in HA to see beta/pre-release versions in updates?

In HACS you have to navigate into the integration card, then go to the menu and choose "Redownload". The you have to turn on the show beta versions switch.

[image]

addinsella, 2024-02-19:
Hi - I am still experiencing repeated disconnects since 2.4.10. I tried the various pre-releases in this thread and updated to 2.4.11. 

[image]

[image]

I have to re-load the integration manually, at which point things go OK for a short while but invariably gets lost again. Once the above error message fires and the entities become unavailable, additional errors show in logs as follows:

[image]

[image]

WillCodeForCats, 2024-02-19:
Upload your diagnostic file.

WillCodeForCats, 2024-02-19:
@addinsella Upload your diagnostic file.

addinsella, 2024-02-19:
Sorry - here is the diagnostic file - [config_entry-solaredge_modbus_multi-1b85b484d42a0c2df324517598ee7358 (1).json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14326470/config_entry-solaredge_modbus_multi-1b85b484d42a0c2df324517598ee7358.1.json)

Siggy101, 2024-02-19:
I am running 2.4.11 and I have experienced 2 more drops that required a reboot to restart data collection:

[image]

markuswesterberg, 2024-02-19:
Same problem here still with 2.4.11. Need to reload integration many times per day.

WillCodeForCats, 2024-02-19:
The last idea I have is clearing the pymodbus object on timeouts.

But if this doesn't help then someone who has this issue is going to have to try to analyze it and suggest a fix or submit a PR to me. Maybe it's related to the network, or how HA is installed, or the hardware, or inverter firmware, or a bug in the pymodbus library. In any case I don't see this problem personally so I can't duplicate it.

WillCodeForCats, 2024-02-19:
If you want to try different versions of the [pymodbus library](https://github.com/pymodbus-dev/pymodbus) you will need to edit the file `custom_components/solaredge_modbus_multi/manifest.json` and change the folowing line:

```
  "requirements": ["pymodbus>=3.5.4"],
```

Change `pymodbus>=3.5.4` to something like `pymodbus==3.5.4` to fix the version rather than use the current version HA ships with (I believe HA 2024.2.2 has pymodbus 3.6.3). The latest version is `pymodbus==3.6.4` which will probably be in HA 2024.3.

After changing this reboot Home Assistant to reload requirements since these are only loaded at startup.

You can try other versions listed here: https://github.com/pymodbus-dev/pymodbus/releases

Siggy101, 2024-02-19:
> [Release v2.4.12-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.12-pre.1) will clear the pymodbus object on timeouts.

Thank you.  I will install and try that version now.
I'm afraid I cannot help with a fix / PR as I have next to no coding skills.  I am very happy to continue to test and to submit logs etc to assist anyone who can suggest a fix.

mtbsteve, 2024-02-20:
> Same problem here still with 2.4.11. Need to reload integration many times per day.

just curious - are you using the Tibber integration in parallel?

mtbsteve, 2024-02-20:
@WillCodeForCats now I am having the first disconnect since I installed Pre4 on Feb 15. In the logs I observed exactly 4 drops in connection over the past days prior to the final disconnect this morning. Could it be that the 5 times-retry-counter is not reset once the interface establishes a reconnect? 
Also, why are you limiting it to 5 retries? Other integrations, such as eg the Shelly adapter are retrying indefinitely once a connection breaks.

WillCodeForCats, 2024-02-20:
There is no actual limit, it's 5 times with an escalating multiplier before the data is considered stale due to failure to refresh it and signals that to HA (unavailable state). Other integrations may prefer to keep showing stale data which appears on the graphs as an unchanging flat line.

You can change that value in `const.py` and adjust the settings under `class RetrySettings`

You can turn on debug to observe if the counter is reset. THe message `Timeout count X limit Y` logs the values before it gets reset to zero. The message `Refresh timeout X limit {RetrySettings.Limit}` shows the values as it increments.

WillCodeForCats, 2024-02-20:
I would recommend trying your own values based on observations, I can't provide any guidance on what would be appropriate since I can't duplicate this issue.

Making the internal retry very long will hide the connection errors and give you flat line graphs if you prefer that. But you can also end up with stale or bad data without any warning.

markuswesterberg, 2024-02-23:
> Example of fixed 5 seconds between attempts for up to 720 times (1 hour of
> retries between attempts, which does not include timeouts during the
> polling cycles, so longer in reality):
>
> class RetrySettings(IntEnum):
>     """Retry settings when opening a connection to the inverter fails."""
>
>     Time = 5000  # first attempt in milliseconds
>     Ratio = 1  # time multiplier between each attempt
>     Limit = 720  # number of attempts before failing
>
> You can try this and try changing the pymodbus library versions.
> Otherwise at the moment I'm out of ideas on this one.
>
> Thanks I configured the settings accordingly 👍

Did this resolve your issues?

addinsella, 2024-02-26:
I am running 2.4.12.pre-1 and the connection has been solid for a week now. Thanks @WillCodeForCats

Siggy101, 2024-02-27:
Another data point here:  I have been running .12Pre2 since it was released and I have not had any drops since.

sirfalo, 2024-02-28:
Unfortunatelly for me this is still an issue...
[image]

Do you want me to enable debuging and send logs?

WillCodeForCats, 2024-02-28:
There may be something common about the people that have this problem compared to those that don't (like me). I don't know what could be though. Hopefully someone will be able to duplicate this problem and suggest a fix.

Feel free to upload logs if you like, but I for now don't have any new ideas other than what I've already suggested to try. Sorry.

mtbsteve, 2024-02-28:
There was a Solaredge Server outage between 9:20am and 10:40am CET this morning. Access through the Solaredge Monitoring App, Modbus and Web API was not possible.
The modbus multi integration automatically restored the connection right after the SE monitoring platform was accessible again.
I am using the retrySettings in const.py as suggested by @WillCodeForCats  above. Seems that this works.

If others are still experiencing problems it would be helpful to list also the other HA adapters in use.
I am suspicious if there is a correlation with the Tibber access since I observed some coincidence with crashes of the HA Tibber link adapter.

WillCodeForCats, 2024-02-28:
> I am using the retrySettings in const.py as suggested by @WillCodeForCats above. Seems that this works.

I am investigating if it's possible to use a YAML config to set overrides for those so they can persist across updates, since altering the file will not. I don't really want those to be able to be changed through the UI options.

Siggy101, 2024-02-29:
> There was a Solaredge Server outage between 9:20am and 10:40am CET this morning. Access through the Solaredge Monitoring App, Modbus and Web API was not possible. The modbus multi integration [...]

Although mine seems to be working nicely now, I wanted to just confirm that I do not, and never have, run the Tibber addin.  (never heard of Tibber).
The only possible lead is that I did update my SolarEdge inverter's firmware just before this all started to go wrong.  Perhaps that is the missing link....

FranciscoBigWings, 2024-02-29:
> There was a Solaredge Server outage between 9:20am and 10:40am CET this morning. Access through the Solaredge Monitoring App, Modbus and Web API was not possible. The modbus multi integration [...]

It is not related directly to SE server disconnects. I have completely block internet connection of the inverter to SE servers and I have random disconnects since many versions ago. Before they were even more often.

When my modbus disconnects with the SE, I stop being able to control the storedge features as the inverter stops responding even I can keep the receiving readings after it reconnects the modbus between HA and inverter.

Full modbus reconnection and storedge responds only the morning after, when the inverter kind of “restores” when it starts producing energy.

I have also noticed that storedge values of maximum discharge and charge value set to 5400W after the disconnect happens

maksyms, 2024-02-29:
I'm getting the same issue since about 21/02/2024. Restarting HA does help. Prior to that, the connection was rock-solid.

[image]

maksyms, 2024-02-29:
I've read this thread in details and about to upgrade to the latest beta.

WillCodeForCats, 2024-02-29:
Also check if your inverter firmware changed with sensor `sensor.solaredge_i1_version`

The version of the pymodbus library can also change with various HA releases. The next release 2024.3 will have pymodbus version 3.6.4.

markuswesterberg, 2024-02-29:
Updating to pre12-2 seems to have resolved all my issues.

[19 more replies]

closed 2024-04-22 by WillCodeForCats